linuxboot/heads #1685: t420 (display port does not work after external flashing)

katharinamarriesdmitry, 2024-05-23:
## Context of the Build

**1. What board are you trying to build?**
t420-hotp-maximized
**2. What repository:branch are you using to build from?**
- [ ] Heads:Master
- [ x] Other (please specify)

**3. What version of coreboot are you trying to build**

CBET4000 Heads-v0.2.0-2147-g1e583e0
**4. In building the rom where did you get the blobs?**
- [ ] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ x] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
- [ ] Yes
- [ x] No

**6. What operating system are you using**
Qubes OS 4.2

## Please describe the problem
after flashing the qubes on my Lenovo 4238W2H t420 i72620M model the displayport does not work anymore. Before I had windows instaled and everything was okay. The VGA controller shows up.

**Describe the bug**
not sure whether it depends on dGPU driver problems? I assume that displayport should work. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

tlaurion, 2024-05-24:
Tagging t420 board testers as documented under https://github.com/linuxboot/heads/issues/692
- [t420](https://doc.coreboot.org/mainboard/lenovo/t420.html) (xx20): @alexmaloteaux @natterangell (iGPU) @akfhasodh @doob85

Can someone replicate and give additional technical information (cbmem, tweaks needed under coreboot config etc) so a PR for fix can be brought up?

I do not own the platform.

@nameisnotprovided Are you technical enough to give cbmem output, journalctl -f output in OS of what is happening when display is connected and see if this is a known upstream bug issue?

natterangell, 2024-05-24:
I don’t have a full size DP cable available right now, but I can check when I’m back in the office over the weekend. I have the igpu model though.

tlaurion, 2024-05-24:
This week has been crazy, not sure about coming station using iGPU.

If I don't reply to you early next week feel free to poke me back by tagging @tlaurion

fhvyhjriur, 2024-05-29:
I got a T420-dGPU for this bugreport do help with testing. And this is really just for this bugreport/heads-development. I really dont like nvidia as a company ( https://pic2.zhimg.com/v2-50e21fbf6935b6a4ee309d236fa54741_b.jpg ).

The issue with the one not working external graphics port you see is just normal when the dGPU is leaved unused or even especially disabled in coreboot.
For example with the Lenovo Thinkpad W500 there is a known issue with dying dGPU's inside the machine. Does this bother for example Libreboot users? No. The dGPU on W500 is disabled by default on Libreboot because its not free as in freedom. But a machine is really technically spoken broken when you have not installed coreboot/libreboot on a W500 with a "broken" dGPU.

(really offtopic quick note: You can fix the W500 dGPU by resoldering the dGPU chip. Its just the solder joints under the GPU that break because of heat problems when the thermal paste/connection get worse over time)

The dGPU on T420 is from the technical coreboot view side the same like dGPU on T520, T430 and T530 based on the information here: https://review.coreboot.org/c/coreboot/+/28392
> It can be tested from linux userspace using util/pmh7tool.
> 
> To turn dGPU power off:
>   pmh7tool -c 0x50 7
>   pmh7tool -c 0x50 3
> 
> To turn it on:
>   pmh7tool -s 0x50 3
>   pmh7tool -s 0x50 7

@tlaurion 
> Maybe all current board configs (not dgpu variants) should be migrated/renamed from BOARD_NAME to BOARD_NAME_iGPU to reduce confusion on what is supported by the board configs?
There was before for example w530-dgpu-K1000m-maximized and w530-dgpu-K2000m-maximized. But after they went to https://github.com/linuxboot/heads/tree/master/unmaintained_boards there is now no device listed with dGPU that could let you expect that when dGPU is not listed in the name, its unsupported.
I wont change now at the moment the naming. I would more likely add back more and more hardware to supported list. For example this t420 could get then the same naming design the now unsupported/untested W530-dGPU version.
When more and more dGPU versions get back listed as supported, this should help people get the logical conclusion that if the dGPU is not listed after the model name, then the dGPU is not enabled.

nameisnotprovided, 2024-05-30:
> I got a T420-dGPU for this bugreport do help with testing. And this is really just for this bugreport/heads-development. I really dont like nvidia as a company ( https://pic2.zhimg.com/v2-50e21fbf6935b6a4ee309d236fa54741_b.jpg ).
Thank you so much! I  probably just bought a wrong model with dGPU. I actually did not intend to do that just got hooked with i7. I updated the heads with new rom. No changes though…

> The dGPU on T420 is from the technical coreboot view side the same like dGPU on T520, T430 and T530 based on the information here: https://review.coreboot.org/c/coreboot/+/28392
Could you please explain how to get to the linux userspace? Should it be done from recovery heads shell?

natterangell, 2024-05-30:
Just confirming that DP works without issue on my iGPU T420.

nameisnotprovided, 2024-06-06:
@tlaurion 
@natterangell 
@fhvyhjriur 
Thank you all! The issue is solved by exchanging old motherboard to one without NVIDIA dGPU chip.

fhvyhjriur, 2024-07-24:
Maybe to finish this the productive way and because i have now the hardware:
@tlaurion 
Is there enough space left in the 8MiB SPI chip for the additional vbios file?
Would you like to create a test image with circleCI for me to test the Display Port connector with enabled Nvidia GPU?

tlaurion, 2024-07-24:
> Maybe to finish this the productive way and because i have now the hardware:
> @tlaurion 
> Is there enough space left in the 8MiB SPI chip for the additional vbios file?
> Would you like to create a test image with circleCI for me to test the Display Port connector with enabled Nvidia GPU?
> Adding t420-dgpu-maximized could also be seen as a preparation for https://github.com/linuxboot/heads/issues/1212 because w520 is much more often with dGPU then without.

I have no clue no time for this even less since nobody owns nor tested dgpu historically.

Reread https://github.com/linuxboot/heads/issues/1685#issuecomment-2129977916 @fhvyhjriur and please stop expecting me to do PR specific to boards I do not own. You should connect to coreboot community, find board owners that use dgpu and have them propose PR if they have a need for that.

It seems that you forget how FOSS works: there needs to be a need, there needs to be a itch, there needs to be technical enough people to propose tic for itch and time from reviewers to review and merge. All FOSS projects are like that.

The fact that you own a t420 dgpu doesn't make this model more used technically. It does not make anybody technical enough to do the job for a platform they own. It doesn't make issues troubleshooter and fixed by technicla enough people.

AFAIK all dgpu board went UNMAINTAINED.
Please again discuss in related issues. One asking to create t429 dgpu, not saying to a board owner please provide logs and that user deciding to buy iGPU variant. My time is limited and nobody uses t420 dgpu I'm sorry, until proven otherwise and those people involving themselves helping me support their need. Not the other way around.

closed 2024-07-25 by tlaurion