linuxboot/heads #1686: Review CONFIG_ONBOARD_VGA_IS_PRIMARY enablement for dGPU enabled boards

tlaurion, 2024-05-23:
Discussion at https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$tNUqVVh-o7ftSbP55-CzJsASEosES22x50A1AaOzIWk?via=matrix.org&via=nitro.chat&via=fairydust.space

Make sure we don't enable conflicting configs

fhvyhjriur, 2024-05-29:
I clicked on the link, after pressing many more buttons at the end i ended up with the note: "Tried to load a specific point in this room's timeline, but was unable to find it."

tlaurion, 2024-05-30:
> I clicked on the link, after pressing many more buttons at the end i ended up with the note: "Tried to load a specific point in this room's timeline, but was unable to find it."

Weird

https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$ryaE99mmRGOOjYMqx3slIvf_LIdzf6X4ex2wuA_BGwQ?via=matrix.org&via=nitro.chat&via=talk.puri.sm

> When `ONBOARD_VGA_IS_PRIMARY` is selected, then coreboot uses the FB from the GPU that is defined in the devicetree.cb. Otherwise the last VGA adpter discovered behind the PCIe ports is picked upWhen `ONBOARD_VGA_IS_PRIMARY` is selected, then coreboot uses the FB from the GPU that is defined in the devicetree.cb. Otherwise the last VGA adpter discovered behind the PCIe ports is picked up