linuxboot/heads #1695: Programmer selection

VoidWa, 2024-05-31:
What programmer do you recommend for installation? The message board is full of CH341A with black circuit boards. I don’t know if it’s 3 or 5 volts... Maybe there is a better solution? Please don't even mention raspberry pi and similar garbage.

nameisnotprovided, 2024-05-31:
I recently used a modified CH341A (3,3 V) with default EEPROM clip programmer offered by Novacustom (https://novacustom.com/product/modded-ch341a-bios-firmware-programmer-3v/ ). It worked out of the box. As a newcomer, I successfully flashed Skulls (x230) and Heads (t420) without any problems. From the reports, other variants should work too.

VoidWa, 2024-05-31:
To be honest, I'm scared to use the CH341A... Skulls says that 3.3V is not enough and you need to use an additional power supply. Coreboot says 5V is normal. Libreboot recommends 3.3V. Well, who to believe?

nameisnotprovided, 2024-05-31:
For me it worked without problems.

fhvyhjriur, 2024-06-26:
@VoidWa 
If you want the best known flasher, take the BeagleBone Black with a 3.3Volt additional dcdc converter (from its 5V). Its most stable flasher out there and the reason why its widely used by coreboot developers since many years. It also support EHCI-Debug for debugging thinkpads.
Yes, those broadcom raspberry pi are crap: https://www.fsf.org/resources/hw/single-board-computers

VoidWa, 2024-06-26:
This fee costs almost 9,000 in local currency. I bought a Thinkpad for 10,000.

tlaurion, 2024-06-28:
This thread is duplicate of documented
https://github.com/linuxboot/heads-wiki/issues/120

Not code related, therefore not relevant to code repo and heads-wiki related in the goal of updating https://osresearch.net.

Please participate there.
Closing here.

closed 2024-06-28 by tlaurion