linuxboot/heads #1711: Switch Haswell boards to NRI (native memory initialization)

tlaurion, 2024-07-02:
NRI work was revived in June 2023.

- [ ] Track https://review.coreboot.org/q/topic:%22haswell-nri-revived%22
- [ ] Have platforms owners involved 
  - [ ] Have them report no S3 regression
  - [ ] Have them report no ram init speed regression
  - [ ] Accept PR
  - [ ] Merge PR

srgrint, 2024-07-03:
I think you probably want to track https://review.coreboot.org/q/topic:%22haswell-nri%22 rather than https://review.coreboot.org/q/topic:%22haswell-nri-revived%22

I think the haswell-nri patch train is actually Angel's more recent version.  For example, it contains https://review.coreboot.org/c/coreboot/+/81897 which as I understand it is vital for working S3 resume.

tlaurion, 2024-07-03:
> I think you probably want to track https://review.coreboot.org/q/topic:%22haswell-nri%22 rather than https://review.coreboot.org/q/topic:%22haswell-nri-revived%22
>
> I think the haswell-nri patch [...]

@srgrint thanks for your input. Modified OP accordingly.

fhvyhjriur, 2024-07-14:
I am interested in testing NRI. I really do not like closed source software and the mrc.bin held me back for some years of getting a T440p(or W541). I did not like to have one more closed source part. Having one less closed source part in a computer is attractive advertisement. Having one more closed source part is bad reputation and should hold people back from getting the hardware.
Now i have a T440p after i find out the existence of NRI and already preflashed a existing maximized image to make it ready to install a NRI image.

fhvyhjriur, 2024-07-22:
I see you added **when ready upstream**
Like i wrote above, i am using on a T440p now NRI thanks to Libreboot releases and its working great. Also ran memtest without any issues. What about you release in the next days Heads NRI images for people like me that do not want to run any additional closed source parts if there is any code that could somehow replace it?
At the moment the images for T440p (and W541) look like this:
heads-t440p-maximized-v0.2.0-2229-g7b0d995.rom
heads-t440p-hotp-maximized-v0.2.0-2229-g7b0d995.rom

After we changed the logic behind the naming of the UNMAINTAINED and added the additional UNTESTED here: https://github.com/linuxboot/heads/tree/master/unmaintained_boards , maybe in combination with the logic you use for the modded x230 heads also support ( x230-maximized-fhd_edp ) and combine both additions with the warning in README.md  "Warning: Do not try to use UNTESTED_ images if you do not have a external flasher."

The image would be then named(following the examples above):
UNTESTED_heads-t440p-maximized-nri-v0.2.0-2229-g7b0d995.rom 
Because the NRI code is not merged upstream but of course testing people are needed, heads could help upstream by releasing such images.
I know you do not like writing documentation/wiki that much. I could rewrite some part of the heads-wiki to implement the now missing UNTESTED and also UNMAINTAINED documentation to the wiki. There i can also write about the NRI-images and why they have UNTESTED in their filename and how to report the tests to coreboot upstream.

In heads the filename UNTESTED_heads-t440p-maximized-nri-v0.2.0-2229-g7b0d995.rom would change to normal heads-t440p-maximized-v0.2.0-2229-g7b0d995.rom (removing the mrc.bin and not mentioning the historic mrc.bin or mentioning NRI any more) when upstream coreboot also merged it (stable release). The default images would then have NRI and no additional images with more closed source software any more released.

tlaurion, 2024-07-23:
@fhvyhjriur from OP

>Most important being

>https://review.coreboot.org/c/coreboot/+/81897 for S3

>I guess the whole patchset being important for ram init being par if not superior in speed and stability when merged.

> @srgrint thanks for your pointers at #1711 (comment)

@fhvyhjriur is suspend to ram (s3) working from libreboot with NRI? I said "once merged upstream" because as per past users reports, there were important regressions both on performance and features.

tlaurion, 2024-07-23:
@Th3Fanbus (Angel from pending patchsets upstream): This issue is about users willing to test roms produced for a pointed commit I could create a PR against and for technical enough users to report back upstream under coreboot if that would help merging this faster, feel free to tag me.

Cross-post under coreboot matrix channel at https://matrix.to/#/!EhaGFZyYcbyhdSgStq:matrix.org/$sn5yonF4e3fJE4aCUGxfQv_xENoq80BeK6h5hqwQnHw?via=matrix.org&via=sibnsk.net&via=nope.chat

@fhvyhjriur I could create testing PR in draft for you to test on a coreboot commit to be tested by upstream; that is easy enough for me to do for one target to be tested if you are willing to take that task of communicating results upstream and if upstream is ready to push this forward. This is not my call and could be discussed with coreboot devs directly following my message there if you will.

A reminder that Heads is a coreboot user, where forks are coreboot devels maintained. On community boards, we try to stay with upstream, and we are behind at least 2 releases most of the time because a lot of things break unfortunately. So if this can happen upstream, I'm willing to modify modules/coreboot to use upstream coreboot + commit, modify one board target for it to use NRI and for you to take responsibility of reporting upstream if there is consent this is useful for them to push this forward. I have no clue of the priority of this for Angel nor have followed NRI improvements and status in libreboot.

fhvyhjriur, 2024-07-23:
@Th3Fanbus Great to see you following this thread. I want to thank you for the NRI Haswell code. This is what changed the situation for me to get a Haswell based device.
I think the most logical thing would be to ask you if you want people having a simplified (prebuild images) ability to run the heads code combined with NRI code and report using such images issues to you.
Or do you share the opinion that Libreboot prebuild images with NRI are enough for testing?

tlaurion, 2024-10-30:
@Th3Fanbus aith current situation outlined at https://github.com/linuxboot/heads/issues/1825#issuecomment-2448636649 it might be a good moment for Heads to switch to NRI for t440p and w541 since augmenting CBFS region results in boot time of >50s with ram init from blobs borrowed from Chromebook.

Would you be willing to analyse cbmem logs provided by board testers if I do PR with testable roms directly built from CI if you point me to a coreboot commit you would like those users to test and iterate?

Otherwise Heads users will boot in >50s for those platforms until NRI is ready :(

tlaurion, 2024-11-07:
Also note official opened issue on coreboot side which didnt get any real attention:  https://ticket.coreboot.org/issues/457

tlaurion, 2025-01-24:
Also see https://github.com/linuxboot/heads/issues/1825#issuecomment-2613236203

tlaurion, 2025-01-26:
Input for https://github.com/linuxboot/heads/issues/1825

tlaurion, 2025-03-09:
>Hopefully there should be movement here with Haswell NRI - all the remaining parts of the patch train have now been given +2, so presumably should be getting merged upstream imminently

_Originally posted by @srgrint in https://github.com/linuxboot/heads/issues/1825#issuecomment-2708516959_

NRI has been merged today.

tlaurion, 2025-03-09:
> > @gaspar-ilom amazing news yesterday on that front: NRI got merged upstream, meaning that the merge patch train is 24.12 + a lot of patches, but, if the patch trail splitted in small commits patches under Heads, and those individual patches adapted to be applied cleanly on top of this PR (later work), then we will have NRI for t440p w541, finally.
> 
> I think there are different options:
> 1. Pull in all the patches and maintain them under 24.12. as you suggested.
> 2. We could also wait for the next coreboot release which is not far away (25.03!?) and bump all boards again or will the testers just hate us too much?
> 3. Or we put this PR on hold yet again and reset the coreboot commit to fetch 24.12 from to 5d291de6011a56bfd767c4bcdfdc3aa6ee87a2dd (current HEAD of main after haswell NRI was merged). Coreboot 24.12 in Heads is anyway pulling an arbitrary commit 70 commits ahead of the release right now: `git log 24.12..2f1e4e5e8515dd350cc9d68b48d32a5b6b02ae6a`
> 
> Personally, I have no strong preference. I just thought that 2 and 3 might be cleaner and less maintenance work as we could ideally just forget about haswell NRI. 1. might have the advantage that we only need to test haswell boards (just W541 and T440p iirc).

To be done in another PR in all case. 

Specific points:
2. T480 won't be merged under 25.03. And all other downstream forks are rebasing on 24.12. As you can see here, it's a lot of work. Last bump was 24.02 (Feb 2024) and I cannot afford to do multiple version bumps in a year unless I have an apprentice /co-maintainer since this is unpaid work, and a lot of work.

Therefore option needs to be 1 until there is synchronized efforts/movements from downstream forks as well. Heads cannot build 10 versions of coreboot under CI under master. I don't anticipate so much work needed to cherry-pick adapt upstream patches, and the effort done here will be done and reusable until next version bump, which should give us a break for a while unless there's a vul discovered under 24.12 which would speed things up for next coreboot version bump. But yet again, this is intuition based and trying to apply patch train to 24.12 git tree might be a disaster, only effort will tell. Discussions about this should continue under NRI issue :)

Just wanted to point you that NRI landed. And t440p/w541 not first citizen because of mrc blob being problematic and adding delays of boot up to 50s per prior tests. There was also myth of being able to increase ram capacity to those machines, limitation thought of being driven by preppy Chromebook blob borrowed, not memory controller nor CPU. 

_Originally posted by @tlaurion in https://github.com/linuxboot/heads/pull/1908#issuecomment-2709038705_

tlaurion, 2025-03-14:
Folloring rabbit hole to

Leads to https://github.com/linuxboot/heads/issues/1324, which leads to https://github.com/linuxboot/heads/issues/1711#issuecomment-2206793780, which leads to https://github.com/linuxboot/heads/issues/1324

Quote from https://github.com/linuxboot/heads/issues/1324 on why we got 51s when increasing CBFS region:
> When developing [#1282](https://github.com/osresearch/heads/pull/1282), I built Heads with an initial CBFS size of 11 MiB. I encountered long boot times compared to other Heads users (close to 30 seconds). Looking at `cbmem` timestamps, we discovered that the `after RAM initialization stage` took approximately 17 seconds. I rebuilt heads with a CBFS size of 8 MiB, and boot times returned to normal (about 9 seconds). While debugging, I also built Heads with a CBFS size of 8192 KiB + 64 KiB, which bricked the laptop.
> 
> We think that this is related to Cache-as-RAM (CAR) and the [Haswell MRC](https://doc.coreboot.org/northbridge/intel/haswell/mrc.bin.html). In the chatroom, [@tlaurion](https://github.com/tlaurion) [said](https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$ylkEmWA-qKgl2LN9S703lnoHLDVj7d58pVdN0Cbsw_g?via=matrix.org&via=nitro.chat&via=talk.puri.sm):
> 
> > icon mentioned before that there's CAR at 0xff7c0000..0xff7effff (I think that was referring to haswell) Taking 0xffffffff - 0xff7effff = 0x810000 = 8454144 = 8256KiB, so if the BIOS region is greater than that the memory range mapped to the BIOS flash will overlap the CAR range
> 
> [@tlaurion](https://github.com/tlaurion) also met with a Coreboot developer at FOSDEM who [suggested](https://ticket.coreboot.org/issues/457/) updating `DCACHE_RAM_BASE` from `0xff7c0000` to `0xfe7c0000`. However, [@srgrint](https://github.com/srgrint) [said](https://github.com/osresearch/heads/pull/1282#issuecomment-1442446129) that that bricked his machine.
> 
> I have a few questions:
> 
>     1. How does `DCACHE_RAM_BASE` work?
> 
>     2. The T440p ROM is 12,288 KiB and starts 12,228 KiB below 4 GiB address space. Right now, `DCACHE_RAM_BASE` starts 8,448 KiB below 4 GiB address space. Why does the Coreboot developer in https://ticket.coreboot.org/issues/457/ suggest that we make `DCACHE_RAM_BASE` start 24,832 KiB below 4 GiB address space?
> 
>     3. Is `DCACHE_RAM_BASE` related to the `RW_MRC_CACHE` cbfstool section?
> 
> 
> $ ./build/x86/coreboot-4.17/util/cbfstool/cbfstool ./build/x86/t440p-hotp-maximized/heads-t440p-hotp-maximized-v0.2.0-1366-g25b4f60-dirty.rom layout -w
> This image contains the following sections that can be accessed with this tool:
> 
> 'BIOS' (read-only, size 8388608, offset 4194304)
> 'RW_MRC_CACHE' (size 65536, offset 4194304)
> 'FMAP' (read-only, size 512, offset 4259840)
> 'COREBOOT' (CBFS, size 8322560, offset 4260352)
> 
>     4. How did icon determine the CAR region goes to `0xff7effff`? I can't find this number in the Coreboot source code. `0xff7effff - 0xff7c0000` (their suggested end - start) is exactly 192 KiB - 1 byte. The size of the `UNIFIED_MRC_CACHE` in the  [`src/mainboard/lenovo/haswell/vboot-rwab.fmd`](https://github.com/coreboot/coreboot/blob/master/src/mainboard/lenovo/haswell/vboot-rwab.fmd) flash descriptor (FMD) file is only 128 KiB. And `DCACHE_RAM_SIZE` is 64 KiB.
> 
> 
> [@tlaurion](https://github.com/tlaurion) [said](https://github.com/osresearch/heads/pull/1282#issuecomment-1442522156):
> 
> > No, it is thought that mrc.bin needs to be patched. While referred issue tells that first iteration of native memory init will give lower dram speeds for the moment, so first iteration of this PR goes with recommendation of having cbfs size of 8mb unfortunately.
> 
> So maybe `DCACHE_RAM_BASE` is set the way it is because the MRC expects that cache to be at a specific address?
> 
> We should answer these questions. Also, initially, we wanted to supply `cbmem` logfiles with timestamps for different variations of CBFS size before and after the slowdown. Is this still useful?

ajpauwels, 2025-06-13:
@tlaurion 

I have a t440p I'm interested in getting running with heads and nri. It's already running libreboot w/ seabios fine. I have the hardware and knowledge to flash the chip with a programmer.

From my understanding, you'd like to pull the nri patch train into patches/coreboot-24.12 folder in the heads repo, make the t440p board, and test it. I am very new to custom firmware/heads and am dipping my toes in the water so would just like to confirm this. From having read through the Makefile, it seems like it reads the patches/ directory, identifies if there's a single patch file/a directory of patch files, and runs a `git apply ...` on each one, so this seems like the way I should go. If I've misunderstood, I apologize for the inconvenience, but please let me know.

Once I have that running, I assume you'll need some sort of evaluation that things work. From the first post in this thread, you want to check that S3 suspend works, and that there isn't a RAM init speed regression. The first I can check, the second one I'm not sure what you're looking for exactly. I see a reference to `cbmem -t` for timing the boot sequence, but I'll admit I'm not sure where to run/set this in order to get the output you're looking. I will continue digging and will join the coreboot channel in the meantime.

Thank you for your help!

tlaurion, 2025-06-13:
hey @ajpauwels !

This issue deserves a little update, which I will try to do here quickly.

https://github.com/linuxboot/heads/pull/1923#issuecomment-2907903906 is the last update from NRI patches. There is a lot of uncertainty as of now on why there is boot delay, including (search under the pull request, there was a lot of things discussed there, new blood might help)
- NRI not supporting ME neutered.
- Not quite clear on where the delays are happening
- `cbmem -1` (that is a one) is to be ran within Heads recovery shell (that you can force skip most of the init script by typing 'r' repeatedly as soon as you see Heads asciiart. 
- There is a patch to be tested on top of @1923 by a technical enough developer. As you can see under #1923 there is some slowing down of the effort because its not so trivial to do and extracting logs requires plugging some specialized equipment.

TLDR: the t440p under Heads currently uses chromebook preppy's MRC blob, which results in the machine booting with last numbers I remember being 30 seconds as documented under #1825. If you are not technical enough to risk bricking/unbricking and redoing @gaspar-ilom raspberri pico debug dongle; then you should not attempt it.

Reminder of next steps:
- https://github.com/linuxboot/heads/pull/1923#issuecomment-2907903906 points to patchset https://review.coreboot.org/87830 needing to be added atop a fork of  https://github.com/linuxboot/heads/pull/1923's https://github.com/gaspar-ilom/heads/tree/haswell-nri branch, build and test that rom.
- reminder that roms build and downloadeable from Heads circleci (per official docs download instructions) will use MRC blob, not NRI.

closed 2025-11-17 by tlaurion