linuxboot/heads #1717: heads fail to take the right USB dongle when 2 are connected

aluciani, 2024-07-12:
**Is your feature request related to a problem? Please describe.**
Under nitropad-nv41, and librem_14, when a kernel modification occurs (update for example), and you want to re-sign the parameters, if it turns out that you have two USB gpg dongles (in my case a nitrokey 3 type usb a and a yubikey type usb c), heads can't recognize which USB dongle corresponds to the GPG key inserted in heads.
In my case: heads should use the nitrokey to sign the new parameters, but tries with the yubikey and fails to sign the parameters. 
One solution (the only one I can think of) should be to unplug the yubikey, and go through the parameter signing procedure again.
Except that at that point the parameters are no longer good and heads fail to sign the parameters, and to continue you have to boot, and rm the kexec files in /boot.
Maybe I'm missing a step, but I didn't see a tutorial in the wiki to solve this problem. 

**Describe the solution you'd like**
Heads should be able to differentiate which usb dongle should be used to sign kexec parameters probably by checking the gpg keys contained in both USB dongles before using the dongle to sign them.
There's also the problem of "corruption" of kexec files, I don't know what happens but even if you remove the USB dongle that doesn't contain the gpg key, it's no longer possible to sign kernel parameters, and you have to delete the kexec files, and restart the whole signing procedure.

**Describe alternatives you've considered**
-None yet

**Additional context**
Perhaps we want to create a way out of the fact that the kexec parameters are corrupted and need to be deleted in order to re-sign the kernel parameters.
I didn't check the differences between the kexec files before and after the alteration. I ll put more information (screenshot, kexec fiels) this week end when I got more time.

tlaurion, 2024-07-12:
I think this is a matter of https://github.com/linuxboot/heads/discussions/1521 again.

For example, I provisioned my USB Security dongle and included  proper identifiers at provisioniong. Will reboot and modifiy this comment with screenshot

tlaurion, 2024-07-12:
Will think about that. 

But mitigation today is putting nail polish/permanent marker on USB Security Dongle to differenting the even if physically identical.  

But yet again I'm not sure if it's a bug or a feature: the old detached /boot files digest signature is being deleted/replaced only after a successful gpg --detach-sign call on a temporary created digest for /boot files. 

Only easy fix today would be to loop instead of failing on that code part until the user inserts the right USB security dongle. Otherwise, we would have to bind serial numbers to public key or something on re-ownership/Factory reset, but asking the user to insert a USB security dongle by available tags won't help the User eXperience (UX) further : that would be a serial number the USB Security Dongle brand name as written under /boot. But we coukd do either. 

@JonathonHall-Purism @aluciani : thoughts?

aluciani, 2024-07-14:
> **Bug or feature?** But yet again I'm not sure if it's a bug or a feature: the old detached /boot files digest signature is being deleted/replaced only after a successful gpg --detach-sign call on a temporary created digest for /boot files. An attacker attempting tampering of unnencrypted /boot would probably try a dongle and fail, or try to sign /boot content digest without plugging the right dongle too.

I agree that it's good that the attacker trying to modify the /boot files can't do it.
My specific problem is that I use a nitrokey for heads and a yubikey to sign things and connect to ssh. I leave my yubikey connected all the time, it's a usb c and doesn't take up much space.
The problem is (I think) that heads still tries to sign kexec files with the wrong key (in my case the yubikey) and corrupts kexec files.
 
> **Quick fix** Only easy fix today would be to loop instead of failing on that code part until the user inserts the right USB security dongle.
> 
> **Refactoring** Otherwise, we would have to bind serial numbers to public key or something on re-ownership/Factory reset, but asking the user to insert a USB security dongle by available tags won't help the User eXperience (UX) further : that would be a serial number the USB Security Dongle brand name as written under /boot. But we could do either.
> 
> **Suggested fix** Using branding in whiptail prompt might be better combined with loop? But that won't help if two dongles of same branding are being connected at the same time or subsequently either. Showing serial number won't help. Asking user to insert usb dongle matching public key identifier won't help if user doesn't identify its dongle properly
> 
> **Question** Are we trying to fix something with code that should be documentation? Ie: "identify usb dongles properly"?

The usb dongles have IDs, is it possible to select the right USB dongle by id if multiple detected ?
And if the right USB security dongle is not inserted, inform the user that the corresponding USB security dongle is not inserted, as is already the case.

> Not the first time this issue is raised. Searching in closed issues should reveal duplicates.

I'm sorry, I tried to look into the issue by filtering, but didn't saw it ...

tlaurion, 2024-07-16:
> I agree that it's good that the attacker trying to modify the /boot files can't do it. My specific problem is that I use a nitrokey for heads and a yubikey to sign things and connect to ssh. I leave [...]

@aluciani important distinction: Heads provides tamper detection/evidence not tamper prevention. This means someone could change your /boot files but signatures won't match and detached signature verification will fail... Hope we are at the same page

> The usb dongles have IDs, is it possible to select the right USB dongle by id if multiple detected ? And if the right USB security dongle is not inserted, inform the user that the corresponding USB [...]

In your specific use case where at factory reset/re-ownership the branding of the nitrokey/librem key was defined (/boot/kexec_hotp_key existing or not) the message could easily tell you to plug the proper one, but from the way gpg works with ID being release by gpg --card-status upon keyring to match private key ID, there is not much that can be done here
1- Heads asks to plug the proper key, then load usb modules and stuff
2- Heads prompts usb dongle with gpg --card-status
3- at this point, it will fail if you already had yubikey connected and there is no easy way to handle this properly but retrying, asking again the user to insert properly branded key

As said, this is a user management issue here an you having multiple keys to use on same machine is pretty uncommon (where most users don't even use their usb security dongles for anything else then HOTP + detached signatures of /boot digest (checksum files: /boot/kexec*.txt -> kexec.sig file)

So unfortunately, the easiest is to simply not boot with any usb dongle connected and insert the one prompted by first solution suggested here as quickfix: looping until user surrenders to inseting the proper one and reminding him 3 times prior of failing (common UX thing) 

> 
> > Not the first time this issue is raised. Searching in closed issues should reveal duplicates.
> 
> I'm sorry, I tried to look into the issue by filtering, but didn't saw it ...

I will find he trail later on, i'm more interested into fixing this then justifying why it wasn't yet.

@JonathonHall-Purism insights would be insightful if you could complement my brain process here.

aluciani, 2024-07-16:
> @aluciani important distinction: Heads provides tamper detection/evidence not tamper prevention. This means someone could change your /boot files but signatures won't match and detached signature verification will fail... Hope we are at the same page

I'm totally on the same page as you, the problem is not related to /boot files being tempered.

> In your specific use case where at factory reset/re-ownership the branding of the nitrokey/librem key was defined (/boot/kexec_hotp_key existing or not) the message could easily tell you to plug the [...]

I'd like to point out that the correct key (nitrokey 3 mini) is plugged in when I start the computer, as is the yubikey.
But heads prefers to take the yubikey instead of the nitrokey 3 mini, probably because of PCI preference .....

> As said, this is a user management issue here an you having multiple keys to use on same machine is pretty uncommon (where most users don't even use their usb security dongles for anything else then [...]

Totally agree, my point here was just to see if a easy fix was possible. it seems it isn't, I will remember to remove the yubikey when rebooting after kernel update.
Thanks for the answer

tlaurion, 2024-07-19:
@aluciani what should be fixed under heads-wiki to clarify this once and for all?

aluciani, 2024-07-19:
> @aluciani what should be fixed under heads-wiki to clarify this once and for all?

That maybe it's better to remove any USB security dongle (yubikeys and others) and plug in only the corresponding one when you want to update kernel settings.

This should be enough

tlaurion, 2024-07-19:
@aluciani 

- What page at https://osresearch.net
- What is leaving user to interpret
- what should be read instead

Thanks. I will create a PR since you don't, but I need feedback, not be left to create changes that won't resolve the issue to your eyes. Thnaks for your valued collaboration. Let's economize back and forth, should we? I value your time as I hope you value mine.

aluciani, 2024-07-19:
closing as https://github.com/linuxboot/heads-wiki/pull/158
(late commit of install-os.md is the signing thing)

closed 2024-07-19 by aluciani