linuxboot/heads #1718: Create CODE_OF_CONDUCT.md

tlaurion, 2024-07-12:
Draft
```
## LinuxBoot/Heads Project Code of Conduct ### Our Pledge In the interest of fostering an open and welcoming environment, we as contributors and maintainers pledge to make participation in our project and our community a harassment-free experience for everyone, regardless of age, body size, disability, ethnicity, sex characteristics, gender identity and expression, level of experience, education, socio-economic status, nationality, personal appearance, race, religion, or sexual identity and orientation. ### Our Principles 1. **Openness**: We embrace open source principles and encourage transparency in our work. 2. **Inclusivity**: We welcome contributors from diverse backgrounds and experiences. 3. **Respect**: Treat all community members with respect and kindness. 4. **Collaboration**: Work together constructively to improve the project. 5. **Technical Merit**: Focus on technical excellence and innovative solutions. ### Communication Channels and Their Purposes For the most up-to-date information on our community channels, please visit our community page: https://osresearch.net/community/ #### 1. Matrix Channels - **Heads Channel**: https://matrix.to/#/#heads:matrix.org - Purpose: Primary platform for community discussions and support. - **Heads-Changes Channel**: https://matrix.to/#/#heads-changes:matrix.org - Purpose: Changelog channel for all GitHub changes. Best Practices: - Use threads for specific topics to keep conversations organized. - Aim to consolidate discussions into actionable GitHub issues or pull requests. - Use for topics not yet addressed on GitHub or in the documentation. #### 2. GitHub Repositories - **linuxboot/heads**: https://github.com/linuxboot/heads - Purpose: Main repository for code, development, and user experience improvements. - **linuxboot/heads-wiki**: https://github.com/linuxboot/heads-wiki - Purpose: Repository for project documentation. - Note: The content of heads-wiki is rendered and searchable at https://osresearch.net ### Best Practices for Issues, Discussions, and Contributions #### Before Opening an Issue or Starting a Discussion 1. **Search Existing Resources**: - Use GitHub's search function in both repositories (heads and heads-wiki) to check if your topic has already been reported or discussed. - Use the search bar at the top of https://osresearch.net to search the entire documentation. - Search the Matrix channel history for related discussions. - Try different keywords related to your topic. - Check closed issues as well, as your problem might have been resolved previously. 2. **Review the Documentation**: - Thoroughly read relevant sections of the documentation on https://osresearch.net. - Check if your question is answered in the FAQ or troubleshooting sections. 3. **Consider the Appropriate Channel**: - For general questions or discussions, use the Heads Matrix channel. - For potential bugs or feature requests, consider opening a GitHub issue. - For documentation improvements, use the heads-wiki repository. #### When Starting a Discussion on Matrix - Clearly state your topic or question. - Provide context, including your environment (OS, hardware, etc.) if relevant. - Be patient and respectful while waiting for responses. #### When Opening an Issue on GitHub - Clearly describe the problem or suggestion. - Include steps to reproduce for bugs. - Mention your environment (OS, hardware, etc.) if relevant. - Use labels appropriately to categorize your issue. #### For Pull Requests - Link to related issues or discussions if applicable. - Provide a clear description of the changes and their purpose. - Ensure your code adheres to the project's coding standards. - Be responsive to feedback and be prepared to make adjustments. ### Expected Behavior Examples of behavior that contributes to creating a positive environment include: - Using welcoming and inclusive language - Being respectful of differing viewpoints and experiences - Gracefully accepting constructive criticism - Focusing on what is best for the community - Showing empathy towards other community members ### Unacceptable Behavior Examples of unacceptable behavior include: - The use of sexualized language or imagery and unwelcome sexual attention or advances - Trolling, insulting/derogatory comments, and personal or political attacks - Public or private harassment - Publishing others' private information, such as a physical or electronic address, without explicit permission - Other conduct which could reasonably be considered inappropriate in a professional setting ### Our Responsibilities Project maintainers are responsible for clarifying the standards of acceptable behavior and are expected to take appropriate and fair corrective action in response to any instances of unacceptable behavior. ### Scope This Code of Conduct applies within all project spaces, and it also applies when an individual is representing the project or its community in public spaces. ### Enforcement Instances of abusive, harassing, or otherwise unacceptable behavior may be reported by contacting the project team at [INSERT EMAIL ADDRESS]. All complaints will be reviewed and investigated promptly and fairly. The project team is obligated to maintain confidentiality with regard to the reporter of an incident. Project maintainers who do not follow or enforce the Code of Conduct in good faith may face temporary or permanent repercussions as determined by other members of the project's leadership. ### Attribution This Code of Conduct is adapted from the Contributor Covenant and reflects the community values of the LinuxBoot/Heads project, as well as best practices from other open firmware projects.
```

tlaurion, 2024-07-18:
@SergiiDmytruk I see you downvoted this issue, comments?

tlaurion, 2024-07-19:
Code of conduct will vanish once I unbrick my nv41. Woukd appreciate review of collaborating.md still

fhvyhjriur, 2024-07-23:
I also downvoted it. My explanation can be find here: https://github.com/linuxboot/heads/pull/1719#issuecomment-2244024191

closed 2024-07-29 by tlaurion