linuxboot/heads #1725: What to do after a Firmware flash?OEM Factory, Boot from usb stick or Clean Boot detected?

SirRufusss, 2024-07-19:
##Request for Assistance with QubesOS Installation

### Nitropad NS50 | v2.5.0 FW_WER: Dasharo Nitrokey-v2.5.0

1. What board are you using? (Choose from the list of boards [here]
Sorry i really dont know

2. Does your computer have a dGPU or is it iGPU-only?
    - [ ] dGPU (Distinct GPU other then internal GPU)
    - [ ] iGPU-only (Internal GPU, normally Intel GPU)
    I am not sure but SytemInfo says: CPU: 12th Gen Intel Core i7-1260P

3. Who installed Heads on this computer?
    - [ ] Insurgo (Issues to be reported at https://github.com/linuxboot/heads/issues)
    - [ X ] Nitrokey (Issues to be reported at https://github.com/Nitrokey/heads/issues)
    I will make after this issue a second issue on nitrokey, i hope this is okay
    - [ ] Purism (Issues to be reported at https://source.puri.sm/firmware/pureboot/-/issues)
    - [ ] Novacustom (Issues to be reported at https://github.com/Dasharo/dasharo-issues)
    - [ ] HardnenedVault (Issues to be reported at https://github.com/hardenedvault/vaultboot/issues)
    - [ ] Other provider
    - [ ] Self-installed

4. What PGP key is being used?
    - [ ] Librem Key (Nitrokey Pro 2 rebranded)
    - [ ] Nitrokey Pro
    - [ ] Nitrokey Pro 2
    - [ X ] Nitrokey 3 NFC
    - [ ] Nitrokey 3 NFC Mini
    - [ ] Nitrokey Storage
    - [ ] Nitrokey Storage 2
    - [ ] Yubikey
    - [ ] Other

5. Are you using the PGP key to provide HOTP verification?
    - [ ] Yes
    - [ ] No
    - [ X ] I don't know

### B. Identify how the board was flashed

1. Is this problem related to updating heads or flashing it for the first time?
    - [ ] First-time flash
    - [ X ] Updating heads 

2. If the problem is related to an update, how did you attempt to apply the update?
    - [ X ] Using the Heads menus
    - [ ] Flashrom via the Recovery Shell
    - [ ] External flashing

3. How was Heads initially flashed?
    - [ ] External flashing
    - [ ] Internal-only / 1vyprep+1vyrain / skulls
    - [ X ] Don't know

4. Was the board flashed with a maximized or non-maximized/legacy rom?
    - [ ] Maximized
    - [ ] Non-maximized / legacy
    - [ X ] I don't know

5. If Heads was externally flashed, was IFD unlocked?
    - [ ] Yes
    - [ ] No
    - [ X ] Don't know

### C. Identify the rom related to this bug report

1. Did you download or build the rom at issue in this bug report?
    - [ X ] I downloaded it
    - [ ] I built it

2. If you downloaded your rom, where did you get it from?
    - [ ] Heads CircleCi
    - [ ] Purism
    - [ X ] Nitrokey
    - [ ] Dasharo DTS (Novacustom)
    - [ ] Somewhere else (please identify)

    *Please provide the release number or otherwise identify the rom downloaded*

3. If you built your rom, which repository:branch did you use?
    - [ ] Heads:Master
    - [ X ] Other i used the Nitrokey rom (https://github.com/Nitrokey/heads/releases/tag/v2.5.0)

4. What version of coreboot did you use in building?
{ You can find this information from github commit ID or once flashed, by giving the complete version from Sytem Information under Options --> menu}

5. In building the rom, where did you get the blobs?
    - [ ] No blobs required
    - [ ] Provided by the company that installed Heads on the device
    - [ ] Extracted from a backup rom taken from this device
    - [ ] Extracted from another backup rom taken from another device (please identify the board model)
    - [ ] Extracted from the online bios using the automated tools provided in Heads
    - [ X ] I don't know

## Please describe the problem

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Main Menu -->option ---> Flash/Update the BIOS'
2. Click on 'Flash the firmware with a new ROM, erase Settings'
3. Scroll down to 'Flash it and reboot the PC'
4. Now i dont know what to do...

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Without Nitrokey:
[image]
 With Nitrokey:
 

[image]

**Additional context**

Hello HEADS-Team,

I hope this message finds you well. I am seeking assistance with the installation of QubesOS following an update of the HEADS.rom.

The documentation indicates that the next step is to "boot from USB stick," which seems straightforward: Boot Options > Boot from USB.
However, I am encountering a problem: I do not see the usual HEADS menu. The first message I receive upon restarting my PC is: "Error GPG Keyring empty." I am presented with the following options:
Add a GPG key to the running BIOS
OEM Factory Reset / Re-Ownership
Ignore error and continue to main menu
Exit to recovery shell
When I start my PC with the Nitrokey plugged in, I receive the message: "CLEAN Boot detected - Perform OEM Factory Reset / Re-Ownership?" My options here are to continue or cancel.
Summary of My Problem:
The documentation advises booting from a USB stick.
Without the Nitrokey, I see an error message with four options.
With the Nitrokey, I am prompted to perform an OEM Factory Reset.
I would greatly appreciate any guidance or tips on how to proceed in this situation.
Thank you in advance for your support!

So my Problem is:
Documentaion says: Boot from USB-Stick
HEADS without Ntrokey: Says Error, here are 4 options
HEADS with Nitrokey says: Do a OEM Factory Reset / Re-Ownership"

Please tell me: Please give me some guidance, a tip or anything on what to do next?

tlaurion, 2024-07-19:
@SirRufusss : Please read official docs and reply here if you still have questions. You are proposed the right actions, since your Nitrokey is either unprovisioned or you flashed Heads without giving it prior public key that matches your provisioned USB Security dongle. Hitting cancel would lead you to main screen, from where documentation should answer other questions.

TLDR: Heads doesn't know your USB Security dongle.
When booting with it attached, it detects a clean install, proposing you to factory reset. What is the problem if that is true?

Documentation:
- https://osresearch.net/
- See Installing and configuring. Click on it
- See steps 1,2,3,4? 
- Step 3: https://osresearch.net/Configuring-Keys/
- Step 4: https://osresearch.net/InstallingOS/

Feel free to close this issue at your earliest convenience.

closed 2024-07-19 by tlaurion

SirRufusss, 2024-07-20:
Please excuse me if I still have questions.

I have really given this a lot of thought and have decided to do an “OEM Factory Reset / Re-Ownership”.
Because, as you said, there is nothing to be said against it.

So HEADS asked me about the new flashing of the BIOS and the OEM Factory Reset, which should be the default boot and strangely enough my first attempts to install Fedora appeared. But shouldn't everything be deleted after a reflash?
Anyway, I decided against Fedora anyway and just want to install QubesOS as recommended

I navigated to the main menu and wanted to continue with the installation of Qubes, I selected “Boot Options > Boot from USB” as described in the documentation.

I can assure you that I have a USB stick in Exfat4 format, and I dragged the QubesOS.iso, the Qubes-R4.2.2-x86_64.iso.sig and the “qubes-release-4.2-signing-key.asc” onto it and checked it with sha256sum after the download.

Unfortunately I still can't boot from the USB stick and I can't find any hint under the section “Installing Qubes 4.X”.
It only tells me that I should have done the following:
“Plug in the EXT3/EXT4/ExFat formatted USB stick containing Qubes iso and iso.asc files into one of the USB port and boot it from USB mode:”

And to my knowledge I have fulfilled exactly these conditions

Here is a picture of my screenshot:
[image]

And here is the error code i get, if i chosse booting from usb-stick:
[image]

tlaurion, 2024-07-20:
> Please excuse me if I still have questions.
>
> I have really given this a lot of thought and have decided to do an “OEM Factory Reset / Re-Ownership”.
> Because, as you said, there is nothing to be said [...]

Only iso and iso.sig/iso.asc are needed to be on root of the ext3/ext4/ExFat usb thumb drive partition. 

> Unfortunately I still can't boot from the USB stick and I can't find any hint under the section “Installing Qubes 4.X”.
> It only tells me that I should have done the following:
> “Plug in the [...]

From recovery shell, do mount-usb. 
then mount. 

Five screenshot. 
> 
> 
> Here is a picture of my screenshot:
> [image]
> 
> 
> And here is the error code i get, if i chosse booting from usb-stick:
> [image]
> 

This is still fedora.

tlaurion, 2024-07-21:
> I can assure you that I have a USB stick in Exfat4

ExFat and ext4 are not the same. Please provide output of

mount-usb
mount

With your qubesos iso+iso.asc file on usb thumb drive connected to usb3 port.

----

As per actionable items from this issue :
- I guess step 3 and 4 should be both step 3, one stating : if OS pre-installed, launch re-ownership wizard, the other being if no OS pre-installed: boot from USB

- where GUI detects it and could propose the right thing to the user.

@SirRufusss @JonathonHall-Purism thoughts?

SirRufusss, 2024-07-25:
Unfortunately, I was busy with work over the past few days.
As requested, I have checked my USB stick, and it is formatted in ext4 and contains the QubesOS .iso and .asc files. I have verified both files using PGP, fingerprint, and sha256.
When I start into HEADS and try to boot from the USB stick, I only see the QubesOS.iso file, which I then select.
After that, I see the following in the Recovery shell:
"gpg: Cant't check signature on kexec: No public key"

[image]

I really can't think of any reason why the ISO signature would fail. As I understand it, the public key is in the .asc file, and I have checked this file for correctness and integrity during the download. This translation maintains the clarity and detail of your original message, ensuring that the recipient will understand your situation accurately.