linuxboot/heads #1726: Why i have 6Admins after flashing the heads.rom (Connected device status in secure-shell)

SirRufusss, 2024-07-20:
## Context of the Build

**1. What board are you trying to build?**

**2. What repository:branch are you using to build from?**
- [ ] Heads:Master
- [ ] Other (please specify)

**3. What version of coreboot are you trying to build**

**4. In building the rom where did you get the blobs?**
- [ ] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ ] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
- [ ] Yes
- [ ] No

**6. What operating system are you using**

*please specify*

## Please describe the problem

**Describe the bug**
After reflashing the heads.rom I still have old .iso files....
In addition, the secure-shell unetr connected devices is set to “admin = 6”.
I think admin are the users and I don't understand why I have more than one...

I used this .rom: https://github.com/Nitrokey/heads/releases/tag/v2.5.0

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'mainmenu ---> options --> Flash/Update the BIOS'
2. Click on 'Flash the firmware with a new ROM, erase settings'
WITH this .rom: https://github.com/Nitrokey/heads/releases/tag/v2.5.0
3. Scroll down to '....'
4. NO ERROR, i got the messaged: Succesfully flashed"

**Expected behavior**
As I understand it, the old .rom will be replaced by the new .rom.

The hard disk and the GPG (i.e. the Nitrkey) should be deleted by a TPM re ownership, which I did...
According to the documentation, TPM is the SecureChip of the hardware.

**Screenshots**
[image]

**Additional context**
Hello HEADS-Team,

can someone please tell me why everything is not reset/deleted after flashing the HEADS.rom?

I can still find my previous attempts to install Fedora.iso in the boot options...

I also received a notification after flashing that the falshing was successful.

Afterwards I also performed an OEM factory reset, which as far as I know also deletes my Nitrokey (i.e. GPG).
Nevertheless I could see with the ESCAPE button in the main menu that I am in the 6Admin usage...
I even checked the nitrokey on my 2nd PC and it is completely empty

SirRufusss, 2024-07-20:
when i flash new my hard disk does not erase.
How do I get the previous .iso files deleted?
Can I access it from the recovery shell and do this?

But I'm still wondering about the 6Admins...

tlaurion, 2024-07-20:
@nestire @daringer @jans23 : hello Nitrokey.
As referred in past issues, nitrokey seems to miss official documentation on what this hotp-verification returned 4.11 version of what applet, and why there are counters going to more then 3 as in traditional opengpg smartcard. I will leave this question for you to answers and to refer here to duplicates of issues raised in your repo.

The user here uses nitrokey firmware and nitrokey 3, I will answer for Heads related questions.

tlaurion, 2024-07-20:
> when i flash new my hard disk does not erase.
> How do I get the previous .iso files deleted?
> Can I access it from the recovery shell and do this?
> 
> But I'm still wondering about the 6Admins...

Your screenshot shows that Heads attempts to hash a lot of things in it's /boot partition. No, Heads doesn't wipe disks. I station instructions are pretty clear under https://osresearch.net where detached signed isos or installation media dd'ed to usb thumb drive should permit installation and where /boot partition needs to be created, seperated of root partition for Heads to not spin trying to hash everything under the same partition. That seems to be your issue.

The screenshot says that heads waits 5 seconds prior of attempting to boot. Any key press typed there will lead you to main menu, where recovery shell is under options menu.

From there it's typical Linux commands. If you want to delete files, you will have to remount filesystem in read write, or cal fdisk and wipe your hard drive, or boot from USB thumb drive again to reinstall your OS properly as per documentation.

tlaurion, 2024-07-22:
Took liberty to rename issue

tlaurion, 2024-07-22:
This issue is about counters and firmware reported from hotp-verification on boot output.

Other issues need to be opened seperately.
This issue will be left open until Nitrokey answers, others issues need to be dealt seperately.

Thank you for your cooperation @SirRufusss .

nestire, 2024-07-22:
The Counters indicate how much trys you have left to enter a wrong pin in thiy case the GPG Pins . This used to be 3 but do to requirements for FIDO2 the Certification we increased this. Please see also here https://docs.nitrokey.com/nitrokey3/linux/set-pins if you have more questions please contact support@nitrokey.com

tlaurion, 2024-07-22:
@SirRufusss diccsuccion should continue at https://github.com/Nitrokey/nitrokey-hotp-verification/issues/36, closing this issue as duplicate of upstream opened issue

closed 2024-07-22 by tlaurion