linuxboot/heads #1736: librem_14 board local build errror

aluciani, 2024-07-27:
## Context of the Build

**1. What board are you trying to build?**
librem_14

**2. What repository:branch are you using to build from?**
- [x] Heads:Master

**3. What version of coreboot are you trying to build**
Coreboot-purism which is I think 24.02.01

**4. In building the rom where did you get the blobs?**
- [X] Extracted from the online bios using the automated tools provided in Heads

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
- [X] Yes

**6. What operating system are you using**
Debian 12

*please specify*

## Please describe the problem

**Describe the bug**
I'm trying to build the librem_14 board on a debian 12 system. But I get an error in the docker image :
```
2024-07-27 06:45:16+00:00 MAKE coreboot-purism
tail /home/user/heads/build/x86/log/coreboot-purism.log
-----

Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20230628
Copyright (c) 2000 - 2023 Intel Corporation

coreboot toolchain v2024-02-18_732134932b
ASL Input:     dsdt.asl -   41461 bytes   1282 keywords      0 source lines
AML Output:    dsdt.aml -   13711 bytes    742 opcodes     540 named objects

Compilation successful. 0 Errors, 0 Warnings, 26 Remarks, 385 Optimizations, 39 Constants Folded
    IASL       3150 warning types were ignored!
    IASL       librem_14/dsdt.aml disassembled correctly.
Makefile:303: *** Missing NSS. Please install libnss3-dev if it is not installed..  Stop.
make[1]: *** [src/security/vboot/Makefile.mk:54: librem_14/external/vboot_reference-ramstage/vboot_fw.a] Error 2
Makefile:303: *** Missing NSS. Please install libnss3-dev if it is not installed..  Stop.
make[1]: *** [src/security/vboot/Makefile.mk:55: librem_14/external/vboot_reference-postcar/vboot_fw.a] Error 2
Makefile:303: *** Missing NSS. Please install libnss3-dev if it is not installed..  Stop.
make[1]: *** [src/security/vboot/Makefile.mk:50: librem_14/external/vboot_reference-bootblock/vboot_fw.a] Error 2
make[1]: Leaving directory '/home/user/heads/build/x86/coreboot-purism'
make: *** [Makefile:563: /home/user/heads/build/x86/coreboot-purism/librem_14/.build] Error 1
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install a bare debian 12 OS
2. Follow the docker and nix install on heads README
```
$ [ -d /nix ] || sh <(curl -L https://nixos.org/nix/install) --no-daemon
$ . /home/user/.nix-profile/etc/profile.d/nix.sh
$ mkdir -p ~/.config/nix
$ echo 'experimental-features = nix-command flakes' >>~/.config/nix/nix.conf
$ nix --print-build-logs --verbose build .#dockerImage && docker load < result
```
4. run ```ocker run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) tlaurion/heads-dev-env:latest -- make BOARD=librem_14```
5. See error

**Expected behavior**
Build the librem_14 rom

**Additional context**
I think the board build on CircleCI so I don't know why in local it's not building, maybe libnss3-dev is cached or something ?

srgrint, 2024-07-27:
You need to install libnss3-dev .  This wasn't needed until the latest coreboot bump yesterday.

aluciani, 2024-07-27:
> You need to install libnss3-dev . This wasn't needed until the latest coreboot bump yesterday.

yes, but packages installed on debian should not be visible in the docker image, no ?

srgrint, 2024-07-27:
> > You need to install libnss3-dev . This wasn't needed until the latest coreboot bump yesterday.
> 
> yes, but packages installed on debian should not be visible in the docker image, no ?

So does it need adding to the flake.nix file?

tlaurion, 2024-07-27:
Really confused by this. Seems you are not on master.

```
user@localhost:~/heads$ cat flake.nix | grep nss
        nss
        openssl # needed for talos-2 kernel build
```

Or that you have not pulled latest nor v0.2.1 docker image to use docker:
`user@localhost:~/heads$ time docker run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) tlaurion/heads-dev-env:latest -- make BOARD=librem_14`

Which latest = v0.2.1 at https://hub.docker.com/r/tlaurion/heads-dev-env/tags:
`sha256:80c66b6f7a4e69a7c05016fa4756b9755a30146fff8049f3967d4ac27211fa8f`

Which both include nss.

tlaurion, 2024-07-27:
Maybe that's ecause coreboot-purism per modules/coreboot depends on 24.02.01 buildstack and CircleCI build dependencies builds and cache x230-hotp-maximized which builds 24.02.01 first.

Restarting on clean commit locally. Will edit

tlaurion, 2024-07-27:
Renamed issue since this is README.md nix docker image replication/nix based docker image replication that is unclear, and not a bug with the buildsystem itself to not let others in the belief that things are brocker outside of replicability of README.md instructions to use nix and flakes.nix, or docker usage and git cloning, or using git pointing at master, not sure.

tlaurion, 2024-07-27:
https://osresearch.net/general-building/
https://github.com/linuxboot/heads/blob/master/README.md#pull-docker-hub-image-to-prepare-reproducible-roms-as-circleci-in-one-call

Things clear to my eyes, but as usual, this is not a guarantee its clear to yours so please modify so that clear to yours @aluciani @srgrint :)

aluciani, 2024-07-28:
```
$ [ -d /nix ] || sh <(curl -L https://nixos.org/nix/install) --no-daemon
$ . /home/user/.nix-profile/etc/profile.d/nix.sh
$ mkdir -p ~/.config/nix
$ echo 'experimental-features = nix-command flakes' >>~/.config/nix/nix.conf
$ sudo rm -rf ./heads
$ git clone https://github.com/linuxboot/heads
$ docker -v
Docker version 27.1.1, build 6312585
$ nix --version
nix (Nix) 2.23.3

tail /home/user/heads/build/x86/log/coreboot-purism.log
-----
    CP         ramstage/lib/program.ld
    GCC        ramstage/libgfxinit/common/dyncpu/hw-gfx-gma-config.o
    GCC        ramstage/libgfxinit/common/haswell_shared/hw-gfx-gma-connectors-ddi.o
    GCC        ramstage/libgfxinit/common/haswell_shared/hw-gfx-gma-connectors.o
    GCC        ramstage/libgfxinit/common/haswell_shared/hw-gfx-gma-ddi_phy_stub.o
    GCC        ramstage/libgfxinit/common/haswell_shared/hw-gfx-gma-port_detect.o
    GCC        ramstage/libgfxinit/common/hw-gfx-dp_aux_ch.o
    GCC        ramstage/libgfxinit/common/hw-gfx-dp_defs.o
    GCC        ramstage/libgfxinit/common/hw-gfx-dp_dual_mode.o
    GCC        ramstage/libgfxinit/common/hw-gfx-dp_info.o
Makefile:303: *** Missing NSS. Please install libnss3-dev if it is not installed..  Stop.
make[1]: *** [src/security/vboot/Makefile.mk:52: librem_14/external/vboot_reference-romstage/vboot_fw.a] Error 2
make[1]: *** Waiting for unfinished jobs....
Makefile:303: *** Missing NSS. Please install libnss3-dev if it is not installed..  Stop.
make[1]: *** [src/security/vboot/Makefile.mk:54: librem_14/external/vboot_reference-ramstage/vboot_fw.a] Error 2
Makefile:303: *** Missing NSS. Please install libnss3-dev if it is not installed..  Stop.
make[1]: *** [src/security/vboot/Makefile.mk:55: librem_14/external/vboot_reference-postcar/vboot_fw.a] Error 2
Makefile:303: *** Missing NSS. Please install libnss3-dev if it is not installed..  Stop.
make[1]: *** [src/security/vboot/Makefile.mk:50: librem_14/external/vboot_reference-bootblock/vboot_fw.a] Error 2
make[1]: Leaving directory '/home/user/heads/build/x86/coreboot-purism'
make: *** [Makefile:563: /home/user/heads/build/x86/coreboot-purism/librem_14/.build] Error 1
```

** I MADE A MISTAKE ON THE OS, I M USING DEBIAN TRIXIE ** (not that it shoudl change something on the build......)
```
$ uname -a
Linux workstation 6.9.10-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.9.10-1 (2024-07-19) x86_64 GNU/Linux
```

tlaurion, 2024-07-28:
> ```
> $ [ -d /nix ] || sh <(curl -L https://nixos.org/nix/install) --no-daemon
> $ . /home/user/.nix-profile/etc/profile.d/nix.sh
> $ mkdir -p ~/.config/nix
> $ echo 'experimental-features = nix-command flakes' >>~/.config/nix/nix.conf
> $ sudo rm -rf ./heads
> $ git clone https://github.com/linuxboot/heads
> $ docker -v
> Docker version 27.1.1, build 6312585
> $ nix --version
> nix (Nix) 2.23.3
> 
> $ docker run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) tlaurion/heads-dev-env:latest -- make BOARD=librem_14
> 
> ....
> 
> tail /home/user/heads/build/x86/log/coreboot-purism.log
> -----
>     CP         ramstage/lib/program.ld
>     GCC        ramstage/libgfxinit/common/dyncpu/hw-gfx-gma-config.o
>     GCC        ramstage/libgfxinit/common/haswell_shared/hw-gfx-gma-connectors-ddi.o
>     GCC        ramstage/libgfxinit/common/haswell_shared/hw-gfx-gma-connectors.o
>     GCC        ramstage/libgfxinit/common/haswell_shared/hw-gfx-gma-ddi_phy_stub.o
>     GCC        ramstage/libgfxinit/common/haswell_shared/hw-gfx-gma-port_detect.o
>     GCC        ramstage/libgfxinit/common/hw-gfx-dp_aux_ch.o
>     GCC        ramstage/libgfxinit/common/hw-gfx-dp_defs.o
>     GCC        ramstage/libgfxinit/common/hw-gfx-dp_dual_mode.o
>     GCC        ramstage/libgfxinit/common/hw-gfx-dp_info.o
> Makefile:303: *** Missing NSS. Please install libnss3-dev if it is not installed..  Stop.
> make[1]: *** [src/security/vboot/Makefile.mk:52: librem_14/external/vboot_reference-romstage/vboot_fw.a] Error 2
> make[1]: *** Waiting for unfinished jobs....
> Makefile:303: *** Missing NSS. Please install libnss3-dev if it is not installed..  Stop.
> make[1]: *** [src/security/vboot/Makefile.mk:54: librem_14/external/vboot_reference-ramstage/vboot_fw.a] Error 2
> Makefile:303: *** Missing NSS. Please install libnss3-dev if it is not installed..  Stop.
> make[1]: *** [src/security/vboot/Makefile.mk:55: librem_14/external/vboot_reference-postcar/vboot_fw.a] Error 2
> Makefile:303: *** Missing NSS. Please install libnss3-dev if it is not installed..  Stop.
> make[1]: *** [src/security/vboot/Makefile.mk:50: librem_14/external/vboot_reference-bootblock/vboot_fw.a] Error 2
> make[1]: Leaving directory '/home/user/heads/build/x86/coreboot-purism'
> make: *** [Makefile:563: /home/user/heads/build/x86/coreboot-purism/librem_14/.build] Error 1
> ```
> 
> 
> **I MADE A MISTAKE ON THE OS, I M USING DEBIAN TRIXIE** (not that it should change something on the build......)
> ```
> $ uname -a
> Linux workstation 6.9.10-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.9.10-1 (2024-07-19) x86_64 GNU/Linux
> ```

Exactly. It should matter if you actually constructed the docker image (which your repro doesn't do above) by continuing following steps from master commit see my repo logs, you miss doing

https://github.com/linuxboot/heads/tree/1519bd74fe5aa16bb930165a8649cdfe45b8f6dc?tab=readme-ov-file

I see no
nix --print-build-logs --verbose develop --ignore-environment --command true

nix --print-build-logs --verbose build .#dockerImage && docker load < result

Which construct the nix docker image and use it.

Then https://github.com/linuxboot/heads/tree/1519bd74fe5aa16bb930165a8649cdfe45b8f6dc?tab=readme-ov-file#jump-into-nix-develop-created-docker-image-for-interactive-workflow

docker run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) linuxboot/heads:dev-env -- make BOARD=nitropad-nv41

Which are all steps to create reproducible docker image which is advanced use which you should not do.

The only thing you need to do is at https://github.com/linuxboot/heads/tree/1519bd74fe5aa16bb930165a8649cdfe45b8f6dc?tab=readme-ov-file#pull-docker-hub-image-to-prepare-reproducible-roms-as-circleci-in-one-call

Which uses recreated docker image for end users (above prior steps are for developers not end users) 

You only need to do, as explicited example to use recreated docker image which will download use and build :
docker run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) tlaurion/heads-dev-env:latest -- make BOARD=x230-hotp-maximized
docker run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) tlaurion/heads-dev-env:latest -- make BOARD=nitropad-nv41

@aluciani at this point, I expect a PR because clearly the documebtariin was nit clear enough to you, the end user, not the developer. Tha KS in advance.

tlaurion, 2024-07-29:
@aluciani : Tldr you are reusing past self-created docker image from old flake.nix that was not containing nss.

Read prior edited comment. If going that path, you cannot skip steps.

tlaurion, 2024-07-29:
@aluciani do

`docker image list`
`docker image rm IMAGE_ID_FROM_PREVIOUS_COMMAND`

This weird situation happens if "latest" matches locally built IMAGE_ID from reproducible builds. For users only using latest, this is not an issue. For people recreating docker image from "result" produce nix image, this is not an issue either.

I just ran in the same issue on an old dev qube that was using docker image v0.1.9 on which I was trying to make qemu target and failed with similar results because nss was not in docker image v0.1.9.

calling
`docker run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) tlaurion/heads-dev-env:latest -- make BOARD=qemu-coreboot-whiptail-tpm1`

built as expected.

---
Now, how should this be documented properly?

aluciani, 2024-07-29:
First of all I apologize, I'm a stupid sandwich, I hadn't read the documentation properly. It's clear and I don't think it needs updating on my part (at least for now).
I'm sorry for the delay in replying, I'm not very available at the moment, and the build is taking a long time.
Anyway, I've just launched it with all the advice (delete the docker image etc), I'll keep you posted.
Do you know if, when I run the 

```docker command run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) tlaurion/heads-dev-env:latest -- make BOARD=whatever``` 

if the docker image is not up to date, is it updated automatically?
Anyway, after deleting the docker image, re pull, I get this :
```
2024-07-29 14:02:47+00:00 MAKE coreboot-purism
tail /home/user/heads/build/x86/log/coreboot-purism.log
-----
    CC         postcar/soc/intel/common/block/cpu/car/exit_car.o
    CC+STRIP   src/cpu/intel/fit/fit_table.c
    HOSTCC     cbfstool/cbfstool.o
    HOSTCC     cbfstool/cbfs_image.o
    HOSTCC     cbfstool/cbfs-mkstage.o
    HOSTCC     cbfstool/cbfs-mkpayload.o
    HOSTCC     cbfstool/partitioned_file.o
    HOSTCC     cbfstool/platform_fixups.o
convert: insufficient image data in file `../../../branding/Heads/bootsplash.jpg' @ error/jpeg.c/ReadOneJPEGImage/1136.
convert: no images defined `bmp:librem_14/mainboard/purism/librem_cnl/cbfs-file.2pZPsr.out' @ error/convert.c/ConvertImageCommand/3362.
convert: unable to open image 'librem_14/mainboard/purism/librem_cnl/cbfs-file.2pZPsr.out': No such file or directory @ error/blob.c/OpenBlob/3596.
convert: no decode delegate for this image format `OUT' @ error/constitute.c/ReadImage/746.
convert: no images defined `jpg:librem_14/mainboard/purism/librem_cnl/cbfs-file.2pZPsr.out' @ error/convert.c/ConvertImageCommand/3362.
make[1]: *** [Makefile:315: librem_14/mainboard/purism/librem_cnl/cbfs-file.2pZPsr.out] Error 1
make[1]: *** Waiting for unfinished jobs....
vboot SHA256 built with tight loops (slower, smaller code size)
vboot SHA256 built with tight loops (slower, smaller code size)
vboot SHA256 built with tight loops (slower, smaller code size)
vboot SHA256 built with tight loops (slower, smaller code size)
make[1]: Leaving directory '/home/user/heads/build/x86/coreboot-purism'
make: *** [Makefile:563: /home/user/heads/build/x86/coreboot-purism/librem_14/.build] Error 1
```

I think this is a symlink issue no ?

tlaurion, 2024-07-30:
As for your symlink issue, yes branding/Heads/bootsplash.jpg is a symlink that you might have broken, but if so, another user issue which cannot be reproduced and would deserve redoing symlink, and unrelated to the original issue opened here. A clean git clone would also show you how things are supposed to work. coreboot config embeds a path to the directory of bootsplash.jpg with a static name, and bootsplash.jpg is a symlink to a bootsplash. Its easier to change a symlink then a config file; therefore the path I chose to implement, where forks can decide for their own what is best for them and maintain the way then intend to do. IF this is an issue, open another one and work with me keeping things seperated.

If this issue is fixed, please close it @aluciani

closed 2024-07-30 by aluciani