linuxboot/heads #1744: Report: The x230-maximized image is working now on x230t without the backspace keyboard error

fhvyhjriur, 2024-08-05:
@tlaurion wrote many times and took this even in the year 2024 often as a example for how missing testing is blocking adding the support for a device. The example: x230t. This was famous for not having a working backspace key when a heads x230 image was installed on it. I can report from years ago, that this was the case. The keyboard was not working as expected but back then, i had not the chance to test further/again.
Now i have again access to a x230t and the first thing i did was to install https://output.circle-artifacts.com/output/job/ddd46474-d44a-47ed-8035-d22d97171c0d/artifacts/0/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-2254-g27d09d4.rom
I was expecting it to have the same keyboard issue like years ago, but keyboard worked perfectly fine. Of course at the moment you have to force flash with flashrom because flashrom correctly reports, that the coreboot model number inside the file is not x230t but its x230 (there was a x230t coreboot image on it).

**Is your feature request related to a problem? Please describe.**

There was a problem that seem to be gone on the latest releases.

**Describe the solution you'd like**

Add x230t with default coreboot x230t config and x230t name to heads to finish up the work that started years ago for this model.

**Describe alternatives you've considered**
None?

tlaurion, 2024-08-05:
@fhvyhjriur stays over reboot? Problem was that it was not staying, same reported in some x230 since then as well.

Coreboot devels contacted, even issues opened upstream... Still considered not working and would require ps2 code under coreboot for force proper keyboard, bypassing detection, or having linux kernel module switch.

Both not implemented. Basically you are lucky here and source of issue still not fixed in code AFAIK.

tlaurion, 2024-08-05:
For some traceability.

2254 is https://github.com/linuxboot/heads/commit/27d09d4a2397bc068262fedd17ac3fe7d0fa4b2a so 24.04.02 coreboot based with ps2 init still on. 

If someone can invest time to see if ps2 code was rewritten or replicate this RAW/TRANSLATED AT2 keyboard detection and can confirm that the problem vanished once and for all and pinpoint why and where this was fixed, then @fhvyhjriur x230 and x230t should be exactly the same board from coreboot perspective last time I checked. There would not be any need of creating/maintaining yet another board variant, and past e-ink across everything searchable x230t should answer your questions in this repo.

tlaurion, 2024-08-05:
Assigned to myself because /i have a x230t somewhere, so if I can have behavior stable across reboots/cold boot/reboots, then x230 could be advertised as also working on x230t and this issue could be closed as resolved when actual code fix si traced back to why the problem is fixed now. Maybe TOP_DOWN allocator?! unsure really, this being fixed without reason is just weird as of now.

tlaurion, 2024-08-16:
Weird, I confirm this works on x230t now.

Time to understand why now. Anybody sees what changed under coreboot?

tlaurion, 2024-08-16:
Hmmm will need to do this local, cannot just do files changes from web GUIs
https://github.com/coreboot/coreboot/compare/4.22_branch...24.02_branch

tlaurion, 2024-08-16:
Nothing directly linked to PC80 not PS/2 from what I see....

tlaurion, 2024-08-16:
Well, could be linux kernel version bump as well, good news is that its fixed :/

@fhvyhjriur: what would fix this issue? As of today, x230t was considered not supported by Heads x230 board flavors. Today it would. A note added under board configs?

tlaurion, 2024-08-19:
@fhvyhjriur ping

tlaurion, 2024-09-06:
@fhvyhjriur reping

fhvyhjriur, 2024-09-10:
Some GPIO's was strange(my expectation) on my test but the keyboard was now fine. 

Could you confirm following issues on your hardware?
- The Powerbutton for example does not work as a powerbutton any more. Even the default ATX-Shutdown that have to always work does not work. You can turn the x230t on with the powerbutton, but then even pressing and holding it for a long time does not turn the x230t off.
- Soudcard is not working

Because this is been asked often on coreboot development: Yes, S3 is working fine

tlaurion, 2024-09-10:
> Some GPIO's was strange(my expectation) on my test but the keyboard was now fine.
>
> Could you confirm following issues on your hardware?
>
> * The Powerbutton for example does not work as a powerbutton [...]

That answers the question then. No I haven't tested extensively x230t but keyboard. This means x230t cannot be supported through x230 board flavor and x230t coreboot config should be brought in: hotp and non-hotp variants just like the x230.

PR welcome, @fhvyhjriur. Use helpers of modules/coreboot end file to modify and save config after haviong copied boar configs, renamed x230t coreboot config to a distinct file and make something that works for you. If you create a PR. I will test and merge.

tlaurion, 2024-09-10:
Renamed issue

fhvyhjriur, 2024-09-11:
> PR welcome, @fhvyhjriur. Use helpers of modules/coreboot end file to modify and save config after haviong copied boar configs, renamed x230t coreboot config to a distinct file and make something that works for you. If you create a PR. I will test and merge.

How often should i tell you, that i am not a developer? Stop forcing me this direction!
You told keyboard backspace issue was untested for long time and complained that no one have tested this. I spend time crossflashing and risking a brick for the heads project and reported the test results. THATS ALL! Remember this! I am a human who open thinkpads, flash coreboot and like running fully free software. I am not a developer. Please stop for all time ever again pushing me towards writing code.

tlaurion, 2024-11-15:
Testing under #1791 reported switchig to x230t coreboot config doesn't fix the issue. I tried numerous times, bug needs to be opened under coreboot. Traces evailable to the developer wanting to fix this: its either a ps2 init bug under coreboot key ps2 keyboard enabled (ra vs at2 translated) that shows on cold boot but not reboot. Either coreboot or linux needs to fixate state of ps2 keyboard to AT translated2 to fix this isssue.

Won't fiz myself, closing. X230t won't be officially supported until fixed one way or the other.
Multiple attempts (20 hours+ of my time) wasted on this with late testing and PR bitrotting.

Pr welcome pointig to this issue and I will help to fix/guide but won't do this without sponsoring somehow. Not enough people care enough, won't take leadership here: not my board.

tlaurion, 2026-10-08:
X230t reported broken by user pointing at https://marc.info/?l=linuxbios&m=176719949022072&w=3

x230t didnt boot between 25.03 to 26.03.

tlaurion, 2026-10-08:
closing and reopening to reflect new name of issue and tracking x230t possible errors in testing