linuxboot/heads #1770: KGPE-D16 Configs

jtf7, 2024-08-31:
I have a significant amount more information to merge in here (it’s handwritten and printouts). I will continue to update here in html text first then add to the plain text configs once finished.

@tlaurion  Is this where I am supposed to add this??

----------------DRAFT-----------[Subject to change]

**### _KGPE-D16 Configurations_**
**_Overview:_**
Prior testing of Heads firmware by independent users has been performed on a variety of revisions of KGPE-D16 motherboards and in combination with a wide variety of versions of CPU models, RAM, GPU, etc.. This wide variety of KGPE-D16 configurations utilized by users appears to have caused test results which are wildly inconsistent. The primary purpose of this updated configuration is to create a standardized configuration to help users, testers, and developers maintain uniformity and subsequently as a result, improve efficiency in the development and testing of Heads Firmware.

**_Target End User:_**
The vast majority of ends users are expected to select the KGPE-D16 in order to be able to utilize a system that is as blobfree, open source, and Qubes capable as possible. Therefor, part numbers have been selected taking this in consideration. Other important customer requirements are: robustness, statelessness, stability, raw power, upgrade-ability, and ease-of-use.

----->>>TL:DR: The purpose of these strict configurations while testing are to make everyone's life easier in the long run.

### **_Part Recommendations_**

**_CPUs_**
Opteron 6200 series should be used exclusively while testing. Some CPUs require RAM operating at 1.5v in order to be capable of obtaining the full 1600 MT/s RAM speed. The RAM listed below will help you avoid some of the many nuisances. CPUs contain the memory controller internally, please be cognoscente of this when testing.

**_RAM_**
Please use Micron MT36JSF2G72PZ-1G6 for best compatibility. There are many reasons for this; the biggest of which is the clarity of documentation, ability to run at max speed with the 6200 series CPUs, 1.5v capable, 16gb per stick, ECC RDIMM, cost effectiveness, etc..

_General Part # Guide_
[RAM-Module Part Numbering Systems.pdf](https://github.com/user-attachments/files/16826066/RAM-Module.Part.Numbering.Systems.pdf)

_Micron MT36JSF2G72PZ-1G6 RAM Documentation_
[RAM-MT36JSF2G72PZ-1G6.pdf](https://github.com/user-attachments/files/16826063/RAM-MT36JSF2G72PZ-1G6.pdf)

_Specific Chip Documentation for RAM_
[RAM-MT41J256M16RE-15EITD.pdf](https://github.com/user-attachments/files/16826062/RAM-MT41J256M16RE-15EITD.pdf)

_GPUs_
Please use a GPU from the NVIDIA Kepler micro-architecture. These are the most recent that still allow for open source display firmware.

tlaurion, 2024-09-02:
> > I have a significant amount more information to merge in here (it’s handwritten and printouts). I will continue to update here in html text first then add to the plain text configs once finished.
> > @tlaurion Is this where I am supposed to add this??
> > ----------------DRAFT-----------[Subject to change]
> > ### KGPE-D16 Configurations
> > ----->>>TL:DR: The purpose of these strict configurations while testing are to make everyone's life easier in the long run.
> > **_Overview:_** Prior testing of Heads firmware by independent users has been performed on a variety of revisions of KGPE-D16 motherboards and in combination with a wide variety of versions of CPU models, RAM, GPU, etc.. This wide variety of KGPE-D16 configurations utilized by users appears to have caused test results which are wildly inconsistent. The primary purpose of this updated configuration is to create a standardized configuration to help users, testers, and developers maintain uniformity and subsequently as a result, improve efficiency in the development and testing of Heads Firmware.
> > **_Target End User:_** The vast majority of ends users are expected to select the KGPE-D16 in order to be able to utilize a system that is as blobfree, open source, and Qubes capable as possible. Therefor, part numbers have been selected taking this in consideration. Other important customer requirements are: robustness, statelessness, stability, raw power, upgrade-ability, and ease-of-use.
> > ### **_Recommendations_**
> > **_CPUs_** Opteron 6200 series should be used exclusively while testing. Some CPUs require RAM operating at 1.5v in order to be capable of obtaining the full 1600 MT/s RAM speed. The RAM listed below will help you avoid some of the many nuisances. CPUs contain the memory controller internally, please be cognoscente of this when testing.
> > **_RAM_** Please use Micron MT36JSF2G72PZ-1G6 for best compatibility. There are many reasons for this; the biggest of which is the clarity of documentation, ability to run at max speed with the 6200 series CPUs, 1.5v capable, 16gb per stick, ECC RDIMM, cost effectiveness, etc..
> > _General Part # Guide_ [RAM-Module Part Numbering Systems.pdf](https://github.com/user-attachments/files/16826066/RAM-Module.Part.Numbering.Systems.pdf)
> > _Micron MT36JSF2G72PZ-1G6 RAM Documentation_ [RAM-MT36JSF2G72PZ-1G6.pdf](https://github.com/user-attachments/files/16826063/RAM-MT36JSF2G72PZ-1G6.pdf)
> > _Specific Chip Documentation for RAM_ [RAM-MT41J256M16RE-15EITD.pdf](https://github.com/user-attachments/files/16826062/RAM-MT41J256M16RE-15EITD.pdf)
> > _**GPUs**_ Please use a GPU from the NVIDIA Kepler micro-architecture. These are the most recent that still allow for open source display firmware.
> > **_SPI Flash Chip_** Winbond W25Q128FVAIG (G version)
> > _Hardware Write Protection_ This is a feature that allows a user to lock a eeprom chip as read-only, typically by shorting two of the pins together. A number of users have had issues getting this working so here are some tips.
> > _Hardware write protection is only available on eeprom chips that explicitly support it and typically requires setting specific fuse settings appropriately to enable this function._ This information is found in the manufacturer’s documentation for the specific part # of chip you intend to use.
> > **Winbond W25Q128FVAIG** The Winbond W25Q128FVAIG is a 16MB eeprom chip capable of hardware write protection and is compatible with the KGPE-D16. This chip is physically engraved with the abbreviated text 25Q128FVIG.
> > Fuse settings required to activate the hardware write protection feature for this specific part number of Winbond chip are as listed below:
> > [S18=WPS=0] [S14=CMP=0]
> > [S9=QE=0] [S8=SRP1=0] [S7=SRP0=1]
> > [S6=SEC=0] [S5=TB=0] [S4=BP2=1] [S3=BP1=1] [S2=BP0=1]
> > Experts can tweak these settings to fine tune the chip’s configuration however, the settings listed above have worked well for me and should work well for most users.
> > Connecting pin 3 (/WP) and pin 4 (GND) together with these fuse settings will enable hardware write protection. It may be required to power off the system before connecting/disconnecting these pins in order for the correct write state to be recognized.
> > Many users have a similar version of Winbond chip with part number W25Q128FVAIQ however, it should be noted that this alternate Q version chip is not capable of hardware write protection.
> > **_Additional Fuse Settings: Security Register Lock Bits_** You may also want to set S11, S12, and S13 as listed below in order to permanently lock the security register as read-only. This is where manufacturer’s info, security related information, and hard coded data for the chip is stored. This is a separate location on the chip from where your ROM image is stored. The vibe I got from reading the Winbond documentation is that this could be a good place for backdoor related shenanigans and it’s probably a good idea to set these fuses in order to lock as read-only. That being said, I’m not entirely sure. I set mine as listed below and it worked for me however, as always, use at your own risk.
> > [S13=LB3=1] [S12=LB2=1] [S11=LB1=1] . . .
> > *********More information on the way soon
> 
> you might want to contribute your information on 15h.org

@jtf7 @arhabd : yes, that would be ideal on 15h.org, and then add pages to board configs. 
An issue won't resolve this, could be heads-wiki->osreearch.net but this is not Heads related specifically. 

@jtf7 I'm still unclear reading the above and with what tools you set additional registers from other discussions that landed in this issue to be created? I understand the whys, but not the how :)

jtf7, 2024-09-05:
> you might want to contribute your information on 15h.org

@arhabd Thanks for the link, I will add this site to my list of resources and request an account from them. 

It's nice to see my results coincide with theirs :)

jtf7, 2024-09-05:
> I'm still unclear reading the above and with what tools you set additional registers from other discussions that landed in this issue to be created? I understand the whys, but not the how :)

@tlaurion Some of the information I determined by reading through the lengthy AMD 15h datasheets, I will cite specific documents once I find in my notes. As far as setting Winbond advanced registers, I simply read through the documentation and provided the fuse settings to someone else to help program on my behalf. I know, I know.....from a security point of view, I should really be programming the SPI flash myself but in the essence of saving time, I hired someone on ebay to help with it for the time being. I plan to come back later and learn the details once I finish the "big picture" type things in getting everything I want set up

Which reminds me, I really need to update my current Heads SPI flash chips in order to continue testing. The person I reached out to for help flashing my SPI flash chips is asking if the link below is correct and I have no idea.....is this the correct link they should be flashing from?
 https://app.circleci.com/pipelines/github/linuxboot/heads/832/workflows/303bedb4-409d-410c-bece-501eec300257/jobs/18656

@tlaurion  Is there anything specific I missed that you would like me to elaborate on? While I think of it, please feel free to reach out to me if you ever have questions about manufacturing or mechanical engineering type related questions--that kind of stuff is much more in my comfort zone. (I'm quite curious in the details of how the Power9 systems are being manufactured, especially if I could help make the platform overall more accessible to the end user!)  Happy to help with anything I can! :+1: