linuxboot/heads #1783: Heads fails to find /boot partition with Qubes OS 4.3

marmarek, 2024-09-06:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

1. What board are you using? (Choose from the list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))

This was found on Nitropad T430 (maximized), but I believe it applies to others too.

2. Does your computer have a dGPU or is it iGPU-only?
    - [ ] dGPU (Distinct GPU other then internal GPU)
    - [x] iGPU-only (Internal GPU, normally Intel GPU)

3. Who installed Heads on this computer?
    - [ ] Insurgo (Issues to be reported at https://github.com/linuxboot/heads/issues)
    - [ ] Nitrokey (Issues to be reported at https://github.com/Nitrokey/heads/issues)
    - [ ] Purism (Issues to be reported at https://source.puri.sm/firmware/pureboot/-/issues)
    - [ ] Novacustom (Issues to be reported at https://github.com/Dasharo/dasharo-issues)
    - [ ] HardnenedVault (Issues to be reported at https://github.com/hardenedvault/vaultboot/issues)
    - [ ] Other provider
    - [x] Self-installed

4. What PGP key is being used?
    - [ ] Librem Key (Nitrokey Pro 2 rebranded)
    - [x] Nitrokey Pro
    - [ ] Nitrokey Pro 2
    - [ ] Nitrokey 3 NFC
    - [ ] Nitrokey 3 NFC Mini
    - [ ] Nitrokey Storage
    - [ ] Nitrokey Storage 2
    - [ ] Yubikey
    - [ ] Other

5. Are you using the PGP key to provide HOTP verification?
    - [x] Yes
    - [ ] No
    - [ ] I don't know

### B. Identify how the board was flashed

1. Is this problem related to updating heads or flashing it for the first time?
    - [ ] First-time flash
    - [x] Updating heads 

2. If the problem is related to an update, how did you attempt to apply the update?
    - [ ] Using the Heads menus
    - [ ] Flashrom via the Recovery Shell
    - [ ] External flashing

3. How was Heads initially flashed?
    - [x] External flashing
    - [ ] Internal-only / 1vyprep+1vyrain / skulls
    - [ ] Don't know

4. Was the board flashed with a maximized or non-maximized/legacy rom?
    - [x] Maximized
    - [ ] Non-maximized / legacy
    - [ ] I don't know

5. If Heads was externally flashed, was IFD unlocked?
    - [ ] Yes
    - [ ] No
    - [ ] Don't know

### C. Identify the rom related to this bug report

1. Did you download or build the rom at issue in this bug report?
    - [x] I downloaded it
    - [ ] I built it

2. If you downloaded your rom, where did you get it from?
    - [x] Heads CircleCi
    - [ ] Purism
    - [ ] Nitrokey
    - [ ] Dasharo DTS (Novacustom)
    - [ ] Somewhere else (please identify)

    v0.3.0-3398-gb226453

3. If you built your rom, which repository:branch did you use?
    - [x] Heads:Master
    - [ ] Other (please identify)

4. What version of coreboot did you use in building?
{ You can find this information from github commit ID or once flashed, by giving the complete version from Sytem Information under Options --> menu}

5. In building the rom, where did you get the blobs?
    - [ ] No blobs required
    - [ ] Provided by the company that installed Heads on the device
    - [ ] Extracted from a backup rom taken from this device
    - [ ] Extracted from another backup rom taken from another device (please identify the board model)
    - [ ] Extracted from the online bios using the automated tools provided in Heads
    - [ ] I don't know

## Please describe the problem

**Describe the bug**

Heads cannot find /boot partition which is /dev/sda2 now, it tries /dev/sda1:
(screenshot)

**To Reproduce**
Steps to reproduce the behavior:
1. Install https://openqa.qubes-os.org/tests/111529/asset/iso/Qubes-4.3.202409051715-x86_64.iso
2. Click reboot after first installation phase
3. See heads trying to mount sda1 for the sealed TOTP / rollback counter.

**Expected behavior**

Heads should find the right partition on its own.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

Extra info about partitions layout:
```
[anaconda root@localhost ~]# blkid /dev/sda*
/dev/sda: PTUUID="7a719adb-544a-42d2-94a0-45db91a7c2b1" PTTYPE="gpt"
/dev/sda1: PARTUUID="b2642cdb-2eb1-47c5-a7be-b140bad955e5"
/dev/sda2: UUID="038ce965-a8d6-446d-a39d-4252af10a70a" BLOCK_SIZE="4096" TYPE="ext4" PARTUUID="284ce8df-4659-4cda-90da-1fd09e249cfc"
/dev/sda3: UUID="fecb55e4-fc49-44fd-987a-0a409ccd3750" TYPE="crypto_LUKS" PARTUUID="ad6681e5-713e-4441-8f0b-1d8010c75d9e"
[anaconda root@localhost ~]# fdisk -l /dev/sda
Disk /dev/sda: 119.25 GiB, 128035676160 bytes, 250069680 sectors
Disk model: TOSHIBA THNSNF12
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 7A719ADB-544A-42D2-94A0-45DB91A7C2B1

Device       Start       End   Sectors   Size Type
/dev/sda1     2048      4095      2048     1M BIOS boot
/dev/sda2     4096   2101247   2097152     1G Linux extended boot
/dev/sda3  2101248 250068991 247967744 118.2G Linux filesystem
```

The "BIOS boot" partition is to allow legacy boot while using GPT (instead of MBR) partition table.

tlaurion, 2024-09-06:
The I guess all boot devices under board configs will need to disappear and rely solely on boot partition detection.

@JonathonHall-Purism thoughts?

tlaurion, 2024-09-08:
@JonathonHall-Purism @marmarek : #1784 satisfying?

closed 2024-09-09 by JonathonHall-Purism