linuxboot/heads #1788: Stamp out exFAT warnings at the root

JonathonHall-Purism, 2024-09-09:
Since adding the exFAT driver, we get spurious warnings about exFAT missing superblocks when trying to detect a partition type.  In particular, this happens a lot when trying to find /boot, but it can happen in other cases too.

I made several changes to try to eliminate them in #1602 - tried exFAT after iso9660, tried to exclude bios-grub, LUKS, and LVM partitions, etc.

@tlaurion is making another change in #1784 - exclude very small partitions.

These help, but they still won't be enough.  It's looking like a whack-a-mole game to try to eliminate every cause of these warnings.  `mount` detects filesystem type just by trying all filesystems, exFAT is the oddball here because every other filesystem just fails to mount without a warning to facilitate this behavior.

As we've discussed before, spurious warnings are a real issue because they hide the causes of real problems and reduce confidence that Heads is working correctly.  IMO, we need to eliminate the warning completely, my ideas are either:
* patch the warning out of the kernel
* stop dumping kernel warnings to the console

tlaurion, 2024-09-09:
@JonathonHall-Purism see https://github.com/linuxboot/heads/pull/1784#issuecomment-2338117972 which I think could be the real solution?

JonathonHall-Purism, 2024-09-09:
Referring to blkid checking for bios-grub - it's still not a complete solution.  exFAT will print this warning for any partition that's not exFAT.  It could be any number of other unknown filesystems, or legitimately unformatted, etc.  The only way to really know if we can mount it is to try.

I guess if you really want to go down the rabbit hole - we could implement some other detection for "is this exFAT", remove exFAT from the automatic filesystem list entirely, and then whenever we try to `mount` something, we have to manually check if it's exFAT and then either try exFAT or try everything else.  TBH that sounds like a mess.

I don't love the idea of carrying more kernel patches but we never want this warning :shrug: