linuxboot/heads #1801: Clarify bootguard key fusing and update platforms limitations and new platforms strong dependenciest against ME/CSME

tlaurion, 2024-09-30:
There seems to be a confusion from community around possibility of getting Celvo laptops with unfused bootguard keys and being able to flash custom firmware on Clevo laptops. Clarify that Clevo laptops have unfused bootguard keys only if ordered through Novacustom bulk orders from Clevo.

----

ME/CSME is now required (cannot be disabled) to obtain s01x sleep state. S3 is not possible on newer Intel platforms. So user has to switch ME on to sleep, or accept that no sleep is possible.

tlaurion, 2024-12-05:
Seems like s3 cannot be set nor ME HAP bit enabled to disable ME on v560ty/v540tu per @mkopec comments under https://github.com/linuxboot/heads/pull/1871 

- S3 not preferred: https://github.com/linuxboot/heads/pull/1871#discussion_r1870135128
- HAP not enabled: ME not disabled: https://github.com/linuxboot/heads/pull/1871#discussion_r1871202320

tlaurion, 2024-12-06:
"Clarify bootguard key fusing and update platforms limitations" will be seperate issue once we have to cross that bridge, no point doing prevention/documenting unknowns fro the moment.

macpijan, 2024-12-09:
> Heads never supported ME enabled platform up to now.

Looking at the Intel's approach to S3, it is rather a matter of **when** not **if** this changes, unless this will be a hard blocker of supporting new platforms.

tlaurion, 2024-12-09:
> > Heads never supported ME enabled platform up to now.
> 
> Looking at the Intel's approach to S3, it is rather a matter of **when** not **if** this changes, unless this will be a hard blocker of supporting new platforms.

@macpijan the point is that we are not there yet.
Also, it will then be a documentation problem. If and when s3 is totally deprecated, and ME is needed for s01x (which QubesOS doesn't support yet, btw) then there will need to be a documentation and/or toggle under Heads giving choice to the user between s01x/ME being disabled.

But we are not there yet, aren't we?

tlaurion, 2024-12-09:
Note to everyone lurking into this issue: we are not there yet and v506tu/v540tu will not have ME enabled, and supports s3. Its a question of adapting the configurations before #1821.

mkopec, 2024-12-10:
ME is now disabled and S3 is enabled and functional: https://github.com/linuxboot/heads/pull/1846#issuecomment-2532346570

As I mentioned elsewhere, it's not supported by Intel so if it breaks with some update to silicon or fw there's a good chance they won't fix it