linuxboot/heads #1821: FEATURE FREEZE PLANNED FOR 2024-11-20

tlaurion, 2024-10-24:
This is follow-up conclusion from discussions having happened after my QubesOS mini-summit _Heads rolling release : roles of upstream and downstream forks (Design Session)_ 

- Takeaway is 30 seconds long at end of the recording, up to the end: https://youtu.be/mAb_kHrF6SQ?list=PLuISieMwVBpL5S7kPUHKenoFj_YJ8Y0_d&t=1848
- Priority review needed on issues labeled [Release Cycle](https://github.com/linuxboot/heads/pulls?q=is%3Apr+is%3Aopen+label%3A%22release+cycle%22)

wessel-novacustom, 2024-10-25:
I will try the latest Heads upstream version tomorrow and provide feedback in tickets.

wessel-novacustom, 2024-10-29:
> I will try the latest Heads upstream version tomorrow and provide feedback in tickets.

I'm sorry for the delay. I have to postpone this for now, as we are now working on the videos for the current Dasharo coreboot+Heads release. I hope to find enough time to continue this on Saturday, otherwise the Saturday after.

20th of November sounds a bit early to me since there is probably also some time needed for realising improvements, if needed from our side of course.

tlaurion, 2024-10-29:
> I'm sorry for the delay. I have to postpone this for now, as we are now working on the videos for the current Dasharo coreboot+Heads release. I hope to find enough time to continue this on Saturday, [...]

@wessel-novacustom I didn't know you were spooling this this fast. There are still changes pending that changes the UX, so please sync with me prior of doing the videos. One example is https://github.com/linuxboot/heads/issues/1641#issuecomment-2252658288 which I cross-linked here exactly for video creation purposes. Once that PR is merged, oem-factory-reset/Re-ownership is not expected to change until feature freeze.

tlaurion, 2024-10-29:
> 
> 20th of November sounds a bit early to me since there is probably also some time needed for realising improvements, if needed from our side of course.

This is date Dasharo picked for coreboot release. This is interesting experience. 

@wessel-novacustom @macpijan @pietrushnic : this needs better synchronization. 

Bugs still awaited to be opened on heads (upstream release) to work on issues. Then fixes need to be tested and then pooled to dasharo for downstream release. 

This is why I said Heads is rolling release in my talk. If I have to wait for input from all parties, and videos are to be produced, this is fitever chicken egg and why rolling release. If this is the case, then videos should be spoiled to when Dasharo will pick Heads commit, but issues still awaited on Heads master as of today and pap Ned feature plan is in 20 days as of now. Time to redefine. 

Sync needed off github people. Let's chat under matrix and come here with realistic date with now better understanding of realities of a release requiring a feature freeze without issues even opened yet nor upstream tested today from downstream, nor stated needed development for next downstreamrelease that is expected to happen only once a year.

srgrint, 2024-11-08:
Not sure whether you want testing reports given the recent version bumps in the last week?  In any case, I can confirm that heads-x220-maximized-v0.2.0-2410-g9d656fc works fine.

srgrint, 2024-11-09:
Also works on my intel t430

tlaurion, 2024-11-20:
So end of November is the goal. Will post blockers and change label name accordingly and tag features and Bugfixes that happened between last release and feature freeze for reference on the 20th. 

I agree with your definition of feature and bug. The thing is that sometimes things considered bugs are features, and some features, if untested properly, have bugs. Features not properly implemented also prevents other features, as can be seen in this feature freeze of merged things and issues created/fixed in the goal of feature freezing with feature requests depending on currently/recently fixed broken things/fix pending yet unreleased. 

This is not just heads here. Heads depends on other moving parts, like security dongles, coreboot, ec, tpm implementations, software stacks, kernel, libraires, etc which also has their own new features and bugs.

Only thing that can be done is thorough testing from multiple angles, and maybe joint effort in unit testing which I'm looking for next when I'll have a chance, outside of this feature freeze with help of qemu targets when pflash and firmware update feasible within q35.

A reminder that up to this, Heads was and stillnis a rolling release, where new features are paused until the freeze, even though new requests were made to postpone feature freeze and blockers not fixed yet for feature freeze. See label.

wessel-novacustom, 2024-12-03:
Delay of the feature freeze is unavoidable until [this Nitrokey issue](https://github.com/Nitrokey/nitrokey-hotp-verification/issues/36) has been fixed. We will plan again afterwards. Meanwhile, @tlaurion is working on features he can already work on.

tlaurion, 2024-12-03:
Waiting for missing parts to come my way, still, to be able to set my own feature freeze date.

- V540TU need to provide a common commit not breaking n4x_adl, ns50 with Pr0.
- hotp_verification fixes landing tested independently in PR for quiet mode and automated oem/re-ownership.
-access to dasharo automated testing platform/result/pooling of next runs and attack manual tests not touched by automated testing (unknowns). 

Other parts have been worked on mostly and should not delay things more. Once we have the above, automated testing can start

tlaurion, 2025-02-12:
[Label linked to work leading to feature freeze](https://github.com/linuxboot/heads/issues?q=label%3A%22release+cycle+-+2025-01-30%22+)

[V560tu Dasahro+Heads 0.9.0 was released on 2025-01-30](https://docs.dasharo.com/variants/novacustom_v560tu/releases_heads/#v090-2025-01-30)

Closing

closed 2025-02-12 by tlaurion