linuxboot/heads #1825: t440p/w541 : CONFIG_CBFS_SIZE is 8mb and prevents stitching roms on master

tlaurion, 2024-10-30:
Why is it not `CONFIG_CBFS_SIZE=0xBE4FFF` as all other 12mb opaque flash chips from ivy bridge?

tlaurion, 2024-10-30:
Pinned issue

ResendeGHF, 2024-10-30:
> Why master was not enforcing `CONFIG_CBFS_SIZE=0xBE4FFF` as all other 12mb opaque flash chips from ivy bridge? (was 8mb).

`CONFIG_CBFS_SIZE=0xBE4FFF` leads to longer RAM initialization times. Relevant information about this issue can be found in [#1282](https://github.com/linuxboot/heads/pull/1282/files#r1071650125). As icon noted, this is likely due to the following reasons:

> I assume with the bigger size, not everything in CBFS can be cached. There’s CAR (Cache as RAM) at 0xff7c0000 to 0xff7effff, which is used to store data before the DRAM is initialized. The BIOS region is mapped directly below 4G (at 0x100000000). Therefore, making the CBFS too large could conflict with CAR.  I don't believe this can be changed without also modifying the MRC blob. The best option is to set CBFS_SIZE to 8MiB. One can still make the rest of the BIOS region available using a custom FMAP, though it would not be part of CBFS.

Additionally, I've just tested on the W541 board from [UNTESTED_w541-maximized](https://app.circleci.com/pipelines/github/linuxboot/heads/867/workflows/863c8bed-0ec7-4b0a-8387-5ce2833e2706/jobs/20275) (The path under UNTESTED_w541-hotp-maximized.config is inconsistent due to renaming to UNTESTED). As expected, boot time is over 50 seconds.

tlaurion, 2024-10-30:
@ResendeGHF Maybe its time to switch to native ram init #1711 ? (With its documented downsides)
Point is: past value for CBFS (8mb) is not big enough anymore as can be seen from CircleCI failed builds :(

tlaurion, 2024-11-05:
@ResendeGHF op updated.

tlaurion, 2024-11-06:
Also note that participating in upstream issue would make this go forward https://ticket.coreboot.org/issues/457

tlaurion, 2024-11-07:
Proposed collaboration in s3 resume from suspend patchset at https://review.coreboot.org/c/coreboot/+/81897/comments/d4d3ed3a_d6c46137

If you see replies faster than I do, please ping me here.

tlaurion, 2025-01-24:
To follow the rabbit https://matrix.to/#/!EhaGFZyYcbyhdSgStq:matrix.org/$6YZmhYf-91bvmgab4IDi-DxfS7EbXCUg-Xa4afsLOoM?via=georgi.family&via=matrix.org&via=tchncs.de

tlaurion, 2025-01-24:
Also see https://github.com/linuxboot/heads/issues/1825#issuecomment-2462654620

MattClifton76, 2025-02-23:
Can I get added as a board tester for the T440p, since it always says untested?

tlaurion, 2025-02-24:
It is marked as untested but will move to unmaintained on next coreboot/kernel version bump since nobody knowledgeable enough fixes upstream issue, here NRI or MRC blob.

Not sure how acceptable booting a platform in +50 seconds is acceptable for boards owners here. Would that be satisfying for your use case @MattClifton76 ?

This is what this issue is about. I can add you to board testers next time I touch t440p.

MattClifton76, 2025-02-26:
When is the timer started, when the power button is pushed? My t440p doesn't seem like it takes that long. I mainly use mine to run Qubes.

MattClifton76, 2025-02-28:
I just flashed the latest Heads build 81151d9, upgraded fine and boots into Qubes.

srgrint, 2025-03-08:
Hopefully there should be movement here with Haswell NRI - all the remaining parts of the patch train have now been given +2, so presumably should be getting merged upstream imminently

tlaurion, 2025-03-09:
> Hopefully there should be movement here with Haswell NRI - all the remaining parts of the patch train have now been given +2, so presumably should be getting merged upstream imminently

Patch train got merged.

Since there is effort bumping all boards to 24.12 (including in other coreboot forks, Dasharo and Purism included AFAIK), maybe it would not be too hard to cherry-pick each individual patches and apply on top of 24.12, after 24.12 bump is merged.

tlaurion, 2025-03-20:
@value:we2.ee report on [Heads matrix channel](https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$iiChFxbRzerky9zuL-MYOWCF9V-vsRbz-rugPYyoUSM?via=matrix.org&via=nitro.chat&via=tchncs.de):

> I thought this may be interesting to share here. I noticed my boot times on the T440p were a lot faster than the 50s claimed time for the board causing it to be marked UNTESTED. My heads  install is from a few months ago. I used cbmem to get the boot times which show a 31s boot (as shown below). 
> 
> I know there is a lot of work happening to reduce the boot times with NRI (Thank you!). I don't know if this is relevant to the discussion but I just wanted to point this out as there might be big variances in boot times between t440p devices?
> 
> 
> Heads-v0.2.0-2442-g4f14058
> 
> 27 entries total:
> 
>    0:1st timestamp                                     8,781 (0)
>   11:start of bootblock                                126,122 (117,340)
>   12:end of bootblock                                  126,655 (532)
>   13:starting to load romstage                         126,655 (0)
>   14:finished loading romstage                         172,720 (46,064)
>    1:start of romstage                                 172,919 (199)
>    2:before RAM initialization                         243,084 (70,164)
>    3:after RAM initialization                          19,983,213 (19,740,128)
>    4:end of romstage                                   22,650,727 (2,667,514)
>  100:start of postcar                                  22,658,705 (7,977)
>  101:end of postcar                                    22,658,705 (0)
>    8:starting to load ramstage                         22,658,707 (2)
>   15:starting LZMA decompress (ignore for x86)         22,758,013 (99,305)
>   16:finished LZMA decompress (ignore for x86)         22,825,844 (67,831)
>    9:finished loading ramstage                         22,826,102 (257)
>   10:start of ramstage                                 22,826,143 (40)
>   30:device enumeration                                22,826,148 (5)
>   40:device configuration                              22,829,980 (3,832)
>   50:device enable                                     22,832,286 (2,306)
>   60:device initialization                             22,832,522 (236)
>   15:starting LZMA decompress (ignore for x86)         22,890,336 (57,813)
>   16:finished LZMA decompress (ignore for x86)         22,890,659 (323)
>   70:device setup done                                 23,307,891 (417,231)
>   75:cbmem post                                        23,307,891 (0)
>   80:write tables                                      23,307,891 (0)
>   85:finalize chips                                    23,426,420 (118,528)
>   90:starting to load payload                          23,426,422 (1)
>   99:selfboot jump                                     31,683,175 (8,256,752)
> 
> Total Time: 31,674,380

Important parts above:

> 2:before RAM initialization 243,084 (70,164)
> 3:after RAM initialization 19,983,213 (19,740,128)
-> 19,740,128 spent in ram initialization on normal boot.

gaspar-ilom, 2025-11-13:
We have a fix with NRI ready for testing: https://github.com/linuxboot/heads/pull/2025 Reports are much appreciated.

tlaurion, 2025-11-17:
Note that CONFIG_CBFS_SIZE=0xBE4FFF

Was bogus and went uncorrected up to now. Sorry folks, it should have been CONFIG_CBFS_SIZE=0xBE4000

Impacted all 4+8mb spi chips.
Calculations at https://github.com/linuxboot/heads/pull/2025#issuecomment-3524672338

Tldr: CBFS region should be ifd's bios size.

PR #1826 (2024-10-30): t440p/w541 move to untested - coreboot configs: bump CONFIG_CBFS_SIZE=0x800000 -> 0xBE4FFF 
(not sure why it was 8mb as opposed to all other 12mb opaque flash chips)

Probably fixes #1825. Those boards don't build on master since kernel + cryptsetup version bump, sorry it was not caught in PR with all the rebasing: each PR built seperately, but not when rebased on top of each other.  

Testing needed from boards testers ([BOARDS_TESTERS.md](https://github.com/linuxboot/heads/blob/bbb96fd80e27757ed294095205623d183b2f28d2/BOARD_TESTERS.md)):
- [ ] t440p: @fhvyhjriur @ThePlexus @srgrint @akunterk [... cut]