linuxboot/heads #1827: Implement Automatic OEM Factory Reset with diceware secrets resulting in QR code to be scanned prior of rebooting

tlaurion, 2024-10-30:
This is to discuss requirements, turning https://github.com/linuxboot/heads/discussions/1521 into actionnable plan.

@wessel-novacustom @jan23 @JonathonHall-Purism  (tag anyone from support teams): if you have time to draft what you would like to see as an OEM to start discussing plan and see if I can do this prior of feature freeze (most probably going to be postponed after November 20th 2024 (#1821) 

Thanks!

wessel-novacustom, 2024-10-31:
Currently, we just flash Heads and the customer should do the re-ownership steps.

What we want is a situation in which Heads has already been setup by us and once booted, the end user can verify the integrity and take over the ownership including reencryption of the LUKS header, completely automatically with as less prompts as possible.

We don't do that yet because setting safe credentials and communicating them manually is time-consuming.

So what we need is an automated OEM factory reset with the most secure options that generates credentials automatically and combines this in only one QR code that contains all the necessary information for the end user.

In our production process, we split this content so that it will be partly communicated to the customer by paper with the order, and partly by email. We already do this for the LUKS disk encryption passphrase at the moment.

We see a risk that the end user would forget his actual LUKS passphrase if it isn't necessary to type this during boot. Can we mitigate this risk somehow? I'm thinking about a statement on the _Important notes_ document that we deliver with every laptop already.

We have to keep things as simple as possible, especially for the end user, but also for our production process. That doesn't mean we cannot implement a solution on the product side to deliver credentials through two channels.

@tlaurion I'm looking forward to your proposal how this can work.

NovaCustom will take care of the funds after approval.

tlaurion, 2024-10-31:
Qrcode can be used to pack all the info, as seen under https://github.com/Kunzisoft/KeePassDX/issues/1443#issuecomment-1629420560

wessel-novacustom, 2024-11-01:
> Qrcode can be used to pack all the info, as seen under [Kunzisoft/KeePassDX#1443 (comment)](https://github.com/Kunzisoft/KeePassDX/issues/1443#issuecomment-1629420560)

That's great!

How to communicate these credentials to the client without manual intervention?

The customer needs to store all security components separately, which sounds like a hassle to me. I see a lot of confusion of customers who don't know what password is for what component.

wessel-novacustom, 2024-11-02:
I will explain how I see how this can work. So when pressing 'o' upon OEM factory reset, the employee will be asked to:

- Insert the Nitrokey
- Provide the DRK recovery key¹, prompted by Heads²

Then, a QR code will be generated with these contents sealed in it:

- Generated TPM ownership passphrase
- Generated GPG Admin PIN
- Generated GPG User PIN
- Generated TPM Disk Unlock Key passphrase

So this all sealed in ONE _transition password_ (or passphrase).

When customer boots the laptop for the first time, the laptop should automatically:

- Ask to insert Nitrokey for validation
- Ask to provide the transition password
- Generate the new passphrases without asking (DRK passphrase, generate new TPM ownership passphrase, generate new GPG Admin PIN, generate new GPG User PIN, generate new TPM Disk Unlock Key passphrase)
- Re-encrypt LUKS container without asking
- Show all the generated credentials and making sure that the customer has saved all. No QR code here³.

And that's it, boot process is normal as it is right now after all steps above.

¹ The process should fail if the DRK is incorrect.
² The prompt should allow a USB keyboard. Real situation is that we can use our QR scan device.
³ To avoid confusion with TOTP scan code.

tlaurion, 2024-11-10:
Blocked by https://github.com/Nitrokey/nitrokey-hotp-verification/issues/36

tlaurion, 2024-11-10:
Prerequisite for Heads introspection of firmware (bugfix of cbmem output for TPM1. 2) done under https://github.com/linuxboot/heads/pull/1824

tlaurion, 2024-11-10:
Needs https://github.com/linuxboot/heads/issues/1831

tlaurion, 2024-11-10:
- [ ] Take discussion https://github.com/linuxboot/heads/discussions/1521 and create smaller issues that can be addressed by individual PR

tlaurion, 2025-01-16:
Supersede by https://github.com/linuxboot/heads/pull/1875

tlaurion, 2025-02-03:
Merged as part of #1875