linuxboot/heads #1829: Local Build-System produces difer

Oessel, 2024-11-01:
## Context of the Build

**1. What board are you trying to build?**

**2. What repository:branch are you using to build from?**
- [ ] Heads:Master
- [ ] Other (please specify)

**3. What version of coreboot are you trying to build**

**4. In building the rom where did you get the blobs?**
- [ ] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ ] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
- [ ] Yes
- [ ] No

**6. What operating system are you using**

*please specify*

## Please describe the problem

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

Oessel, 2024-11-01:
[hashes.txt](https://github.com/user-attachments/files/17595608/hashes.txt)

tlaurion, 2024-11-02:
From : https://github.com/linuxboot/heads-wiki/issues/70#issuecomment-1503487734

```
#Download CircleCI hashes.txt and put it in local dir:
HASH1=hashes.txt
#point to local build hashes.txt for same build:
HASH2=~/heads/build/x86/t430-hotp/maximized/hashes.txt
egrep '^[0-9a-f]{64}' $HASHES1 | while read line; do HASH_REF=$(echo $line|awk -F " " {'print $1'}); FILE_REF=$(echo $line|awk -F "/" {'print $NF'}); if ! grep -q "$HASH_REF" $HASHES2; then echo "$FILE_REF doesn't match";fi; done

```
Same hashes.
There is aa problem though with git reporting git error which makes the build system not fill git commit ID etc

```
user@heads-tests-deb12-nix:~/heads$ docker run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) tlaurion/heads-dev-env:v0.2.3 -- make BOARD=t430-hotp-maximized
fatal: detected dubious ownership in repository at '/home/user/heads'
To add an exception for this directory, call:

	git config --global --add safe.directory /home/user/heads
warning: Not a git repository. Use --no-index to compare two paths outside a working tree
usage: git diff --no-index [<options>] <path> <path>

Diff output format options
    -p, --patch           generate patch
    -s, --no-patch        suppress diff output
    -u                    generate patch
    -U, --unified[=<n>]   generate diffs with <n> lines context
    -W, --[no-]function-context
                          generate diffs with <n> lines context
    --raw                 generate the diff in raw format
    --patch-with-raw      synonym for '-p --raw'
    --patch-with-stat     synonym for '-p --stat'
    --numstat             machine friendly --stat
    --shortstat           output only the last line of --stat
    -X, --dirstat[=<param1>,<param2>...]
                          output the distribution of relative amount of changes for each sub-directory
    --cumulative          synonym for --dirstat=cumulative
    --dirstat-by-file[=<param1>,<param2>...]
                          synonym for --dirstat=files,<param1>,<param2>...
    --check               warn if changes introduce conflict markers or whitespace errors
    --summary             condensed summary such as creations, renames and mode changes
    --name-only           show only names of changed files
    --name-status         show only names and status of changed files
    --stat[=<width>[,<name-width>[,<count>]]]
                          generate diffstat
    --stat-width <width>  generate diffstat with a given width
    --stat-name-width <width>
                          generate diffstat with a given name width
    --stat-graph-width <width>
                          generate diffstat with a given graph width
    --stat-count <count>  generate diffstat with limited lines
    --[no-]compact-summary
                          generate compact summary in diffstat
    --binary              output a binary diff that can be applied
    --[no-]full-index     show full pre- and post-image object names on the "index" lines
    --[no-]color[=<when>] show colored diff
    --ws-error-highlight <kind>
                          highlight whitespace errors in the 'context', 'old' or 'new' lines in the diff
    -z                    do not munge pathnames and use NULs as output field terminators in --raw or --numstat
    --[no-]abbrev[=<n>]   use <n> digits to display object names
    --src-prefix <prefix> show the given source prefix instead of "a/"
    --dst-prefix <prefix> show the given destination prefix instead of "b/"
    --line-prefix <prefix>
                          prepend an additional prefix to every line of output
    --no-prefix           do not show any source or destination prefix
    --default-prefix      use default prefixes a/ and b/
    --inter-hunk-context <n>
                          show context between diff hunks up to the specified number of lines
    --output-indicator-new <char>
                          specify the character to indicate a new line instead of '+'
[... 80 lines cut ...]
To add an exception for this directory, call:

	git config --global --add safe.directory /home/user/heads
----------------------------------------------------------------------
!!!!!! BUILD SYSTEM INFO !!!!!!
System CPUS: 12
System Available Memory: 15146 GB
System Load Average: 0.98
----------------------------------------------------------------------
Used **CPUS**: 12
Used **LOADAVG**: 18
Used **AVAILABLE_MEM_GB**: 15146 GB
----------------------------------------------------------------------
**MAKE_JOBS**: -j12 --load-average=18 

Variables available for override (use 'make VAR_NAME=value'):
**CPUS** (default: number of processors, e.g., 'make CPUS=4')
**LOADAVG** (default: 1.5 times CPUS, e.g., 'make LOADAVG=54')
**AVAILABLE_MEM_GB** (default: memory available on the system in GB, e.g., 'make AVAILABLE_MEM_GB=4')
**MEM_PER_JOB_GB** (default: 1GB per job, e.g., 'make MEM_PER_JOB_GB=2')
----------------------------------------------------------------------
!!!!!! Build starts !!!!!!
```

Problem vanishes if `sudo chown root:root ~/heads` but that is not a real solution. Investigating.

tlaurion, 2024-11-03:
Problem was safedir in doker image (flake.nix) not creating gitconfig under /root dir. Testing fix under #1833

tlaurion, 2024-11-03:
Reasoning of discrepancies for prosperity: git commands under Makefile were not able to get git commit/clean state, resulting in differences in produced binaries because version exported and not reused under kernel modules (modules.cpio) and tools/libraries packed (tools.cpio) packed in initrd.cpio.xz, resulting in payload (kernel+initramfs) packed as payload, resulting in coreboot image being different locally/ci (where CI runs docker in privileged mode and check for git was bypassed per how CircleCI runs docker images resulting in different behavior). Details:
- `GIT_HASH	:= $(shell git rev-parse HEAD)` call failed under local docker image usage, resulting in no versioning info reused under Heads buildsystem using Heads hash as version. GIT_HASH exported and reused was null.
- `HEADS_GIT_VERSION	:= $(shell git describe --abbrev=7 --tags --dirty)` call failed under local docker image usage, resulting in no versioning user Heads buildsystem using Heads as version. This variable wass used in rom filename and was also null.

tlaurion, 2024-11-03:
@Oessel #1833 fixes the root cause of the issue, but not README.md which seems to need improvement so that users wanting to do local builds that matches hashes of CircleCI can do so arriving to same results.

Can you repro and also provide PR/comment here what is missing under README.md so that it's clearer that CircleCI used versioned docker image is needed to produce reproducible builds that match from CircleCI and local builds? I seem to have failed writing docs that permit others, like you to arrive to same results.

Thanks.

tlaurion, 2024-11-03:
NEeds README.md fix, reopeing util other PR fixes README.md with end user reported needed fixes

Oessel, 2024-11-03:
I think i accidently caused some trouble here. I applied changes to some files in my local branch, so i usually don't check all hashes, but when i noticed the missing commit-id, i tried to figure out if this is caused by my changes. After some testing, i tried to build with a fresh build-system and fresh cloned source, but i simply forgot to run the right command and then checked the hashes. I caused the wrong hashes with my incorrect testing, so i am really sorry for this! I mixed up this two different problems and opened the issue, you're docs are correctly saying what is needed to build reproductible Roms. I would like to change the title of this issue and close it as completed if you are OK with this.

tlaurion, 2024-11-03:
@Oessel please replicate with master :)
I pushed a fix that was needed following this issue and pushed docker v0.2.4 used by Circleci.

Oessel, 2024-11-03:
Now i got proper named output files and the same hashes as CircleCI, but both hashes.txt contain the hash oft the bottom.rom two times and no hash of the combined rom.

Oessel, 2024-11-03:
Not related to this issue, but another thing came to my mind. Of course this is a very specific scenario, but if a high skilled attacker would be able to manipulate my local source files, wouldn't it be possible that this attacker is able to replace the hashes.txt in my build-directory with a downloaded file from CircleCI? Does it make sense to compute the hashes of you local built roms on your local system and check them against CircleCI too? But as i said, not related to this issue at all and very paranoid ;)

tlaurion, 2024-11-04:
> Now i got proper named output files and the same hashes as CircleCI, but both hashes.txt contain the hash of the bottom.rom twice and no hash of the combined rom.

This is known issue #1534 fixed by https://github.com/linuxboot/heads/pull/1834

tlaurion, 2024-11-04:
> Not related to this issue, but another thing came to my mind. Of course this is a very specific scenario, but if a high skilled attacker would be able to manipulate my local source files, wouldn't it be possible that this attacker is able to replace the hashes.txt in my build-directory with a downloaded file from CircleCI? Does it make sense to compute the hashes of you local built roms on your own system (e.g. in a disposable VM on qubes) and check them against CircleCI too? But as i said, not related to this issue at all and very paranoid ;)

This is why reproducible builds exists, hashes.txt are produced and final ROM hash is the only thing that matters at the end.

PR #1833 (2024-11-03): Docker image 0.2.4: Fix git dubious ownership in repository
fix #1829 

.gitconfig containing safe dir was not under /root so not taken into consideration.
closed 2024-11-04 by Oessel