linuxboot/heads #1835: Port Heads to Novacustom V560TU

tlaurion, 2024-11-04:


tlaurion, 2024-11-13:
Test: https://docs.dasharo.com/variants/novacustom_v560tu/releases/#v090-2024-07-18
- Dasharo commit to first try https://github.com/Dasharo/coreboot/tree/316f964c

tlaurion, 2024-11-14:
@macpijan this commit reference config points to bad FSP paths, no recipe to build EC.

Please provide a commit that builds with Heads for that platform.

pietrushnic, 2024-11-14:
https://github.com/Dasharo/dasharo-issues/issues/1139

tlaurion, 2024-11-15:
> This is not the correct repo for Dasharo bugs; let me create the correct reference based on what I get from you in DM.

Commented at https://github.com/Dasharo/dasharo-issues/issues/1139#issuecomment-2477521692

Note: meteorlake fb init for #1846 was reporting having issue (GOP+coreboot fb init not working) under Heads channel around https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$Uu8mvr1LvKZQeIeMsmlcrrDW_HqyIbOTs1fvVeLFWaM?via=matrix.org&via=nitro.chat&via=envs.net and following threads

Provided hints, reviewed librem_11 GOP enabled board (used to port nv41) related PR for references:
- https://github.com/linuxboot/heads/pull/1522 
- https://github.com/linuxboot/heads/pull/1492
- https://github.com/linuxboot/heads/commit/a15f77e3361d5baeb7500e75d749b92f0d1e9e0e : when some tunings were discovered needed because libgfxinit  does not yet support Jasper Lake.
- Initial PR switching i915 to libgfxinit's now efifb compliant FB prepared by coreboot: https://github.com/linuxboot/heads/pull/1403

mkopec, 2024-11-28:
>  Need an upstream valid commit that can build Heads for testing. 

@tlaurion here's the commit I've been using for testing: https://github.com/Dasharo/coreboot/commit/db1d9cd59d0d7d5b708bbdf8670780914061410c from this branch: https://github.com/Dasharo/coreboot/commits/mtl_heads

It's just two changes that haven't been merged upstream yet, a fix for the FSP error and a commit enabling LPSS UART console for debugging.

tlaurion, 2024-11-28:
> @tlaurion here's the commit I've been using for testing: https://github.com/Dasharo/coreboot/commit/db1d9cd59d0d7d5b708bbdf8670780914061410c from this branch: [...]

Will ping you to cherry-pick commit of flashrom once merge upstream and then will reuse your fork with merged commits missing. Seems like a plan @mkopec ?

tlaurion, 2024-11-28:
> > > Need an upstream valid commit that can build Heads for testing.
> > 
> > 
> > @tlaurion here's the commit I've been using for testing: [Dasharo/coreboot@db1d9cd](https://github.com/Dasharo/coreboot/commit/db1d9cd59d0d7d5b708bbdf8670780914061410c) from this branch: https://github.com/Dasharo/coreboot/commits/mtl_heads
> > It's just two changes that haven't been merged upstream yet, a fix for the FSP error and a commit enabling LPSS UART console for debugging.
> 
> Will ping you to cherry-pick commit of flashrom once merge upstream and then will reuse your fork with merged commits missing. Seems like a plan @mkopec ?

https://github.com/linuxboot/heads/pull/1846#issuecomment-2506734728 @mkopec 

Ping me here directly with @tlaurion when coreboot commit is good including needed fixes and basic tests done for v560TU (build, EC build recipes) so I can help like it happened #1846 which now builds and boot Heads.

tlaurion, 2025-01-16:
https://github.com/linuxboot/heads/pull/1846#issuecomment-2593197284 builds and doesn't take 30 seconds on bootsplash to enter heads payload.

#1846 merged, closing

closed 2025-01-16 by tlaurion