linuxboot/heads #1836: Port Heads to Novacustom V540TU

tlaurion, 2024-11-04:


tlaurion, 2024-11-05:
Trace of failed attempts branches (with associated CircleCI failed build traces)

- Dasharo v0.9 branch from release attempt: https://github.com/tlaurion/heads/tree/V540TU_PoC
  - Doesn't build 
- Dasharo novacustom_branch attempt: https://github.com/tlaurion/heads/tree/V540TU_PoC-dasharo_novacustom_upstream_branch
  - Builds but bricks 
- Dasharo upstream (latest, yesterday commit) branch attempt : https://github.com/tlaurion/heads/tree/V540TU_PoC-dasharo_upstream_branch
  - Doesn't build with coreboot-blobs not being in sync?
    - 3rdparty/blobs/mainboard/clevo/mtl-h/FSP/MeteorLake/Include/  doesn't exist?

mkopec, 2024-11-29:
@tlaurion I've rebased https://github.com/linuxboot/heads/pull/1846 to include flashprog support for meteorlake. Will clean up the git history today and then add V56 variant too

closed 2025-01-16 by tlaurion