linuxboot/heads #1842: Investigate emulation possibilities (kvm/qemu) of Nitrokey usb security dongles

tlaurion, 2024-11-08:
This guide introduces the USB/IP runner and shows how it can be used to simulate a Nitrokey 3 device.
https://github.com/Nitrokey/nitrokey-3-firmware/blob/main/docs/usbip.md

That would resolve most of my qemu (tcg, not kvm) problems in HOTP testing, and most of automated testings needs (outside of #1203)

tlaurion, 2024-11-08:
Alternative that could implement reverse HOTP sealing support:
- yubikey (refused but don't find a trace)
- canokey needing use case justification at https://github.com/canokeys/canokey-qemu/issues/7

mkopec, 2024-11-21:
I got the usbip NK3 runner working according to the [doc](https://github.com/Nitrokey/nitrokey-3-firmware/blob/main/docs/usbip.md) , the key appears in lsusb but it looks like the usbip runner is built without the secrets app (so no HOTP support). I'm still looking into how difficult it would be to enable

macpijan, 2024-11-21:
@mkopec So this would need to be added to the NK3 emulator? Why don't we ask @daringer about that? We may also create issue in NK repo with such a question, and cross-reference here. If that can be done, I am sure NitroKey folks would know better than us.

mkopec, 2024-11-21:
Created https://github.com/Nitrokey/nitrokey-3-firmware/issues/564

tlaurion, 2024-11-23:
@mkopec details requested under https://github.com/Nitrokey/nitrokey-3-firmware/issues/564#issuecomment-2493738891

tlaurion, 2024-11-23:
> @mkopec details requested under https://github.com/Nitrokey/nitrokey-3-firmware/issues/564#issuecomment-2493738891

@mkopec asked specifics for Heads use case and docs improvements needed for qemu under following comment