linuxboot/heads #1864: `generic-init` drops to recovery shell immediately upon entering

mkopec, 2024-11-28:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

1. What board are you using? (Choose from the list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))

qemu-coreboot-fbwhiptail-tpm1-hotp

3. Does your computer have a dGPU or is it iGPU-only?
    - [ ] dGPU (Distinct GPU other then internal GPU)
    - [ ] iGPU-only (Internal GPU, normally Intel GPU)
    - [x] neither? :)

4. Who installed Heads on this computer?
    - [ ] Insurgo (Issues to be reported at https://github.com/linuxboot/heads/issues)
    - [ ] Nitrokey (Issues to be reported at https://github.com/Nitrokey/heads/issues)
    - [ ] Purism (Issues to be reported at https://source.puri.sm/firmware/pureboot/-/issues)
    - [ ] Novacustom (Issues to be reported at https://github.com/Dasharo/dasharo-issues)
    - [ ] HardnenedVault (Issues to be reported at https://github.com/hardenedvault/vaultboot/issues)
    - [ ] Other provider
    - [x] Self-installed

5. What PGP key is being used?
    - [ ] Librem Key (Nitrokey Pro 2 rebranded)
    - [ ] Nitrokey Pro
    - [ ] Nitrokey Pro 2
    - [ ] Nitrokey 3 NFC
    - [ ] Nitrokey 3 NFC Mini
    - [ ] Nitrokey Storage
    - [ ] Nitrokey Storage 2
    - [ ] Yubikey
    - [x] Other

6. Are you using the PGP key to provide HOTP verification?
    - [ ] Yes
    - [x] No
    - [ ] I don't know

### B. Identify how the board was flashed

`make BOARD=qemu-coreboot-fbwhiptail-tpm1-hotp run`

1. Is this problem related to updating heads or flashing it for the first time?
    - [x] First-time flash
    - [ ] Updating heads 

2. If the problem is related to an update, how did you attempt to apply the update?
    - [ ] Using the Heads menus
    - [ ] Flashrom via the Recovery Shell
    - [ ] External flashing

3. How was Heads initially flashed?
    - [ ] External flashing
    - [ ] Internal-only / 1vyprep+1vyrain / skulls
    - [ ] Don't know

4. Was the board flashed with a maximized or non-maximized/legacy rom?
    - [ ] Maximized
    - [ ] Non-maximized / legacy
    - [ ] I don't know

5. If Heads was externally flashed, was IFD unlocked?
    - [ ] Yes
    - [ ] No
    - [ ] Don't know

### C. Identify the rom related to this bug report

1. Did you download or build the rom at issue in this bug report?
    - [ ] I downloaded it
    - [x ] I built it

2. If you downloaded your rom, where did you get it from?
    - [ ] Heads CircleCi
    - [ ] Purism
    - [ ] Nitrokey
    - [ ] Dasharo DTS (Novacustom)
    - [ ] Somewhere else (please identify)

    *Please provide the release number or otherwise identify the rom downloaded*

3. If you built your rom, which repository:branch did you use?
    - [ ] Heads:Master
    - [x] Other (please identify) - https://github.com/tlaurion/heads/tree/dasharo-add_novacustom_v540tu

4. What version of coreboot did you use in building?
{ You can find this information from github commit ID or once flashed, by giving the complete version from Sytem Information under Options --> menu}

5. In building the rom, where did you get the blobs?
    - [ ] No blobs required
    - [ ] Provided by the company that installed Heads on the device
    - [ ] Extracted from a backup rom taken from this device
    - [ ] Extracted from another backup rom taken from another device (please identify the board model)
    - [ ] Extracted from the online bios using the automated tools provided in Heads
    - [ ] I don't know

## Please describe the problem

**Describe the bug**
A clear and concise description of what the bug is.

generic-init always exits to recovery shell, preventing usage without a graphical display 

**To Reproduce**
Steps to reproduce the behavior:
1. Build and run:
   ```
   docker run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) tlaurion/heads-dev-env:latest -- make BOARD=qemu-coreboot-fbwhiptail-tpm1-hotp
   make BOARD=qemu-coreboot-fbwhiptail-tpm1-hotp run
   ```
2. Press Enter in serial console to drop to recovery shell
3. Run `generic-init`

**Expected behavior**
A clear and concise description of what you expected to happen.

Can use menu to configure heads

**Screenshots**
If applicable, add screenshots to help explain your problem.

Log

```
~ # generic-init
y) Default boot
n) TOTP does not match
r) Recovery boot
u) USB boot
m) Boot menu
[   27.561287] TRACE: /etc/functions(148): confirm_totp
[   27.567165] TRACE: /bin/unseal-totp(8): main
[   27.589087] TRACE: /bin/tpmr(612): tpm1_unseal
[   27.612625]  !!! ERROR: Unable to read sealed file from TPM NVRAM !!!
[   29.625931] DEBUG: Running at_exit handlers
[   29.627757] TRACE: /bin/tpmr(388): cleanup_shred
[   29.629366]  !!! ERROR: Unable to unseal TOTP secret !!!
[   31.635234] TRACE: Under /etc/ash_functions:recovery
!!!!! TOTP code generation failed
[   31.644298] DEBUG: Board qemu-coreboot-fbwhiptail-tpm1-hotp - version Heads-v0.2.0-2451-ga141ebe-dirty
TPM: Extending PCR[4] to prevent any further secret unsealing
[   31.671296] DEBUG: TPM: Will extend PCR[4] with hash of filename recovery
[   31.674936] TRACE: /bin/tpmr(788): main
[   31.700719] DEBUG: TPM: Extending PCR[4] with hash 8b60e9d739b5a5bfd87cbfe67501ab4fa6e41504
[   31.703876] DEBUG: exec tpm extend -ix 4 -ic recovery
New value of PCR[4]: 83250e39d4e4af122e5f21f1e0f45afd25f3345b
!!!!! Starting recovery shell
~ #
```

**Additional context**
Add any other context about the problem here.

closed 2024-11-28 by mkopec

mkopec, 2024-11-28:
closing as a misunderstanding of what whiptail and fbwhiptail are. Will use whiptail for tests in QEMU.

tlaurion, 2024-11-28:
> 
> Thierry also clarified that gui-init should be used on all boards, and generic-init is a legacy implementation.

generic-init is the 'generic' implementation by a lack of a better name. At the time of writing those lines, kgpe-d16 server board variant still points to it, driven by bmc console. 

I guess this is duplicate somehow of https://github.com/linuxboot/heads/issues/477