linuxboot/heads #1879: Fail building t430-hotp-maximized

BreakfastCharm, 2024-12-17:
## Context of the Build

**1. What board are you trying to build?**
t430-hotp-maximized

**2. What repository:branch are you using to build from?**
- [ x] Heads:Master
- [ ] Other (please specify)

**3. What version of coreboot are you trying to build**

I dont know, git clone https://github.com/linuxboot/heads

**4. In building the rom where did you get the blobs?**
- [ ] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ ] Extracted from the online bios using the automated tools provided in Heads
- [ x] I don't know

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
- [ ] Yes
- [ ] No

**6. What operating system are you using**

Ubuntu 24.04.1 LTS

## Please describe the problem

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. git clone https://github.com/linuxboot/heads
2. cd heads
3. ./docker_repro.sh make BOARD=t430-hotp-maximized
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

```
./docker_repro.sh make BOARD=t430-hotp-maximized
Using CircleCI Docker image: tlaurion/heads-dev-env:v0.2.4
[sudo] password for BreakfastCharm: 
----
Usage reminder: The minimal command is 'make BOARD=XYZ', where additional options, including 'V=1' or 'CPUS=N' are optional.
For more advanced QEMU testing options, refer to targets/qemu.md and boards/qemu-*/*.config.

Type exit within docker image to get back to host if launched interactively!
----
2024-12-17 19:49:23+00:00 MAKE coreboot-24.02.01
tail /home/BreakfastCharm/heads/build/x86/log/coreboot-24.02.01.log
-----
make -C "/home/BreakfastCharm/heads/build/x86/coreboot-24.02.01/t430-hotp-maximized" -C /home/BreakfastCharm/heads/build/x86/coreboot-24.02.01 obj=/home/BreakfastCharm/heads/build/x86/coreboot-24.02.01/t430-hotp-maximized DOTCONFIG=/home/BreakfastCharm/heads/build/x86/coreboot-24.02.01/t430-hotp-maximized/.config XGCCPATH=/home/BreakfastCharm/heads/build/x86/coreboot-24.02.01/util/crossgcc/xgcc/bin/  BUILD_TIMELESS=1 -j2 --load-average=3 
make[1]: Entering directory '/home/BreakfastCharm/heads/build/x86/coreboot-24.02.01'
toolchain.mk:183: The coreboot toolchain for 'x86_32' architecture was not found.
toolchain.mk:193: The coreboot toolchain version of iasl 'R06_28_23' was not found
toolchain.mk:216: 
toolchain.mk:217: Path to your toolchain is currently set to '/home/BreakfastCharm/heads/build/x86/coreboot-24.02.01/util/crossgcc/xgcc/bin/'
toolchain.mk:219: 
toolchain.mk:220: To build the entire coreboot toolchain: run 'make crossgcc'
toolchain.mk:222: To build just IASL: run 'make iasl'
toolchain.mk:224: For more toolchain build targets: run 'make help_toolchain'
toolchain.mk:225: 
toolchain.mk:227: To try to use any toolchain in your path, run 'make menuconfig', then select
toolchain.mk:229: the config option: 'General setup', and 'Allow building with any toolchain'
toolchain.mk:231: Note that this is NOT supported. Using it means you're on your own.
toolchain.mk:233: 
toolchain.mk:235: *** Halting the build.  Stop.
make[1]: Leaving directory '/home/BreakfastCharm/heads/build/x86/coreboot-24.02.01'
make: *** [Makefile:570: /home/BreakfastCharm/heads/build/x86/coreboot-24.02.01/t430-hotp-maximized/.build] Error 1
```

tlaurion, 2024-12-17:
@BreakfastCharm I cannot reproduce this issue.

Do

Let's go to basics, and remove heads git cloned directory and start fresh.
 
You most probably have build artifacts from your prior attempt without using the docker image?

This is exactly what is done by CircleCI to build roms downloadable by https://osresearch.net instructions.

See https://app.circleci.com/pipelines/github/linuxboot/heads/888/workflows/f06c63b2-ea63-4995-ba13-fdf1acb0b096/jobs/21246/parallel-runs/0/steps/0-102

What it basically does is to to call hidden by ./docker_repro.sh:
`docker run --device=/dev/bus/usb:/dev/bus/usb -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) $DOCKER_IMAGE`

which translates today by:
`docker run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) tlaurion/heads-dev-env:v0.2.4 -- make V=1 BOARD=t430-hotp-maximized   | ts`

As per README.md, you could as well extrapolate instructions and do
- `docker run -e DISPLAY=$DISPLAY --network host --rm -ti -v $(pwd):$(pwd) -w $(pwd) tlaurion/heads-dev-env:v0.2.4` to jump inside of v.0.2.4 pulled docker image and then
- `make V=1 BOARD=t430-hotp-maximized` when V=1 is verbose mode for the build. And do make V=1 BOARD=t430-hotp-maximized CPUS=1 to build in single thread mode so that logs make sense to understand.

But in your logs above, coreboot crossgcc buildstack was not built, and troubleshooting this makes no sense if you followed the reproducible instructions at the first place.

Feel free to tag @tlaurion if you need additional help or close this issue once your reproduced the documentation starting from a fresh clone.

tlaurion, 2024-12-17:
> To Reproduce
> Steps to reproduce the behavior:
> 
>     git clone https://github.com/linuxboot/heads
>     cd heads
>     ./docker_repro.sh make BOARD=t430-hotp-maximized
>     See error
> 

TLDR:
- rm -rf ~/heads
- git clone https://github.com/linuxboot/heads
- cd heads
- ./docker_repro.sh make BOARD=t430-hotp-maximized

closed 2024-12-18 by BreakfastCharm