linuxboot/heads #1885: Heads support for T480/T480s

doritos4mlady, 2025-01-14:
Looks like a great candidate as it recently got libreboot support and is much faster than the T420/T430. 

Is this something the devs would be interested in supporting?

notgivenby, 2025-01-21:
The platform may be indeed attractive. I am not sure about being much faster.  i7 3740qm/3840qm which can be installed on t420/t430 performs similar to  i7 8550u. The newer CPU are less hungry though (45W vs 15 W). Both are EOL in terms of microcode updates. One issue for heads could be the TPM problem.  
https://libreboot.org/docs/install/t480.html "The TPM is disabled on this device, to prevent hanging/boot delay in SeaBIOS, due to buggy TPM drivers there." Is this true only for SeaBIOS or for GRUB too?

tlaurion, 2025-01-28:
Some discussion over [Heads matrix channel thread on porting process for t480](https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$ODBCVcCf317FpchLRYMKhneFTDmCf1Ahz7y0xDZzF6U?via=matrix.org&via=nitro.chat&via=envs.net) if thread discussion is better for you to organize leadership of this port.

tlaurion, 2025-01-29:
TPM works under coreboot but is "vulnerable to gpio attack" [source](https://matrix.to/#/!EhaGFZyYcbyhdSgStq:matrix.org/$fl912PPDSzPMkTlRJu-txuykoe4uDi1ppdKml5c9Z-8?via=matrix.org&via=tchncs.de&via=tedomum.net)

tlaurion, 2025-01-29:
@notgivenby 

>"The TPM is disabled on this device, to prevent hanging/boot delay in SeaBIOS, due to buggy TPM drivers there." Is this true only for SeaBIOS or for GRUB too?

Would not be relevant to Heads in any case.

Any board owner here willing to take leadership into starting and leading the port?

akunterkontrolle, 2025-01-29:
Just want to chime in that I am the happy new owner of a T480 which I explicitly bought for playing around with coreboot and the hope to ultimately run heads. So I am more than willing to help with the port and test, but I don't think I am qualified to take the leadership in any way.

Maybe to add my perspective on why the T480(s) is an interesting platform: Yes, it is indeed not more powerful than a e.g. a maxed out t440p (which is sad) but uses way less power (good thing on a Laptop) and has all the bells and whistles of USB-C (which is helpful e.g. at my workplace where we have set up monitors to act as a USB-C dock). And while not faster, it is much quieter then my t440p under load.

tlaurion, 2025-01-30:
> TPM works under coreboot but is "vulnerable to gpio reset attack" [source](https://matrix.to/#/!EhaGFZyYcbyhdSgStq:matrix.org/$fl912PPDSzPMkTlRJu-txuykoe4uDi1ppdKml5c9Z-8?via=matrix.org&via=tchncs.de&via=tedomum.net)

https://ticket.coreboot.org/issues/576

notgivenby, 2025-01-31:
@doritos4mlady @akunterkontrolle I am trying to evaluate if it is really a good investment in a hobby project. Would you be interested in a community effort of porting heads to t480? Do you know anybody else who would be interested and would/can contribute to the effort?

Board advantages: 
The IPS panel is good (in comparison to t430, x230). You can put up to 64gb RAM, which is good for example for Qubes OS.  It is not that expensive (2x32gb sodimm ddr4 3200 costs around 100$). It is much lighter than t430 too. The SPI chip is very easy to access. One need to remove only the back panel which has only 4 screws or so. The used board itself depending on the condition is also around 250-300$. I even found couple of refurbished boards on Amazon/ebay. In European countries, it means you have a 1 year warranty and can return it for free in 14 days. So the chances are high that will have couple of users interested. This means the port may be of use for the whole community.

Some questions/obstacles :
1. Is your spi chip free? Mine has this https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$BHJDXHAaVjaQyrYu6UVtDyC425Da_y5AuGUzFZP0a2E?via=matrix.org&via=nitro.chat&via=envs.net 
[image]
2. Are you guys able to see the cpu temperature sensors and do fans work after flashing libreboot? There was one issue perhaps a single one https://forum.qubes-os.org/t/t480-problems-after-librebooting/31427/12
3. TPM GPIO vulnarability better to be fixed otherwise the board is not reasonably safe under heads https://ticket.coreboot.org/issues/576 https://mkukri.xyz/2024/06/01/tpm-gpio-fail.html. Since it maybe relevant to other boards it is likely that it will be fixed. 
4. @tlaurion in case we screw with the porting by our self would you mind to share in which range of consulting fee we may motivate you to do that? We would have the boards bought and if 10 people would donate like 100$ each which is kind of sum you expect for transition from dasharo to dasharo/heads would it be enouph? I have no ideas how expensive it can be, so if it is too low please do not take it personally. Perhaps community effort would be also to find some money for that.

mkopec, 2025-01-31:
> Is your spi chip free? Mine has this https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$BHJDXHAaVjaQyrYu6UVtDyC425Da_y5AuGUzFZP0a2E?via=matrix.org&via=nitro.chat&via=envs.net

That's not the BIOS SPI flash, the BIOS flash sits close to the middle of the board and is 16MB

notgivenby, 2025-01-31:
> > Is your spi chip free? Mine has this https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$BHJDXHAaVjaQyrYu6UVtDyC425Da_y5AuGUzFZP0a2E?via=matrix.org&via=nitro.chat&via=envs.net
> 
> That's not the BIOS SPI flash, the BIOS flash sits close to the middle of the board and is 16MB

It is Thunderbolt firmware’s SPI flash, corrected...

akunterkontrolle, 2025-01-31:
> [@doritos4mlady](https://github.com/doritos4mlady) [@akunterkontrolle](https://github.com/akunterkontrolle) I am trying to evaluate if it is really a good investment in a hobby project. Would you be [...]

I am certainly interested in a community effort to port heads to the t480 (note: not the t480s, I got the t480 and it is highly unlikely that I will get a t480s as well). I do not know anybody else who could be interested though.

> Board advantages: The IPS panel is good (in comparison to t430, x230). You can put up to 64gb RAM, which is good for example for Qubes OS. It is not that expensive (2x32gb sodimm ddr4 3200 costs [...]

Fully agree. I just flashed Libreboot a few minutes ago for the first time and the hardware-based flashing was so much less painful compared to the t440p. I think I even managed this time to not break any plastic clips, yeah!

> 
> Some questions/obstacles :
> 
>     1. Is your Thunderbolt firmware’s spi chip free? Mine has this https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$BHJDXHAaVjaQyrYu6UVtDyC425Da_y5AuGUzFZP0a2E?via=matrix.org&via=nitro.chat&via=envs.net
>        [image]

Well should have read that first, I had it open just minutes ago but did not look closely at the Thunderbolt chip. However, I do not remember having seen any obstacles around it.  Although, I am not sure if it matters, because you can easily upgrade the Thunderbolt firmware via software with fwupdmgr from within Linux (at least if your fwupd version is recent enough, Debian/Devuan stable obviously lagging behind). I am not sure why Libreboot recommends flashing that directly with a hardware programmer as well. Maybe they are not aware of the possibility of updating from within Linux.

> 
>     2. Are you guys able to see the cpu temperature sensors and do fans work after flashing libreboot? There was one issue perhaps a single one https://forum.qubes-os.org/t/t480-problems-after-librebooting/31427/12

Fans work and spin up automatically. I can see the cpu temp sensors but no fan speed and the thinkpad_acpi module does not load. I have to check if the force option will do anything.

> 
>     3. TPM GPIO vulnarability better to be fixed otherwise the board is not reasonably safe under heads https://ticket.coreboot.org/issues/576 https://mkukri.xyz/2024/06/01/tpm-gpio-fail.html. Since it maybe relevant to other boards it is likely that it will be fixed.

Yep. Sadly, this is something I can probably not help with at all. I don't thin I even understand fully how that attack works in the first place.

> 4. [@tlaurion](https://github.com/tlaurion) in case we screw with the porting by our self would you mind to share in which range of consulting fee we may motivate you to do that? We would have the [...]

My current plan on how to get heads to the t480 is as following and probably a very bad idea that will result in burning Thinkpads:
1. Copy-and-past the coreboot config for the t480 from libreboot. 
2. Take the kernel config from the t440p.
3. Copy the blobs (me and stuff) created by libreboot into appropriate folders for heads
4. Compile heads with a new t480 target and profit?

This may work at least as a proof-of-concept and we could go from there.

akunterkontrolle, 2025-02-02:
Thank you very much for that detailed plan on how a port of heads for the t480 could come along! @tlaurion 

I will do my best to read up on the various things like signing my git commits. I knew that this is a thing but have never done it myself before so I am sure I will be on a fun ride :D

Well, it seems that I have now "taken the lead" somewhat … Well, I promise I will do my best and I am sure there will be some result at the end. I will try to document my work as best as I can to help the documentation. However, I will say upfront, that it could take a bit until results show up. Sadly this is just a hobby and for various unfair reasons I have to do other stuff as well …

notgivenby, 2025-02-03:
Yesterday I started to compare the coreboot configs like outlined in the plan. My t480 will arrive tomorrow. @akunterkontrolle what would be the best way to communicate during the port and help you?

akunterkontrolle, 2025-02-04:
> I don't know if this would impact the Heads config, but there a couple of things worth mentioning about the T480. It has two variants: one with integrated graphics, and another with integrated & dedicated graphics. At present, the dedicated graphics do not function in coreboot and are essentially disabled by default.
> 

Good point, the same applies to the T440p. I think it is handled there somehow as well. My T480 is the one without a dedicated graphics so I can't really test that. I think it should be documented that the dedicated graphics is untested as well as the screen mods and anyone having this hardware config will play guinea pig.

> Additionally, in light of the previous X230 maximised config, the T480 has become a commonly modified laptop by the ThinkPad community with do-it-yourself screen replacements at 1440p and 4K being fairly popular.

akunterkontrolle, 2025-02-04:
> Yesterday I started to compare the coreboot configs like outlined in the plan. My t480 will arrive tomorrow. [@akunterkontrolle](https://github.com/akunterkontrolle) what would be the best way to communicate during the port and help you?

Nice to hear that you also already started working on it. I have created a fork from heads under my username with a new branch for the T480, see https://github.com/akunterkontrolle/heads/tree/board_t480

Things I started to do there:
1. Added a coreboot fork under modules/coreboot pointing to the coreboot git repo and the commit hash that Libreboot is referencing for the T480. Although the question remains if just adding a new coreboot version in this file and referencing it from the boards config is really enough?
2. Added the coreboot patches from Libreboot. I just added all of them since finding out which of them really apply to the T480 is a bit above my capabilities. I am not sure if it is enough to just dump the patches in or if some script has also to be added to actually apply them.
3. Added a coreboot config based on Libreboot's coreboot config expanded by settings I think are relevant / needed for heads. I used the T440p config from heads to get clues what may be needed. I don't know if it is still missing stuff. I choose to enable TPM2 instead of TPM1, but no idea if that is a good or bad decision, heads seems to have boards with TPM1 and TPM2 … so it should work? Note: I did use the 'menuconfig' utility from coreboot that can be invoked by the libreboots build system to change the config. First of all because it says clearly in the config file that you are not supposed to edit it manually and because the TUI utility has at least a bit more description for the various options.
4. I added files for the board config. I just copied them over from the T440p and slowly started to adapt them. They still have the wrong references for the needed blobs.

So I think the next steps are to add a linux kernel config for the T480 build and tackle the download and modification of the blobs (note: ME needs deguard and we have to check if and when to apply me_cleaner)

Regarding communication: I think we should use mainly this issue which keeps the discussion publicly accessible and helps to document steps, hurdles and question when porting to a new platform. But for some discussion this may not be the right place so you can reach me on Matrix via @ak_unterkontrolle:matrix.org

mblanqui, 2025-02-04:
>> My T480 is the one without a dedicated graphics so I can't really test that. I think it should be documented that the dedicated graphics is untested as well as the screen mods and anyone having this hardware config will play guinea pig.

My T480 has dedicated graphics, I can play guinea pig.

mblanqui, 2025-02-04:
> So I think the next steps are to add a linux kernel config for the T480 build and tackle the download and modification of the blobs (note: ME needs deguard and we have to check if and when to apply me_cleaner)

You would have to check with Máté, but if iirc from when I built coreboot+edk2 for this board, deguard automatically HAP disables the Management Engine. It also leaves the ME region of the bios completely unsigned, so there's the potential to run your own code in place of Intel's.

notgivenby, 2025-02-04:
Great! 

> 2\. I am not sure if it is enough to just dump the patches in or if some script has also to be added to actually apply them.

Perhaps we need to check later the log of the lb build when complied locally to figure out what patches were applied. Perhaps lbmk config can also help. 

> 3\. TPM2 … so it should work? Note: I did use the 'menuconfig' utility from coreboot that can be invoked by the libreboots build system to change the config. First of all because it says clearly in the config file that you are not supposed to edit it manually and because the TUI utility has at least a bit more description for the various options.

I believe TPM2 should be correct, since the t480 has it instead of TPM1. I puzzled a bit about the TPM2. When you enable the TPM2 in coreboot config under the lbmk the rom build gives an error.  For more details see my fork of heads/wiki https://notgivenby.github.io/heads-wiki/t480-port/. I will try to document changes and steps we did there too.  In, short the error is because the menuconfig untility changes the path to 2 vendor files: `Fsp_M.fd` and `Fsp_S.fd`. We need to keep it in mind. After changing it to the correct path everything worked.

notgivenby, 2025-02-04:
> My T480 has dedicated graphics, I can play guinea pig.

I think it will not be a huge problem. Somebody raised an issue on heads a while ago: the person had t420 with the dGPU.  heads was flashed on it actually without knowing that the machine has dGPU. The only think that did not work was DP output to the external display. So, perhaps you should not play a guinea pig yet.

tlaurion, 2025-02-04:
> > Yesterday I started to compare the coreboot configs like outlined in the plan. My t480 will arrive tomorrow. [@akunterkontrolle](https://github.com/akunterkontrolle) what would be the best way to communicate during the port and help you?
> 
> Nice to hear that you also already started working on it. I have created a fork from heads under my username with a new branch for the T480, see https://github.com/akunterkontrolle/heads/tree/board_t480
> 
> Things I started to do there:
> 1. Added a coreboot fork under modules/coreboot pointing to the coreboot git repo and the commit hash that Libreboot is referencing for the T480. Although the question remains if just adding a new coreboot version in this file and referencing it from the boards config is really enough?
> 2. Added the coreboot patches from Libreboot. I just added all of them since finding out which of them really apply to the T480 is a bit above my capabilities. I am not sure if it is enough to just dump the patches in or if some script has also to be added to actually apply them.
> 3. Added a coreboot config based on Libreboot's coreboot config expanded by settings I think are relevant / needed for heads. I used the T440p config from heads to get clues what may be needed. I don't know if it is still missing stuff. I choose to enable TPM2 instead of TPM1, but no idea if that is a good or bad decision, heads seems to have boards with TPM1 and TPM2 … so it should work? Note: I did use the 'menuconfig' utility from coreboot that can be invoked by the libreboots build system to change the config. First of all because it says clearly in the config file that you are not supposed to edit it manually and because the TUI utility has at least a bit more description for the various options.
> 4. I added files for the board config. I just copied them over from the T440p and slowly started to adapt them. They still have the wrong references for the needed blobs.
> 
> So I think the next steps are to add a linux kernel config for the T480 build and tackle the download and modification of the blobs (note: ME needs deguard and we have to check if and when to apply me_cleaner)
> 
> @notgivenby Regarding communication: I think we should use mainly this issue which keeps the discussion publicly accessible and helps to document steps, hurdles and question when porting to a new platform. But for some discussion this may not be the right place so you can reach me on Matrix via @ak_unterkontrolle:matrix.org
> 
> 

`+# CONFIG_TPM_MEASURED_BOOT is not set`

@akunterkontrolle rhis needs to be on!

mblanqui, 2025-02-04:
> I think it will not be a huge problem. Somebody raised an issue on heads a while ago: the person had t420 with the dGPU. heads was flashed on it actually without knowing that the machine has dGPU. [...]

I have three T480s, one with a dGPU (the one I am writing this message on), and two motherboards with an iGPU waiting to go into dead machines, so if testing is needed let me know.

FYI, although Thunderbolt isn't working in coreboot rn, DisplayPort output via the Thunderbolt port (functioning as a simple USB-C port) does work.

notgivenby, 2025-02-09:
@akunterkontrolle @MattClifton76 @mblanqui Today I successfully finished one of the ME blob magic for heads. As soon as I get back to my signing key, I will commit the script. In process, I noticed that the me.bin for libreboot was extracted from a dell - donor which had no vPro ME. Since it would be great to use faster CPU under heads,/Qubes, I wonder  whether you have a t480 with i7 8650u or 8550u (both vPro)? Does it work with libreboot, is it stable? Can it be an issue @tlaurion?

MattClifton76, 2025-02-09:
I have a i7-8650u, it appears to have vPro. Firmware 1.53

notgivenby, 2025-02-09:
> I have a i7-8650u, it appears to have vPro. Firmware 1.53

Did you flash libreboot on it? does it work fine?

MattClifton76, 2025-02-09:
I have not, I plan on giving it a shot after the Super Bowl. I know they say use 1.52 for the uart but I don't have a need for it so I'm going to try it with 1.53. I also updated the thunderbolt chip through fwupd, not sure why they wanted to flash the chip separately.

notgivenby, 2025-02-10:
> add a new targets/t480.mk which deals with calling blobs/* download and extraction of blobs putting them in proper places

@akunterkontrolle I managed to create blobs scripts that were mostly relied on the libreboot approach: [https://github.com/notgivenby/heads/tree/board_t480].  We are still missing modifications of the  .circleci/config.yml to add t480.

mblanqui, 2025-02-11:
> I wonder whether you have a t480 with i7 8650u or 8550u (both vPro)? Does it work with libreboot, is it stable? Can it be an issue [@tlaurion](https://github.com/tlaurion)?

I have a i7 8650u (vPro). I built coreboot+edk2 from upstream in early December. It's perfectly stable, and I've had no issues other than the known ones at the time of my build (Fn keys don't work, and the Thunderbolt port doesn't have full functionality). Other than that, it works like a treat. Thanks to @notgivenby for giving me the heads up (no pun intended) to your question. Let me know if you need more info.

tlaurion, 2025-02-11:
For the adventurous: please report testing of https://github.com/linuxboot/heads/pull/1906 there! :)

MattClifton76, 2025-02-11:
[image] this was as fair as it would boot.

notgivenby, 2025-02-11:
for all brave please keep in mind that t480 is not the same board as t480s. There are few GPIO differences that can destroy the t480s board if flashed with t480 rom as notified on the libreboot.

notgivenby, 2025-02-12:
> For the adventurous: please report testing of [#1906](https://github.com/linuxboot/heads/pull/1906) there! :)

@tlaurion OMG! I am so exited! Thank you!!!!

notgivenby, 2025-02-13:
@MattClifton76 @doritos4mlady would you guys be interested in becoming a heads board tester for t480? Additional information is available here https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$6zRiL6bw55FJLwlhB4VLcc8TbxB_FAu_zaolgtaXbaw?via=matrix.org&via=nitro.chat&via=tchncs.de.
In case you do not have a matrix account, here is the copy paste:

https://github.com/linuxboot/heads/blob/master/BOARD_TESTERS.md is the in-tree willing board testers (with external programmers) to be called for testing, basically when there is coreboot version bumps, linux kernel version bumps or board specific changes that could result in bricking platforms, requiring external programmer recovery. This is why board testers are needed: so that things are tested in pull requests by board owners to at least check that the changes do not result in bricks and no important regressions are found prior of merging changes upstream
https://github.com/linuxboot/heads/issues/692 is pinned issue for people to tell their interest into becoming a platform tester with external programmer. Please show interest of being added to https://github.com/linuxboot/heads/blob/master/BOARD_TESTERS.md in this issue.

MattClifton76, 2025-02-14:
> @MattClifton76 @doritos4mlady would you guys be interested in becoming a heads board tester for t480? Additional information is available here [...]

I'll flash that bios chip to the moon and back, I even bought a new nitro key for this occasion. As stated previously I'm very new at this, I still have to write down commands and look up explanations.

doritos4mlady, 2025-02-18:
> [@MattClifton76](https://github.com/MattClifton76) [@doritos4mlady](https://github.com/doritos4mlady) would you guys be interested in becoming a heads board tester for t480? Additional information [...]

Interested. I also just got a T480, but I won't be able to start contributing to this until hopefully some time next month.

gaspar-ilom, 2025-02-19:
> Interested. I also just got a T480, but I won't be able to start contributing to this until hopefully some time next month.

I am adding both you @doritos4mlady and @MattClifton76 as board testers: https://github.com/tlaurion/heads/pull/17/commits/0416162dedadd7b3402eb37ca0a0e6b443b2e238

tlaurion, 2025-02-20:
For now, t480S won't be added by #1906. A subsequent issue must be opened for t480s, whith its own set of board owners, willing testers, and port leaders.

kjkent, 2025-03-16:
> For now, t480S won't be added by #1906. A subsequent issue must be opened for t480s, whith its own set of board owners, willing testers, and port leaders.

@tlaurion when the time comes, I would happily volunteer as a board tester. I have a T480S and an external programmer, etc. I currently have Libreboot running on it.