linuxboot/heads #1887: ns70 being supported or not supported by ns50 under Dasharo coreboot fork?

tlaurion, 2025-01-15:
This is to track ns50 HCL support.

I was contacted today by ns70 owner having Heads when platform was received and asked to be paid to support that user.

Doing 1:1 support won't help ecosystem. Clarify HCL and compatibility for ns70 with ns50 under Heads tree is needed if ns70 was sold by Nitrokey and therefore Heads, which I was unaware up to today.

@daringer ?

tlaurion, 2025-01-15:
Note that ns50 is becoming untested under https://github.com/linuxboot/heads/pull/1846#issuecomment-2593629570

I will need answer, and changes under board config for HCL, prior of moving back the ns50 from untested to tested in subsequent PR opened by Nitrokey.

tlaurion, 2025-01-15:
Cross-linking https://github.com/Nitrokey/heads/issues/52

Natekb7024, 2025-01-17:
> This is to track ns50 HCL support.
> 
> I was contacted today by ns70 owner having Heads when platform was received and was asked to be paid to support that user.
> 
> Doing 1:1 support won't help ecosystem. Clarify HCL and compatibility for ns70 with ns50 under Heads tree is needed if ns70 was sold by Nitrokey and therefore Heads, which I was unaware up to today.
> 
> [@daringer](https://github.com/daringer) ?

If I'm the one trying to get help, as the release change log says V2.4 NS70 is supported on NS50.

[image]

In support they tell me that they do not help to do a reflashing, except that I must pay (150 euros + shipping) and send the laptop to Germany, what is cumbersome and cost is also contrary to the company's policies. I saw a similar situation in the official forum a user needed help but they answered with the typical answer "1: 1"

So I seek help in exchange for a reward. Unlike Nitrokey, Novacustom and Dasharo if you have a good support for this type of situation.