linuxboot/heads #1893: v560tu : user left in the dark for 1 minute after each firmware upgrade on 96GB units (memory training)

tlaurion, 2025-01-20:
Should be documented somewhere

cc @macpijan

macpijan, 2025-01-20:
Yes, that's a lot of RAM, especially DDR5 which takes longer than DDR4 to train.

Do we always drop the training data on heads update?

@mkopec I assume for UEFI update we would not update this region?

macpijan, 2025-01-20:
WRT documentation: somewhere here: https://osresearch.net/Updating ?

tlaurion, 2025-01-20:
> WRT documentation: somewhere here: https://osresearch.net/Updating ?

I was more thinking of dasharo guide for Heads and heads refer to downstream docs on that matter. Dasharo includes disassembly guides and recovery which heads doesn't plan duplicating.

Ideally heads refer to those pages in board configs. Even more ideally, coreboot doesn't wipe trained data. Not sure if this is bug or feature. But 1m in dark as per video shared in business channel is scary to say the least.

Worst case scenario, heads checks memory amount from kernel and shows message to direct en user expectations if this is gonna become norm for those users ordering new machines with such beefy ram.

mkopec, 2025-01-21:
> @mkopec I assume for UEFI update we would not update this region?

Depends on the flashrom arguments, in some cases we update the entire BIOS region (MRC cache is lost) and in others we only modify WP_RO and RW_SECTION_[A,B] (MRC cache is preserved). As I understand it in Heads we do always drop the MRC cache region.

coreboot has an early sign of life feature exactly for this (shows a text message on screen while memory is training) but I was not able to get it to work with any of the FSP versions I tested with. I'm hoping whenever client FSP is released on GitHub, it'll be fixed.

mkopec, 2025-01-21:
> WRT documentation: somewhere here: https://osresearch.net/Updating ?

> I was more thinking of dasharo guide for Heads and heads refer to downstream docs on that matter. Dasharo includes disassembly guides and recovery which heads doesn't plan duplicating.

We already refer to Heads documentation in our docs: https://docs.dasharo.com/unified/novacustom/firmware-update/

We could add a note to this page that the first boot after updating may take a long time and leave the rest as is. Would that be sufficient?

macpijan, 2025-01-21:
Let's do this. Maybe even at the top if it can be useful for UEFi updates as well?

mkopec, 2025-01-21:
Added: https://github.com/Dasharo/docs/pull/974/commits/0a44e05c78a5595166134245a250fa89f5879233

BeataZdunczyk, 2025-01-22:
@tlaurion The change is live on docs.dasharo: https://docs.dasharo.com/unified/novacustom/firmware-update/. Please close the issue if you find it sufficient.

tlaurion, 2025-01-22:
Considering my nv41 has also 64gb ram, is this sufficient as an explanation not blaming ddr5 explicitly? Or is there something else involved in the longer ram init for v560tu as compared to nv41 for same ram amount?

I'm drafting a PR pointing to relative docs for review.

tlaurion, 2025-01-22:
@macpijan @BeataZdunczyk Please review/approve #1896 which should be enough to close this issue.

closed 2025-01-22 by tlaurion