linuxboot/heads #1894: v560tu seems slow on IO, needs confirmation

tlaurion, 2025-01-20:
Only thing I can do is compare across devices.

This is `systemd-analyze blame` output, for reference

My nv41 has bigger templates and is my main driver.

### nv41

[image]

### v560tu

[image]

cc @macpijan

macpijan, 2025-01-20:
I assume it can be useful to document disks in both setups as well

tlaurion, 2025-01-20:
Not sure. Reencryption works @1.4GiB/s, with power draw of 35w, no fan spinning. 

[image]

mkopec, 2025-01-21:
What are the models of SSDs in both laptops? V560TU could have a DRAM-less disk which are much slower in random IO

macpijan, 2025-01-21:
I think it's reasonably to assume disks are from what @wessel-novacustom is offering in the configurator.

V56 offers only one option: https://novacustom.com/product/v56-series/

While NV41 used to offer Samsung disks: https://novacustom.com/product/nv41-series/

Can we please summarize the exact hw configurations being compared here?

mkopec, 2025-01-21:
NV41 can come with 980 Pro (with DRAM cache) or 980 (DRAM-less). Apparently the PX700 disk in V560TU does not have DRAM cache but they sell it as `HMB 3.0` which uses host RAM for caching. I have no idea whether or not that works in Qubes

mkopec, 2025-01-21:
Here's what it looks like with V560TU on my side (96GB / 2TB)

tlaurion, 2025-01-21:
@mkopec @macpijan @wessel-novacustom : updated OP

> nv41 64 GB ram, drive: Samsung SSD 980 PRO 2TB

and 

>v560tu 96gb ram, drive: SSDPR-PX700-04T-80

Opened issue https://github.com/QubesOS/qubes-issues/issues/9723

Want me to run some tests? Please detail.

Also note that as pointed out at https://github.com/linuxboot/heads/pull/1889#issuecomment-2598849847:

> https://github.com/Dasharo/coreboot/compare/94e5f5d5b808cf8d8fd5c70d4ef6a08a054f8986...048ca832325d716fcab596822b10f5d493fc2312

When we attempted that coreboot version bump, perf worsen to the point of systemd services timing out on first stage of qubesos installation, and templates installation took more than an hour.

tlaurion, 2025-01-21:
> Also note that as pointed out at [#1889 (comment)](https://github.com/linuxboot/heads/pull/1889#issuecomment-2598849847):
> 
> > [Dasharo/coreboot@94e5f5d...048ca83](https://github.com/Dasharo/coreboot/compare/94e5f5d5b808cf8d8fd5c70d4ef6a08a054f8986...048ca832325d716fcab596822b10f5d493fc2312)
> 
> When we attempted that coreboot version bump, perf worsen to the point of systemd services timing out on first stage of qubesos installation, and templates installation took more than an hour.

Some additional notes from testing, will edit (re-testing https://github.com/tlaurion/heads/tree/perf_comparison_with_reverted_coreboot_version_bump_v560tu)
- ram init takes around 2m on 96gb ram, not 1
- Heads fb refresh (fbwhiptail drawing) slower.
- Heads hashing of /boot content slower
- Resealing TPM Disk Unlock Key (cpu based derivation of DRK to validate slot + new key from luks DRK into new slot) way slower

Insight is not only IO being slower, but as if CPU speed was limited to be turtle speed as well. Will repost systemd-analyze blame in next edit

tlaurion, 2025-01-21:
`systemd-analyze blame` with https://github.com/linuxboot/heads/commit/36e30d0174dbd351ea13ee1a5659979843939da3 for comparison

[systemd-analyze_blame_master.txt](https://github.com/user-attachments/files/18495363/systemd-analyze_blame_master.txt)

CC @macpijan @mkopec

macpijan, 2025-01-22:
> If I compare I/O write and especially read operations (especially with small files), the -TU models are performing bad compared to the NVIDIA variants. Now that's for UEFI of course, but might still make sense to compare to see if there is a difference and why.

I agree this is to be checked. Did we have a tracking of this problem already?

Also, do we want to do it right now?
Should this be gating heads release if (to be confirmed) performance would be similar on the same device across UEFI and heads firmware?

Ideally, we release heads release from the same (similar) coreboot base as the previous UEFI release.

macpijan, 2025-01-22:
> Some additional notes from testing, will edit (re-testing https://github.com/tlaurion/heads/tree/perf_comparison_with_reverted_coreboot_version_bump_v560tu)
> 
>     * ram init takes around 2m on 96gb ram, not 1
> 
>     * Heads fb refresh (fbwhiptail drawing) slower.
> 
>     * Heads hashing of /boot content slower
> 
>     * Resealing TPM Disk Unlock Key (cpu based derivation of DRK to validate slot + new key from luks DRK into new slot) way slower
> 
> 
> Insight is not only IO being slower, but as if CPU speed was limited to be turtle speed as well.
> 
> `systemd-analyze blame` with branch including changes to coreboot version bump:

Thanks for this test, this will be useful testing point for the future. We should focus on testing what we are aiming to release right now, however.

We propose that we both confirm once again that the binary from [this commit](https://github.com/linuxboot/heads/commit/36e30d0174dbd351ea13ee1a5659979843939da3) is "OK" in terms of performance as reported by us here previously: https://github.com/linuxboot/heads/issues/1894#issuecomment-2604624721

If confirmed, we propose that we use this commit as a release for Dasharo+heads @tlaurion @wessel-novacustom to not postpone it any longer.

We can continue investigating the performance concerns, such as raised by @wessel-novacustom here https://github.com/linuxboot/heads/issues/1894#issuecomment-2604602279 in individual dasharo issues.

wessel-novacustom, 2025-01-22:
> Ideally, we release heads release from the same (similar) coreboot base as the previous UEFI release.

- I can see that, but this performance issue is one of the biggest issues the -TU series has.

Ideally, it would be fixed, despite the coreboot base being slightly different in that case.

In case a coreboot patch could be made to fix this issue, I'm interested in getting a link to that patch, so I can assist customers with UEFI firmware who are complaining about this.

**My suggestion is to make a `v0.9.2-rc1`/`v1.0.0-rc1` untested UEFI release with just this fix patch.** We can then use that version as the base for Heads.

macpijan, 2025-01-22:
> Ideally, it would be fixed, despite the coreboot base being slightly different in that case.

I will move this off-channel to discuss quicker and come back here with conclusions.

mkopec, 2025-01-22:
> We propose that we both confirm once again that the binary from [this commit](https://github.com/linuxboot/heads/commit/36e30d0174dbd351ea13ee1a5659979843939da3) is "OK" in terms of performance as reported by us here previously: https://github.com/linuxboot/heads/issues/1894#issuecomment-2604624721

- 28.983s - 155H, 32G RAM, 1GB SSD

mkopec, 2025-01-22:
...isn't `qubes-vm@sys-whonix.service` dependent on network connection anyway? So we're no longer comparing only I/O speeds

mkopec, 2025-01-22:
Here's my `kdiskmark` result in `personal` Qube on the same laptop with 155H, 32GB RAM, 1TB SSD:

[image]

all with default settings.

@tlaurion can you share yours for comparison? I think that would be the best option for comparing just I/O performance

tlaurion, 2025-01-22:
> Here's my `kdiskmark` result in `personal` Qube on the same laptop with 155H, 32GB RAM, 1TB SSD, AC in:
> 
> [image]
> 
> all with default settings. Of course with LUKS, LVM and Xen between disk and benchmark this isn't measuring *just* I/O but it should be more objective.
> 
> @tlaurion can you share yours for comparison?
> 
> 

Sorry 64gb, will edit prior posts :(
Past v560tu had 96gb and 2tb drive. This one has 64gb and 4tb m2 drive. 

[image]

[image]

[image]

mkopec, 2025-01-22:
Okay, so you are having significantly worse performance than me... I think next we will test on the same 4TB disk model as in your V560TU

wessel-novacustom, 2025-01-22:
We should mark this as a known issue for the Dasharo corboot+Heads release. @macpijan @tlaurion

tlaurion, 2025-01-23:
> We should mark this as a known issue for the Dasharo corboot+Heads release. @macpijan @tlaurion 

How come is this a Heads issue? As opposed to Dasharo-UEFI or simply coreboot or m2 drive? I'm not sure I follow.

wessel-novacustom, 2025-01-23:
It's not specifically a Heads issue, @tlaurion. I just mean that we won't further investigate this for the Dasharo version that @macpijan is about to release.

tlaurion, 2025-01-23:
> It's not specifically a Heads issue, @tlaurion. I just mean that we won't further investigate this for the Dasharo version that @macpijan is about to release.

So I guess what you mean is that the same notes should be present for both Dasharo-UEFI and Dasharo-Heads, since it's not a Heads specific.

wessel-novacustom, 2025-01-23:
Issue has been raised separately for UEFI: https://github.com/Dasharo/dasharo-issues/issues/1216

marmarek, 2025-01-25:
@tlaurion do you have any CLI version of a benchmark that shows this issue? Something that would produce a text output that I can then parse with a script, compare with plain diff etc. I would like to add disk performance test to our CI, but manually comparing graphical screenshots is not going to fly.

marmarek, 2025-01-25:
Maybe some specific configs to the `fio` tool?

tlaurion, 2025-01-25:
> @tlaurion do you have any CLI version of a benchmark that shows this issue? Something that would produce a text output that I can then parse with a script, compare with plain diff etc. I would like [...]

I don't have insights on this matter, unfortunately.

tlaurion, 2025-01-25:
> Maybe some specific configs to the `fio` tool?

@marmarek any command line command you propose I can reuse. kdisktats is what is used by end users on forum to show things visually. Replicating that kdisktats does, from a command line perspective, coukd help here, yes.

tlaurion, 2025-01-26:
> Maybe some specific configs to the `fio` tool?

Untested, please advise @marmarek 

### Random Read Test
```bash
fio --name=randread-test --filename=/path/to/testfile --rw=randread --bs=4k \
    --size=1G --numjobs=4 --time_based --runtime=60 --iodepth=32 \
    --ioengine=libaio --direct=1 --group_reporting
```

### Random Write Test
```bash
fio --name=randwrite-test --filename=/path/to/testfile --rw=randwrite --bs=4k \
    --size=1G --numjobs=4 --time_based --runtime=60 --iodepth=32 \
    --ioengine=libaio --direct=1 --group_reporting
```

### Mixed Random Read/Write Test
```bash
fio --name=randrw-test --filename=/path/to/testfile --rw=randrw --bs=4k \
    --size=1G --numjobs=4 --time_based --runtime=60 --iodepth=32 \
    --ioengine=libaio --direct=1 --group_reporting
```

### Sequential Read Test
```bash
fio --name=seqread-test --filename=/path/to/testfile --rw=read \
    --bs=256k --size=1G --numjobs=4 --time_based --runtime=60 \
    --iodepth=32 --ioengine=libaio --direct=1 --group_reporting
```

### Sequential Write Test
```bash
fio --name=seqwrite-test --filename=/path/to/testfile --rw=write \
    --bs=256k --size=1G --numjobs=4 --time_based --runtime=60 \
    --iodepth=32 --ioengine=libaio --direct=1 --group_reporting
```

### Explanation of Parameters:
- **`--rw`**: Specifies the type of I/O pattern (`randread`, `randwrite`, `randrw`, `read`, or `write`).
- **`--bs`**: Block size (e.g., 4k for random, 256k for sequential).
- **`--iodepth`**: Queue depth for I/O operations.
- **`--numjobs`**: Number of parallel jobs.
- **`--runtime`**: Duration of the test in seconds.
- **`--direct`**: Bypasses OS caching for direct disk access.

These commands will output metrics such as IOPS, bandwidth (throughput), average latency, and queue depth, allowing you to replicate the performance insights provided by `kdisktats`.

Citations :
[1] Sample FIO Commands for Block Volume Performance Tests on ... https://docs.oracle.com/en-us/iaas/Content/Block/References/samplefiocommandslinux.htm
[2] Benchmark Persistent Disk performance on a Linux VM https://cloud.google.com/compute/docs/disks/benchmarking-pd-performance-linux
[3] 1. fio - Flexible I/O tester rev. 3.38 - FIO's documentation! https://fio.readthedocs.io/en/latest/fio_doc.html
[4] Fio Benchmark | JuiceFS Document Center https://juicefs.com/docs/cloud/benchmark/fio/
[5] Running and understanding write workload using FIO – Yugabyte https://support.yugabyte.com/hc/en-us/articles/13874049853965-Running-and-understanding-write-workload-using-FIO
[6] Performance benchmarking with Fio on Nutanix https://portal.nutanix.com/kb/12075
[7] Elastic Compute Service:Test the performance of block storage ... https://www.alibabacloud.com/help/en/ecs/user-guide/test-the-performance-of-block-storage-devices
[8] Review fio test : r/linuxadmin - Reddit https://www.reddit.com/r/linuxadmin/comments/13u4xub/review_fio_test/

marmarek, 2025-01-26:
I prepared something similar based on example configs provided with the fio tool: https://github.com/QubesOS/qubes-core-admin/pull/649
First results: https://openqa.qubes-os.org/tests/126917/file/system_tests-tests-qubes.tests.integ.storage_perf.log
https://openqa.qubes-os.org/tests/126917/logfile?filename=system_tests-perf_test_results.txt (import to oocalc or any other spreadsheet for easier reading)

This is running on a "hw1" runner (some HP laptop), and I'm just repeating the test with kernel-latest. So far, results are pretty much the same (no regression detected). I'll repeat the tests on NV41 or v56. If difference is confirmed with this tool, I can for example run automated git bisect on the kernel between good and bad version to identify specific change causing the regression.

tlaurion, 2025-01-26:
> I prepared something similar based on example configs provided with the fio tool: https://github.com/QubesOS/qubes-core-admin/pull/649
> First results: [...]

@marmarek note that my results for nv41 were on btrfs as note above the screenshots in comment. It was only a reminder that if at cause, v560tu depending on kernel-latest might explain partly perf degradation, where unfortunately, one cannot test kernel-stable on v560tu which requires kernel-latest on initial install for meteorlake support as opposed to nv41.

marmarek, 2025-01-27:
I did a test with fio on NV41, using very similar config to "Mixed Random Read/Write Test"  above (difference: I let it run for 90s, not 60s; and iodepth 16 not 32) with different kernel versions. I do see a small slowdown with 6.12 kernel in dom0, but nowhere near the effect you see. I'm looking at "read_bandwidth_kb" and "write_bandwidth_kb" columns (column 7 and 48 respectively):

fio in dom0:
[fio-kernel-6.6.log](https://github.com/user-attachments/files/18553131/fio-kernel-6.6.log)
[fio-kernel-6.11.log](https://github.com/user-attachments/files/18553132/fio-kernel-6.11.log)
[fio-kernel-6.12.log](https://github.com/user-attachments/files/18553136/fio-kernel-6.12.log)
fio in VM:
[fio-dom0-6.6-vm-6.11.log](https://github.com/user-attachments/files/18553133/fio-dom0-6.6-vm-6.11.log)
[fio-dom0-6.6-vm-6.6.log](https://github.com/user-attachments/files/18553135/fio-dom0-6.6-vm-6.6.log)
[fio-dom0-6.12-vm-6.12.log](https://github.com/user-attachments/files/18553134/fio-dom0-6.12-vm-6.12.log)

@tlaurion can you check if you see the drastic difference with fio on your side? I did the test on LVM (very default installation of R4.2), if fio reports larger difference for you, then maybe it's btrfs-specific? But if you don't see the difference either, then we need another test. In that case, can you check other examples you provided above? Does any of them show the 2x+ difference between kernel versions for you?

tlaurion, 2025-01-27:
> I did a test with fio on NV41, using very similar config to "Mixed Random Read/Write Test" above (difference: I let it run for 90s, not 60s; and iodepth 16 not 32) with different kernel versions. I [...]

dom0 comparatives and used script @marmarek. btrfs, no revision to keep, 2tb drive with sdram cache, 64gb ram.

[dom0_6.6.68-1.qubes.fc37.x86_64_additional-seq-read_output.txt](https://github.com/user-attachments/files/18563916/dom0_6.6.68-1.qubes.fc37.x86_64_additional-seq-read_output.txt)
[dom0_6.6.68-1.qubes.fc37.x86_64_additional-rand-read_output.txt](https://github.com/user-attachments/files/18563922/dom0_6.6.68-1.qubes.fc37.x86_64_additional-rand-read_output.txt)
[dom0_6.6.68-1.qubes.fc37.x86_64_additional-rand-write_output.txt](https://github.com/user-attachments/files/18563923/dom0_6.6.68-1.qubes.fc37.x86_64_additional-rand-write_output.txt)
[dom0_6.6.68-1.qubes.fc37.x86_64_additional-seq-write_output.txt](https://github.com/user-attachments/files/18563925/dom0_6.6.68-1.qubes.fc37.x86_64_additional-seq-write_output.txt)
[dom0_6.6.68-1.qubes.fc37.x86_64_seq-read_output.txt](https://github.com/user-attachments/files/18563930/dom0_6.6.68-1.qubes.fc37.x86_64_seq-read_output.txt)
[dom0_6.6.68-1.qubes.fc37.x86_64_rand-read_output.txt](https://github.com/user-attachments/files/18563926/dom0_6.6.68-1.qubes.fc37.x86_64_rand-read_output.txt)
[dom0_6.6.68-1.qubes.fc37.x86_64_rand-write_output.txt](https://github.com/user-attachments/files/18563927/dom0_6.6.68-1.qubes.fc37.x86_64_rand-write_output.txt)
[dom0_6.6.68-1.qubes.fc37.x86_64_seq-write_output.txt](https://github.com/user-attachments/files/18563931/dom0_6.6.68-1.qubes.fc37.x86_64_seq-write_output.txt)
[dom0_6.12.9-1.qubes.fc37.x86_64_additional-seq-read_output.txt](https://github.com/user-attachments/files/18563928/dom0_6.12.9-1.qubes.fc37.x86_64_additional-seq-read_output.txt)
[dom0_6.12.9-1.qubes.fc37.x86_64_additional-rand-read_output.txt](https://github.com/user-attachments/files/18563919/dom0_6.12.9-1.qubes.fc37.x86_64_additional-rand-read_output.txt)
[dom0_6.12.9-1.qubes.fc37.x86_64_additional-rand-write_output.txt](https://github.com/user-attachments/files/18563918/dom0_6.12.9-1.qubes.fc37.x86_64_additional-rand-write_output.txt)
[dom0_6.12.9-1.qubes.fc37.x86_64_additional-seq-write_output.txt](https://github.com/user-attachments/files/18563917/dom0_6.12.9-1.qubes.fc37.x86_64_additional-seq-write_output.txt)
[dom0_6.12.9-1.qubes.fc37.x86_64_seq-read_output.txt](https://github.com/user-attachments/files/18563921/dom0_6.12.9-1.qubes.fc37.x86_64_seq-read_output.txt)
[dom0_6.12.9-1.qubes.fc37.x86_64_rand-read_output.txt](https://github.com/user-attachments/files/18563929/dom0_6.12.9-1.qubes.fc37.x86_64_rand-read_output.txt)
[dom0_6.12.9-1.qubes.fc37.x86_64_rand-write_output.txt](https://github.com/user-attachments/files/18563924/dom0_6.12.9-1.qubes.fc37.x86_64_rand-write_output.txt)
[dom0_6.12.9-1.qubes.fc37.x86_64_seq-write_output.txt](https://github.com/user-attachments/files/18563920/dom0_6.12.9-1.qubes.fc37.x86_64_seq-write_output.txt)

tar.gz with scripts used inside:

[nv41_fio_tests.tar.gz](https://github.com/user-attachments/files/18563946/nv41_fio_tests.tar.gz)

tlaurion, 2025-01-28:
For the sake of testing, here are the fio commands called under the hood with default profile when opening kdiskmark @marmarek 
```
/usr/bin/fio --output-format=json --create_only=1 --filename=/kdiskmark-skCGNj.tmp --size=1024m --zero_buffers=0 --name=prepare
/usr/bin/fio --output-format=json --ioengine=libaio --randrepeat=0 --refill_buffers --end_fsync=1 --direct=1 --rwmixread=70 --filename=/kdiskmark-skCGNj.tmp --name=randread --size=1024m --zero_buffers=0 --bs=4k --runtime=5 --rw=randread --iodepth=1 --numjobs=1
/usr/bin/fio --output-format=json --ioengine=libaio --randrepeat=0 --refill_buffers --end_fsync=1 --direct=1 --rwmixread=70 --filename=/kdiskmark-skCGNj.tmp --name=randread --size=1024m --zero_buffers=0 --bs=4k --runtime=5 --rw=randread --iodepth=32 --numjobs=1
/usr/bin/fio --output-format=json --ioengine=libaio --randrepeat=0 --refill_buffers --end_fsync=1 --direct=1 --rwmixread=70 --filename=/kdiskmark-skCGNj.tmp --name=randwrite --size=1024m --zero_buffers=0 --bs=4k --runtime=5 --rw=randwrite --iodepth=1 --numjobs=1
/usr/bin/fio --output-format=json --ioengine=libaio --randrepeat=0 --refill_buffers --end_fsync=1 --direct=1 --rwmixread=70 --filename=/kdiskmark-skCGNj.tmp --name=randwrite --size=1024m --zero_buffers=0 --bs=4k --runtime=5 --rw=randwrite --iodepth=32 --numjobs=1
/usr/bin/fio --output-format=json --ioengine=libaio --randrepeat=0 --refill_buffers --end_fsync=1 --direct=1 --rwmixread=70 --filename=/kdiskmark-skCGNj.tmp --name=read --size=1024m --zero_buffers=0 --bs=1024k --runtime=5 --rw=read --iodepth=1 --numjobs=1
/usr/bin/fio --output-format=json --ioengine=libaio --randrepeat=0 --refill_buffers --end_fsync=1 --direct=1 --rwmixread=70 --filename=/kdiskmark-skCGNj.tmp --name=read --size=1024m --zero_buffers=0 --bs=1024k --runtime=5 --rw=read --iodepth=8 --numjobs=1
/usr/bin/fio --output-format=json --ioengine=libaio --randrepeat=0 --refill_buffers --end_fsync=1 --direct=1 --rwmixread=70 --filename=/kdiskmark-skCGNj.tmp --name=write --size=1024m --zero_buffers=0 --bs=1024k --runtime=5 --rw=write --iodepth=1 --numjobs=1
/usr/bin/fio --output-format=json --ioengine=libaio --randrepeat=0 --refill_buffers --end_fsync=1 --direct=1 --rwmixread=70 --filename=/kdiskmark-skCGNj.tmp --name=write --size=1024m --zero_buffers=0 --bs=1024k --runtime=5 --rw=write --iodepth=8 --numjobs=1
```

So we should abort this and simply consider the the is a btrfs perf regression for 6.12.9. Irrelevant.

> The laptop has Dasharo (coreboot+heads) 0.9.0. Is there any relevant change in 0.9.1 that could affect results?

I do not know what to do with nv41 vs v560tu performance, where nv41 performs better. 
That would be a question for @macpijan, is https://github.com/Dasharo/dasharo-issues/issues/1216 and currently planned to be marked as "known issue" for v560tu release @wessel-novacustom @marmarek.

marmarek, 2025-01-29:
Ok, so I reinstalled the NV41 with BTRFS, and I got different results. In the order of running:

[image]
[image]
[image]
[image]
[image]
[image]

So, first of all, most are a lot slower than on LVM. Same hardware, also fresh install (almost empty disk etc). I added also `discard=async` as suggested in https://github.com/QubesOS/qubes-issues/issues/6476#issuecomment-2512664792, but later tests were still quite slow even with this option.
But more importantly, two tests with 6.12 kernel in dom0 are "fast", in fact, those are the best results of all 6 runs - better than 6.6. But then, after restarting the VM (and later dom0 too), I couldn't replicate the good results anymore. So, there is more to it than just the kernel version.

Next tests I need to do with a script (update with the new fio commands etc), to be sure I haven't missed any step. But that will need to wait for the next week unfortunately.

tlaurion, 2025-02-03:
> We should mark this as a known issue for the Dasharo corboot+Heads release. [@macpijan](https://github.com/macpijan) [@tlaurion](https://github.com/tlaurion)

removed release tag

tlaurion, 2026-01-09:
@macpijan this branch fixes a couple of defconfig that otherwise cause issues

https://github.com/tlaurion/heads/commits/pre_release_dasharo_101_rc3/