linuxboot/heads #1900: dmsetup.c:(.text+0x294e): undefined reference to `dm_filemapd_mode_from_string'

saper, 2025-01-25:
## Context of the Build

**1. What board are you trying to build?**

**2. What repository:branch are you using to build from?**
- [X] Heads:Master 6b73d6d7cd7eb22b4623b4ebd1ecfffe34d63c31
- [ ] Other (please specify)

**3. What version of coreboot are you trying to build**

I think coreboot-24.02.01 will be pulled in

**4. In building the rom where did you get the blobs?**
- [ ] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [X] Extracted from another backup rom taken from another device (please identify the board model)
- [ ] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
- [ ] Yes
- [X] No

**6. What operating system are you using**

Fedora 33

## Please describe the problem

```
make[3]: Entering directory '/home/saper/space/heads/build/x86/lvm2.2.03.23/libdm/dm-tools'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/saper/space/heads/build/x86/lvm2.2.03.23/conf'
make -C include
make[2]: Entering directory '/home/saper/space/heads/build/x86/lvm2.2.03.23/include'
    [CC] dmsetup
/home/saper/space/heads/crossgcc/x86/bin/x86_64-linux-musl-gcc -fdebug-prefix-map=/home/saper/space/heads=heads -gno-record-gcc-switches -D__MUSL__ --sysroot  /home/saper/space/heads/install/x86 -isystem /home/saper/space/heads/install/x86/include -L/home/saper/space/heads/install/x86/lib  -Os  -fPIC   -Wl,--export-dynamic \
      -o dmsetup dmsetup.o -L../../libdm/ioctl -ldevmapper -lpthread    -lm
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/saper/space/heads/build/x86/lvm2.2.03.23/include'
make -C libdaemon
make[2]: Entering directory '/home/saper/space/heads/build/x86/lvm2.2.03.23/libdaemon'
/home/saper/space/heads/crossgcc/x86/bin/../lib/gcc/x86_64-linux-musl/9.4.0/../../../../x86_64-linux-musl/bin/ld: dmsetup.o: in function `_stats_update_file':
dmsetup.c:(.text+0x2b10): undefined reference to `dm_stats_bind_from_fd'
/home/saper/space/heads/crossgcc/x86/bin/../lib/gcc/x86_64-linux-musl/9.4.0/../../../../x86_64-linux-musl/bin/ld: dmsetup.c:(.text+0x2b7a): undefined reference to `dm_stats_start_filemapd'
/home/saper/space/heads/crossgcc/x86/bin/../lib/gcc/x86_64-linux-musl/9.4.0/../../../../x86_64-linux-musl/bin/ld: dmsetup.c:(.text+0x2bb3): undefined reference to `dm_stats_update_regions_from_fd'
/home/saper/space/heads/crossgcc/x86/bin/../lib/gcc/x86_64-linux-musl/9.4.0/../../../../x86_64-linux-musl/bin/ld: dmsetup.o: in function `_status':
dmsetup.c:(.text+0x8200): undefined reference to `dm_task_ima_measurement'
/home/saper/space/heads/crossgcc/x86/bin/../lib/gcc/x86_64-linux-musl/9.4.0/../../../../x86_64-linux-musl/bin/ld: dmsetup.o: in function `_stats_create':
dmsetup.c:(.text+0x97e7): undefined reference to `dm_stats_bind_from_fd'
/home/saper/space/heads/crossgcc/x86/bin/../lib/gcc/x86_64-linux-musl/9.4.0/../../../../x86_64-linux-musl/bin/ld: dmsetup.c:(.text+0x98f1): undefined reference to `dm_stats_start_filemapd'
/home/saper/space/heads/crossgcc/x86/bin/../lib/gcc/x86_64-linux-musl/9.4.0/../../../../x86_64-linux-musl/bin/ld: dmsetup.o: in function `_stats_get_filemapd_mode':
dmsetup.c:(.text+0x294e): undefined reference to `dm_filemapd_mode_from_string'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:66: dmsetup] Error 1
make[3]: Leaving directory '/home/saper/space/heads/build/x86/lvm2.2.03.23/libdm/dm-tools'
make[2]: *** [../libdm/make.tmpl:337: dm-tools.device-mapper] Error 2
make[2]: Leaving directory '/home/saper/space/heads/build/x86/lvm2.2.03.23/libdm'
make[1]: *** [make.tmpl:398: libdm] Error 2
``` 

**To Reproduce**
Steps to reproduce the behavior:

`make BOARD=x230-maximized`

**Additional context**

Are we using host installation by chance and not the built code?

tlaurion, 2025-01-25:
@saper long time no see!!!

You seem to have missed buildsystem having switched to nix and nix docker images being used by CircleCI and for local builds from https://github.com/linuxboot/heads/pull/1660 up to latest changes on master with https://github.com/linuxboot/heads/pull/1855

Basically as README.md documents at https://github.com/linuxboot/heads?tab=readme-ov-file#pull-docker-hub-image-to-prepare-reproducible-roms-as-circleci-in-one-call

you should do, after having docker ce installed:
`./docker_repro.sh make BOARD=x230-maximized`

Do that answer your question? That was done so that issues like this are never opened again (Fedora 33 vs Fedora 40 or any other OS related issues for build should never be opened again, hopefully). If so, please close this issue!

tlaurion, 2025-01-25:
Some confusion seems to stick on how to build Heads, also happened under [matrix channel discussion](https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$T-rnpOUNx6_BflyFAieBDfYo2TW55lNDzDPaTm_EV4s?via=matrix.org&via=nitro.chat&via=envs.net) any advice on doc, and proper renaming of this issue, would be welcome.

saper, 2025-01-25:
Hello :) long time indeed. Wanted to play with some cool stuff on the ThinkPad and thought I need to upgrade. 

Quick answer:

The problem was easy to solve: lvm2.2.02.168 has been there before and now we have lvm2.2.03.23. lvm2.2.02.168 does not have this function yet and has been installed in `heads/install/x86` before.

This means that `lvm2` regrettably used the stuff that has been installed before, so the build is not fully reproducible. 

If anyone runs into this: just remove anything with `libdevmapper*` in the `install/x86` and you'll be fine.

Long answer:

We don't do nix and docker here, we are strong believers in `make` and incremental builds :) 
but of course will check out the flags, the `Dockerfile` and stuff. Basically I get what you want - a clean build every time, but I don't like to wait when I do small changes. Will check out `nix` though. Ideally I would build it all on FreeBSD where I have most computing power but that ship might have sailed long time ago.

tlaurion, 2025-01-25:
> Long answer:
> 
> We don't do nix and docker here, we are strong believers in `make` and incremental builds :) 

Hmm. Podman maybe on BSD? Not tried though. 

> but of course will check out the flags, the `Dockerfile` and stuff. 

flake.nix in tree is used to create the docker image from nix locked package list from flake.lock; there is no Dokerfile per se. Nix is used to create pushed docker hub versioned images. 

> Basically I get what you want - a clean build every time, but I don't like to wait when I do small changes. Will check out `nix` though. Ideally I would build it all on FreeBSD where I have most computing power but that ship might have sailed long time ago.

Maybe it would be better to depend on podman than docker, completely forgot about that use case @saper

saper, 2025-01-25:
Don't worry about it too much, please, it is only for only one crazy person here. Especially given kexec cannot boot FreeBSD so why bother.  

I think the problem here is relevant to docker setup as well - it is about having files lying withing the build/install tree that interfere with the update packages. Not a super easy one to solve.

tlaurion, 2025-01-25:
> Don't worry about it too much, please, it is only for only one crazy person here. Especially given kexec cannot boot FreeBSD so why bother.  

OK. 

> 
> I think the problem here is relevant to docker setup as well - it is about having files lying withing the build/install tree that interfere with the update packages. Not a super easy one to solve.

I use helpers I created to deal with this:
https://github.com/linuxboot/heads/blob/2ba5a0420b91e15f4c9663edbd723a6c704f54e9/Makefile#L911

tlaurion, 2025-03-21:
Solution posted under https://github.com/linuxboot/heads/issues/1900#issuecomment-2614099769

closed 2025-03-21 by tlaurion