linuxboot/heads #1922: IOMMU DMA attack

Lapushy6351, 2025-03-09:
Hi,

IOMMU It is used to protect against AMD attack in Dasharo + coreboot, does this during system startup.

[https://osresearch.net/Heads-threat-model/#peripheral-firmware](url)

[https://docs.dasharo.com/dasharo-menu-docs/dasharo-system-features/#dasharo-security-options](url)

On Clevo NV41 and Clevo NS50 models this is disabled, and I don't understand why.

[https://github.com/linuxboot/heads/blob/master/config/coreboot-novacustom-nv4x_adl.config#L426](url)

[https://github.com/linuxboot/heads/blob/d4c4e5699b89365a88d9d49748dbcc11b6394907/config/coreboot-nitropad-ns50.config#L426](url)

@tlaurion why?

tlaurion, 2025-03-09:
It's not supported because FSP blobs.

Details upstream https://github.com/Dasharo/dasharo-issues/issues/985#issuecomment-2678494719

tlaurion, 2025-03-09:
> IOMMU It is used to protect against AMD attack in Dasharo + coreboot, does this during system startup.

Not sure about what you meant here @Lapushy6351

I renamed issue for better tracking and linked to upstream coreboot fork known issue for Dasharo

Lapushy6351, 2025-03-15:
> It's not supported because Intel FSP blobs missing requirements, to be troubleshooted and FSP requirements updated, fixed upstream and then downstream under Heads, pointing to new coreboot commit that will include the fix.
> 
> Details upstream [Dasharo/dasharo-issues#985 (comment)](https://github.com/Dasharo/dasharo-issues/issues/985#issuecomment-2678494719)

and what are the compatible models or as the identifications?