linuxboot/heads #1942: Garbled graphics with higher display resolutions on T420

haxalicious, 2025-04-01:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

1. What board are you using? t420-maximized

2. Does your computer have a dGPU or is it iGPU-only?
    - [ ] dGPU (Distinct GPU other then internal GPU)
    - [x] iGPU-only (Internal GPU, normally Intel GPU)

3. Who installed Heads on this computer?
    - [ ] Insurgo (Issues to be reported at https://github.com/linuxboot/heads/issues)
    - [ ] Nitrokey (Issues to be reported at https://github.com/Nitrokey/heads/issues)
    - [ ] Purism (Issues to be reported at https://source.puri.sm/firmware/pureboot/-/issues)
    - [ ] Novacustom (Issues to be reported at https://github.com/Dasharo/dasharo-issues)
    - [ ] HardnenedVault (Issues to be reported at https://github.com/hardenedvault/vaultboot/issues)
    - [ ] Other provider
    - [x] Self-installed

4. What PGP key is being used?
    - [ ] Librem Key (Nitrokey Pro 2 rebranded)
    - [ ] Nitrokey Pro
    - [ ] Nitrokey Pro 2
    - [ ] Nitrokey 3 NFC
    - [ ] Nitrokey 3 NFC Mini
    - [ ] Nitrokey Storage
    - [ ] Nitrokey Storage 2
    - [x] Yubikey
    - [ ] Other

5. Are you using the PGP key to provide HOTP verification?
    - [ ] Yes
    - [x] No
    - [ ] I don't know

### B. Identify how the board was flashed

1. Is this problem related to updating heads or flashing it for the first time?
    - [] First-time flash
    - [ ] Updating heads 

2. If the problem is related to an update, how did you attempt to apply the update?
    - [ ] Using the Heads menus
    - [ ] Flashrom via the Recovery Shell
    - [ ] External flashing

3. How was Heads initially flashed?
    - [ ] External flashing
    - [ ] Internal-only / 1vyprep+1vyrain / skulls
    - [ ] Don't know

4. Was the board flashed with a maximized or non-maximized/legacy rom?
    - [x] Maximized
    - [ ] Non-maximized / legacy
    - [ ] I don't know

5. If Heads was externally flashed, was IFD unlocked?
    - [ ] Yes
    - [ ] No
    - [x] Don't know

### C. Identify the rom related to this bug report

1. Did you download or build the rom at issue in this bug report?
    - [x] I downloaded it
    - [x] I built it

2. If you downloaded your rom, where did you get it from?
    - [x] Heads CircleCi
    - [ ] Purism
    - [ ] Nitrokey
    - [ ] Dasharo DTS (Novacustom)
    - [ ] Somewhere else (please identify)

Latest Git commit

3. If you built your rom, which repository:branch did you use?
    - [x] Heads:Master
    - [ ] Other (please identify)

4. What version of coreboot did you use in building?
24.12

5. In building the rom, where did you get the blobs?
    - [ ] No blobs required
    - [ ] Provided by the company that installed Heads on the device
    - [ ] Extracted from a backup rom taken from this device
    - [ ] Extracted from another backup rom taken from another device (please identify the board model)
    - [x] Extracted from the online bios using the automated tools provided in Heads
    - [ ] I don't know

## Please describe the problem

Graphics are garbled when using a 1440p panel on T420. This doesn't happen in the OS itself.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Have 1440p panel + mod board installed
2. Boot Heads
3. See image

**Expected behavior**
Graphics display as normal

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Additional context**
Likely missing display mode as 1440p isn't possible with any stock configuration, 1080p may be affected too.

tlaurion, 2025-04-01:
@haxalicious best guess is that vbt in coreboot is not correct for that panel, just like there are custom board config for x230 edp variant, that required lots of work/testing, after there was a coreboot patchset at least worked on upstream at https://review.coreboot.org/c/coreboot/+/28950

You can check edp variant vs non-edp variant for t420: you are first user known to use such panel.
Is that even supported by coreboot upstream, just like it was for x230?

There were efforts upstream (and myself) to get this merged not so long ago. You can look effort and follow rabbit from https://github.com/linuxboot/heads/pull/1604

Pull request/funding needed here. The EDP work from my side under https://github.com/linuxboot/heads/pull/1604 should lead the way, I do not plan to support all high resolution panels outside of standard HCL without funding/pull requests and upstream (coreboot) work.

tlaurion, 2025-04-01:
TLDR: this is not planned to work out of the box if coreboot doesn't support it outside of the box.

@haxalicious feel free to reopen this issue if coreboot supports such panel under the t420.

tlaurion, 2025-04-01:
Also see https://osresearch.net/Porting/

closed 2025-04-01 by tlaurion