linuxboot/heads #1956: Regenerating HOTP fails with ...

marmarek, 2025-04-16:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

1. What board are you using? (Choose from the list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))

x230

2. Does your computer have a dGPU or is it iGPU-only?
    - [ ] dGPU (Distinct GPU other then internal GPU)
    - [x] iGPU-only (Internal GPU, normally Intel GPU)

3. Who installed Heads on this computer?
    - [x] Insurgo (Issues to be reported at https://github.com/linuxboot/heads/issues)
    - [ ] Nitrokey (Issues to be reported at https://github.com/Nitrokey/heads/issues)
    - [ ] Purism (Issues to be reported at https://source.puri.sm/firmware/pureboot/-/issues)
    - [ ] Novacustom (Issues to be reported at https://github.com/Dasharo/dasharo-issues)
    - [ ] HardnenedVault (Issues to be reported at https://github.com/hardenedvault/vaultboot/issues)
    - [ ] Other provider
    - [ ] Self-installed

4. What PGP key is being used?
    - [ ] Librem Key (Nitrokey Pro 2 rebranded)
    - [ ] Nitrokey Pro
    - [x] Nitrokey Pro 2
    - [ ] Nitrokey 3 NFC
    - [ ] Nitrokey 3 NFC Mini
    - [ ] Nitrokey Storage
    - [ ] Nitrokey Storage 2
    - [ ] Yubikey
    - [ ] Other

5. Are you using the PGP key to provide HOTP verification?
    - [x] Yes
    - [ ] No
    - [ ] I don't know

### B. Identify how the board was flashed

1. Is this problem related to updating heads or flashing it for the first time?
    - [ ] First-time flash
    - [x] Updating heads 

2. If the problem is related to an update, how did you attempt to apply the update?
    - [x] Using the Heads menus
    - [ ] Flashrom via the Recovery Shell
    - [ ] External flashing

3. How was Heads initially flashed?
    - [ ] External flashing
    - [ ] Internal-only / 1vyprep+1vyrain / skulls
    - [ ] Don't know

4. Was the board flashed with a maximized or non-maximized/legacy rom?
    - [x] Maximized
    - [ ] Non-maximized / legacy
    - [ ] I don't know

5. If Heads was externally flashed, was IFD unlocked?
    - [ ] Yes
    - [ ] No
    - [ ] Don't know

### C. Identify the rom related to this bug report

1. Did you download or build the rom at issue in this bug report?
    - [x] I downloaded it
    - [ ] I built it

2. If you downloaded your rom, where did you get it from?
    - [x] Heads CircleCi
    - [ ] Purism
    - [ ] Nitrokey
    - [ ] Dasharo DTS (Novacustom)
    - [ ] Somewhere else (please identify)

    *Please provide the release number or otherwise identify the rom downloaded*

3. If you built your rom, which repository:branch did you use?
    - [x] Heads:Master
    - [ ] Other (please identify)

4. What version of coreboot did you use in building?
{ You can find this information from github commit ID or once flashed, by giving the complete version from Sytem Information under Options --> menu}

5. In building the rom, where did you get the blobs?
    - [ ] No blobs required
    - [ ] Provided by the company that installed Heads on the device
    - [ ] Extracted from a backup rom taken from this device
    - [ ] Extracted from another backup rom taken from another device (please identify the board model)
    - [ ] Extracted from the online bios using the automated tools provided in Heads
    - [ ] I don't know

## Please describe the problem

**Describe the bug**

After Heads update (via the menu) regenerating HOTP secret fails with message that it fails to unseal the HOTP secret. But I'm regenerating it exactly because it (as expected) couldn't be unsealed after the update!

**To Reproduce**
Steps to reproduce the behavior:
1. Fetch `heads-x230-hotp-maximized_usb-kb-v0.2.0-2731-gf572998.zip` from CI.
2. Update Heads via the menu.
1. Reboot the system after update, see "TOTP Generation Failed" message
2. Choose "Generate new HOTP/TOTP secret"
3. Confirm ("Yes")
4. See QR code, confirm to "configure your HOTP USB Security dongle"
5. See the message "Error PCR mismatch from TPM_Unseal"

**Expected behavior**

New secret written to the USB dongle (after the admin pin prompt).

**Screenshots**

[image]

**Additional context**
Add any other context about the problem here.

marmarek, 2025-04-16:
I tried also "Reset TPM" flow, with similar result.

tlaurion, 2025-04-16:
Asked same question under [Matrix thread of this issue](https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$SHSntf44m34igyy9GyRMpZrJj3-b_nwqRJ4mp1mzxXo?via=matrix.org&via=nitro.chat&via=tchncs.de)

> I cannot replicate this issue with x230-hotp-maximized (non-usb_kb), can you turn DEBUG output on under Options -> Change configuration settings -> Configure Heads informational output -> Debug
> 
> I am working on HOTP resealing/TPM TOTP resealing under https://github.com/linuxboot/heads/pull/1935
> 
> But nothing should prevent you to reset TPM1.2 on x230, which would take reownership of TPM and reseal TPMTOP

tlaurion, 2025-04-16:
My hypothesis is that unseal operations is not taking into consideration that usb HID driver is required in unsealing operation. Will try to find time testing this, thought this was tested as part of standard, non usb variant enabled through configuration?

marmarek, 2025-04-16:
[debug.log](https://github.com/user-attachments/files/19781939/debug.log)

tlaurion, 2025-04-16:
> [debug.log](https://github.com/user-attachments/files/19781939/debug.log)
> 
> The above is from executing the same flow as in the issue description.

Sorry @marmarek but the log provides is related to 
> Xen 4.21-unstable

Not Heads. Try again :)

tlaurion, 2025-04-16:
> Oops, wrong file, lets try again:
> 
> [debug.log](https://github.com/user-attachments/files/19782011/debug.log)

My log

[thierry-debug.log](https://github.com/user-attachments/files/19783046/thierry-debug.log)

`diff -u marek-debug.log thierry-debug.log`:

```
--- marek-debug.log	2025-04-16 14:22:30.397393800 -0400
+++ thierry-debug.log	2025-04-16 14:45:04.250000000 -0400
@@ -2,20 +2,20 @@
 TPM: Extending PCR[7] with filename /.gnupg/pubring.kbx and then its content
 TPM: Extending PCR[7] with hash 7ccf4f64044946cf4e5b0efe3d959f00562227ae
 LOG: exec stdout: New value of PCR[7]: 1beb88e70c9acf0fa187725205db206675dea4ab
-TPM: Extending PCR[7] with hash e6b956db64eb14ef1a22feef42576d18152d0c7d
-LOG: exec stdout: New value of PCR[7]: 8675167a245590244381acae46f90b9355b60e14
+TPM: Extending PCR[7] with hash 68affd0a2a8aa8b3c0385183a5e57ec41153065e
+LOG: exec stdout: New value of PCR[7]: 101de4a7cb48393c87f40dcb5a4cfc0ded37b21c
 Extracting CBFS file heads/initrd/.gnupg/trustdb.gpg into /.gnupg/trustdb.gpg
 TPM: Extending PCR[7] with filename /.gnupg/trustdb.gpg and then its content
 TPM: Extending PCR[7] with hash 7236ea8e612c1435259a8a0f8e0a8f1f5dba7042
-LOG: exec stdout: New value of PCR[7]: b6f378512ae1df0b71e105a983d6499aadcc6561
-TPM: Extending PCR[7] with hash 9c7fd882a5a3ec0322403a079ceca5dcc4a7f004
-LOG: exec stdout: New value of PCR[7]: 03acfe701eab6376abc594008bff0f6e3734f251
+LOG: exec stdout: New value of PCR[7]: ee4a9d98d87cbd9e14b936e4affe9c0a47fa7055
+TPM: Extending PCR[7] with hash ed593a95731274240266893865c6a6a7b220801b
+LOG: exec stdout: New value of PCR[7]: a6ab5b287f6ac0e7a6a5acf7a41dae5fafcbab5b
 Extracting CBFS file heads/initrd/etc/config.user into /etc/config.user
 TPM: Extending PCR[7] with filename /etc/config.user and then its content
 TPM: Extending PCR[7] with hash a0c02a9f82b98bc70224059ca53af19782ca0b28
-LOG: exec stdout: New value of PCR[7]: 42bef5f461984d203c7abcb8f9d0d5ec5c11747f
-TPM: Extending PCR[7] with hash 936e97235f2076c83dc362d8507604385ac766a9
-LOG: exec stdout: New value of PCR[7]: 88883d8ab62c642ef4d50527960695bd94f0bbe6
+LOG: exec stdout: New value of PCR[7]: 700dea51122158abb7a6b2881dbd7b31462cbe61
+TPM: Extending PCR[7] with hash bb52e739677822a892e4f3cb0417749a83684e13
+LOG: exec stdout: New value of PCR[7]: 4c5334aad32424a1969d4d5f11b202702f86e003
 TRACE: /init(102): main
 DEBUG: Full debug output enabled from this point: output both in dmesg and on console (equivalent of passing debug to kernel cmdline)
 DEBUG: NOTE: DO_WITH_DEBUG std_err and std_out will be redirected to /tmp/debug.log
@@ -92,14 +92,8 @@
 TRACE: /etc/functions(1046): is_gpt_bios_grub
 TRACE: /dev/sda1 is partition 1 of sda
 TRACE: /etc/functions(1023): find_lvm_vg_name
-TRACE: /etc/functions(1109): mount_possible_boot_device
-DEBUG: Partition /dev/sda1 is very small, likely BIOS boot. Skipping mount.
-TRACE: /etc/functions(1086): mount_possible_boot_device
-TRACE: /etc/functions(1046): is_gpt_bios_grub
-TRACE: /dev/sda2 is partition 2 of sda
-TRACE: /etc/functions(1023): find_lvm_vg_name
 TRACE: /etc/functions(1113): mount_possible_boot_device
-DEBUG: Try mounting /dev/sda2 as /boot
+DEBUG: Try mounting /dev/sda1 as /boot
 TRACE: /bin/gui-init(310): clean_boot_check
 TRACE: /bin/gui-init(338): check_gpg_key
 TRACE: /bin/gui-init(178): update_totp
@@ -128,7 +122,7 @@
 PCR-04: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 PCR-05: 15 E5 9E D1 7F 19 E8 B6 2F 35 7C 00 72 51 31 11 1A 10 06 35
 PCR-06: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
-PCR-07: 88 88 3D 8A B6 2C 64 2E F4 D5 05 27 96 06 95 BD 94 F0 BB E6
+PCR-07: 4C 53 34 AA D3 24 24 A1 96 9D 4D 5F 11 B2 02 70 2F 86 E0 03
 TRACE: /bin/gui-init(150): generate_totp_hotp
 TRACE: /bin/seal-totp(10): main
 DEBUG: Sealing TOTP with actual state of PCR0-3
@@ -152,15 +146,14 @@
 TRACE: /bin/tpmr(197): replay_pcr
 TRACE: /bin/tpmr(108): extend_pcr_state
[... 103 lines cut ...]
-DEBUG: Mounting /dev/sdb as read-write
+DEBUG: Mounting /dev/sdb2 as read-write

```

Analysis:
Well, for unknown reasons to me, your PCR4 is not zeroed (clean boot state).

Snippets of diff
```
-DEBUG: Replayed cbmem -L clean boot state of PCR=4 ALG=sha1 : 01a7743dbaff104014b36d254f5c3615cd9b7c51
+DEBUG: Replayed cbmem -L clean boot state of PCR=4 ALG=sha1 : 0000000000000000000000000000000000000000
```

So of course, trying to unseal from a expected clean state (without having been in recovery shell) fails and TPMTOTP secret sealed in nvram. Even our dynamic config saved under CBFS are the same:
```
 TPM: Extending PCR[7] with filename /etc/config.user and then its content
 TPM: Extending PCR[7] with hash a0c02a9f82b98bc70224059ca53af19782ca0b28
```

So runtime config is the same :(

I do not see anything explaninng why replayed PCR4 state from cbmem -L is different here. Will take another shot at this later.

@marmarek and what about tpm reset logs/use case?

tlaurion, 2025-04-16:
@marmarek 

> Only PCR:7 differs.
> EDIT: the runtime config is different:
> ```
>  Extracting CBFS file heads/initrd/etc/config.user into /etc/config.user
>  TPM: Extending PCR[7] with filename /etc/config.user and then its content
>  TPM: Extending PCR[7] with hash a0c02a9f82b98bc70224059ca53af19782ca0b28
> -LOG: exec stdout: New value of PCR[7]: 42bef5f461984d203c7abcb8f9d0d5ec5c11747f
> -TPM: Extending PCR[7] with hash 936e97235f2076c83dc362d8507604385ac766a9
> -LOG: exec stdout: New value of PCR[7]: 88883d8ab62c642ef4d50527960695bd94f0bbe6
> +LOG: exec stdout: New value of PCR[7]: 700dea51122158abb7a6b2881dbd7b31462cbe61
> +TPM: Extending PCR[7] with hash bb52e739677822a892e4f3cb0417749a83684e13
> +LOG: exec stdout: New value of PCR[7]: 4c5334aad32424a1969d4d5f11b202702f86e003
> ```
> 

I initially missed it.
Can you please upload your /etc/config.user?

Otherwise I have no clue of what creates our difference at runtime for the moment. And I doubt this will explain it :/

marmarek, 2025-04-16:
my config.user:
```
export CONFIG_TPM2_TOOLS=""
export CONFIG_TPM="y"
export CONFIG_AUTO_BOOT_TIMEOUT=""
export CONFIG_DEBUG_OUTPUT="y"
export CONFIG_ENABLE_FUNCTION_TRACING_OUTPUT="y"
export CONFIG_QUIET_MODE="n"
```

marmarek, 2025-04-16:
With debug enabled, I noticed cbmem SEGV on screen, could it be related?

[image]

marmarek, 2025-04-16:
`cbmem -L` output (it crashes before finishing):

[cbmem.txt](https://github.com/user-attachments/files/19784534/cbmem.txt)

I think something seriously went wrong there...

marmarek, 2025-04-16:
Should I dump flash content to compare against clean version? Or maybe the above cbmem.txt is enough that you could compare if it isn't corrupted flash content (few initial measurements seems to be saved correctly)?

tlaurion, 2025-04-17:
> Should I dump flash content to compare against clean version? Or maybe the above cbmem.txt is enough that you could compare if it isn't corrupted flash content (few initial measurements seems to be saved correctly)?

Since PCR2 is exactly the same, I take for granted the everything firmware related is the same, including the payload which is extracted and executed, creating our initramfs. I exclude SPI flash chip corruption.

@marmarek will test again. This would mean cbmem/coreboot had a regression when bumped to 24.12 and will probably need a coreboot bug report.

tlaurion, 2025-04-17:
> Current hpothesis is built rom corruption of cbmem tool, which is built with musl-cross-make to be added into tools.cpio and part of Heads recovery shell packed tools. Digging.

Nope, roms from CI and local builds were reproducible. @marmarek you found a bug in coreboot 24.12 cbmem/coreboot measured boot attestation parsing from cbmem.

repro
```
user@heads-tests-deb12-nix:~/heads$ git checkout f572998ea4db1581b15f29e313380e00407a09d2
HEAD is now at f572998ea4 Merge pull request #1953 from tlaurion/patch_revert_if_apply_fails-canary_wipe_rebuild_improvements
user@heads-tests-deb12-nix:~/heads$ ./docker_repro.sh make BOARD=x230-hotp-maximized_usb-kb
[...]
user@heads-tests-deb12-nix:~/heads$ wget -q https://output.circle-artifacts.com/output/job/e3d862e0-7088-49e5-af2c-fe0a3b6d6f08/artifacts/0/build/x86/x230-hotp-maximized_usb-kb/hashes.txt -O hashes.txt 
user@heads-tests-deb12-nix:~/heads$ HASHES1=hashes.txt
user@heads-tests-deb12-nix:~/heads$ HASHES2=/home/user/heads/build/x86/x230-hotp-maximized_usb-kb/hashes.txt 
user@heads-tests-deb12-nix:~/heads$ egrep '^[0-9a-f]{64}' $HASHES1 | while read line; do HASH_REF=$(echo $line|awk -F " " {'print $1'}); FILE_REF=$(echo $line|awk -F "/" {'print $NF'}); if ! grep -q "$HASH_REF" $HASHES2; then echo "$FILE_REF doesn't match";fi; done
user@heads-tests-deb12-nix:~/heads$
```

tlaurion, 2025-04-17:
Also tried to mitigate TPM 1.2 measured boot event log replay issue by switching to prior used "coreboot custom format" for TPM event log, but this is also broken, resulting in `cbmem -L` errlog output of "Unknown TPM log specification"

Config change:
```
diff --git a/config/coreboot-x230-maximized.config b/config/coreboot-x230-maximized.config
index ea9f5c9a13..576c364f99 100644
--- a/config/coreboot-x230-maximized.config
+++ b/config/coreboot-x230-maximized.config
@@ -568,8 +568,8 @@ CONFIG_TPM=y
 CONFIG_MAINBOARD_HAS_TPM1=y
 # CONFIG_TPM_DEACTIVATE is not set
 # CONFIG_DEBUG_TPM is not set
-# CONFIG_TPM_LOG_CB is not set
-CONFIG_TPM_LOG_TPM1=y
+CONFIG_TPM_LOG_CB=y
+# CONFIG_TPM_LOG_TPM1 is not set
 # CONFIG_TPM_LOG_TPM2 is not set
 CONFIG_TPM_MEASURED_BOOT_RUNTIME_DATA=""
 CONFIG_PCR_BOOT_MODE=1

```

`cbmem -L -V > log 2>&1` log:
[cbmem_L_V_coreboot_custom_format_broken.log](https://github.com/user-attachments/files/19797888/cbmem_L_V_coreboot_custom_format_broken.log)

tlaurion, 2025-04-17:
Piggybacked rebasing to 24.12 issue of Dasharo first to see if they encountered the issue yet at https://github.com/Dasharo/dasharo-issues/issues/1241#issuecomment-2813523903

tlaurion, 2025-04-17:
@marmarek 

> Should I dump flash content to compare against clean version? Or maybe the above cbmem.txt is enough that you could compare if it isn't corrupted flash content (few initial measurements seems to be saved correctly)?

Yes please. 
- mount-usb --mode rw
- flashprog --programmer internal -r /media/1956.rom
- umount /media
- upload 1956.rom here (might need to be zipped)

I upgraded to https://output.circle-artifacts.com/output/job/1b40b319-5288-4cb1-b6ae-208f04046802/artifacts/0/build/x86/x230-hotp-maximized_usb-kb/heads-x230-hotp-maximized_usb-kb-v0.2.0-2733-g7626f62.zip and now I cannot replicate....

I did bunch of clean flash (no preserving of settings, so equivalent of `flashprog --programmer internal -w`, and seem to have cleared what cause the segfault.

tlaurion, 2025-04-17:
@miczyg1 answered in dasharo general channel.

I forgot to cherry-pick patch that is still unmerged upstream.... https://review.coreboot.org/c/coreboot/+/84926

Testing

closed 2025-04-18 by tlaurion