linuxboot/heads #1960: 3mdeb/Dasharo has discontinued Heads on MSI desktops, time to upstream?

mblanqui, 2025-04-21:
3mdeb has discontinued all development of Heads for their coreboot distribution on MSI desktops.

> @scallyob great, thanks for testing. However, as the heads product on MSI desktops is discontinued (meaning we don't plan to provide new release for it), everybody will have to rebuild their ROMs themselves to fix it :(  

 _Originally posted by @miczyg1 in [#1189](https://github.com/Dasharo/dasharo-issues/issues/1189#issuecomment-2757761242)_

Could development/updates be handled upstream from now on?

miczyg1, 2025-04-22:
> Could development/updates be handled upstream from now on?

That would be ideal. But somebody would have to take this effort.

@tlaurion already tried very hard to coordinate it and nobody at 3mdeb has any spare resources to do it.

miczyg1, 2025-08-26:
> [@Tonux599](https://github.com/Tonux599) attempted port recently (he uses his fork meanwhile) [#1962 (comment)](https://github.com/linuxboot/heads/pull/1962#issuecomment-2899371045)
> 
> > This is stalled until the stability issue [#1962 (comment)](https://github.com/linuxboot/heads/pull/1962#issuecomment-2851091754) is resolved.
> > Until then the following is what I'm using on my Z790-P (DDR5) board. It uses Dasharo's v0.9.2 release.
> > https://github.com/Tonux599/heads/tree/my-custom-msi_z790p_ddr5
> 
> For reference, stability issue is
> 
> > > [@miczyg1](https://github.com/miczyg1) I'm getting frequent kernel panics (in heads only, not when booted to Qubes) like:
> > > ```
> > > mce: CPUs not responding to MCE broadcast (may include false positives): 2-3
> > > Kernel panic - not syncing: Timeout: Not all CPUs entered broadcast exception handler
> > > ```
> > >   
> > > Looking around about MCE's, I came across your issue below, do you think they are related? https://mail.coreboot.org/hyperkitty/list/coreboot@coreboot.org/thread/MW44TIEMFMVDWPVCAFPE2QUFXXGZYYAX/
> 
> [@Tonux599](https://github.com/Tonux599) [@miczyg1](https://github.com/miczyg1) status update?

None at this point. For some reason it doesn't trigger on our releases.