linuxboot/heads #1963: LUKS change passphrase sets an empty passphrase

gaspar-ilom, 2025-04-27:
When changing the luks passphrase, an empty passphrase is set. The user is not even asked to enter a passphrase.
Afterwards the user cannot decrypt their OS anymore with the luks disk recovery passphrase. From the prompts in heads and OS'es such as Qubes it is not possible to enter an empty passphrase when booting and decrypting the passphrase. Therefore, besides the obvious confidentiality/integrity issue with using an empty passphrase key file, non-technical user will also lose access and might believe their disk is no longer decryptable.

**To Reproduce**
Steps to reproduce the behavior:
1. Select Luks change passphrase in the heads gui
2. Type the current passphrase and confirm
3. See that you are not asked for a new passphrase
4. Try booting the encrypted OS and type the old passphrase or an empty passphrase
5. See that you cannot decrypt it anymore

**Expected behavior**
The user should be asked to provide a new passphrase and the new passphrase should replace the old one. Afterwards it should be possible to decrypt withe the new passphrase but no longer with the old one.

**Recovery**
For those that were already hit by the bug:

1. Go to a recovery shell
2. Create an empty file `touch empty_pass`
3. Replace the empty passphrase with a proper one: `cryptsetup luksChangeKey <device> --key-file empty_pass`, where device is something like `/dev/sda2`.
4. Enter the new passphrase (twice) and confirm.
5. You can use the new passphrase to decrypt the disk/OS now.

tlaurion, 2025-04-27:
Seems a legit bug that happens when passphrase is changed without reencryption (which is not recommended since old luks header can be used, and was not tested properly consequently).

Will review proposed fix tomorrow.

tlaurion, 2025-04-28:
@gaspar-ilom Thanks for reporting this bug and a fix.

closed 2025-04-28 by tlaurion

gaspar-ilom, 2025-04-29:
> [@gaspar-ilom](https://github.com/gaspar-ilom) Thanks for reporting this bug and a fix.

Thanks @tlaurion for testing and merging it so promptly!