linuxboot/heads #1975: QSB-107 -  Multiple CPU branch prediction vulnerabilities - WILL AFFECT < 8th gen CPU forever

tlaurion, 2025-05-20:
QubesOS posted [QSB-107](https://github.com/QubesOS/qubes-secpack/blob/master/QSBs/qsb-107-2025.txt)

- The description clearly states that CPU gen up to 7th gen were tested affected with all mitigations applied (xen + CPU microcode updates)
- We know that EOL CPUs are not tested in new vulnerability bulletins
- We know Intel doesn't say older gens then supported ones at time of vuln discovery are affected.
- So at the time of writing those lines, < 8th gen CPUs are not supported anymore, meaning no microcode update will be made available to older then 8th gen cpus
- I made an announceemnt and a call for comments on how to update the docs in [matrix channel post](https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$--1YLz_7gNgJJGIIzKYZQbSJaM1JEhPkhZkP_LFWCMs?via=matrix.org&via=nitro.chat&via=tchncs.de) which showed a lot of misunderstanding about how this works
- Heads documentation already states that older hardware not receiving microcode updates have limited mitigation, meaning users wanting a plug-and-pray security experience need to use more recent hardware otherwise putting themselves in possible danger of leaking some inforation through transcient execution vulnerabilities at https://osresearch.net/Heads-threat-model/#binary-blobs-microcode-updates-and-transient-execution-vulnerabilities

I posted this to be built upon and update something in the docs or the boards, I do not know and is where I look for advice, not being sure who/where this should be documented....

----

On **Multiple CPU branch prediction vulnerabilities [QSB-107](https://github.com/QubesOS/qubes-secpack/blob/master/QSBs/qsb-107-2025.txt)** : that CPU vulnerability affects T480 and all older Intel platforms currently under Heads, tested up to 7th gen CPU (so we can generalize safely to Sandy(gen 2: xx20;x220), Ivy (gen 3:xx30;x230), Haswell (gen 4: x4xx; t440p), Skylake (gen 6th), KabLake (gen 7th), Coffee Lake (gen 8th: t480).

**TLDR: Oldest Intel CPU gen still receiving CPU microcode updates _today_ is Coffee Lake (Intel 9th gen)**

- https://github.com/linuxboot/heads/blob/master/BOARD\_TESTERS.md

---

Actually, Speculation vulnerabilities discovered in 2017-up to recently deployed CPU based microcode mitigations never really preventing some speculation attacks (even thought we thought they did.... until it was discovered they didn't, again, recently...). The point being here, even if you run QubesOS; if you run unsafe stuff under a disposable qube that triggers specifically those CPU vulnerabilities; then secrets supposedly protected in other qubes running in parallel are not properly protected. This was discussed again and again... The speed of accessing secrets through speculation vulnerabilities is still slow; but fact remains: it is becoming difficult to defend platforms against users who most probably do not have the proper opsec to not run qubes that are unsafe (disposable qubes) in parallel of qubes that are meant to protect secrets (vault)... In an OS (QubesOS) that is supposed to protect the user exactly against that.

---

I'm not sure this deserves a separate documentation update; but if it does, how to properly do it. Your opinion matters on that: microcode updates and _EOL platforms support is a moving target_.

Current docs at https://osresearch.net/Heads-threat-model/#binary-blobs-microcode-updates-and-transient-execution-vulnerabilities

Enough? Thoughts?

tlaurion, 2025-05-20:
I covered things back in 2022 https://forum.qubes-os.org/t/short-list-of-laptops-desktops-that-work-well-with-qubes-os/5197/245

Thing is situation is different now since new research confirms that if things are aligned differently, previous speculation mitigation can be bypassed altogether.

tlaurion, 2025-05-20:
Best I could do now is cross-link things
- https://forum.qubes-os.org/t/short-list-of-laptops-desktops-that-work-well-with-qubes-os/5197/326
- https://forum.qubes-os.org/t/qsb-107-multiple-cpu-branch-prediction-vulnerabilities/33925/2

Hopefully, this issue will get enough attention so that we, as an ecosystem, document this properly.

notgivenby, 2025-05-21:
@tlaurion Should all vulnerable platforms be moved to LIKELY_UNSAFE category? The label perhaps should also appear in the rom name? So, in case of folks building the HEADS themselves or downloading the rom they are at least triggered to ask or search info in the docs? As the experience shows, very few indeed study the docs in detail…
In general, it is quite sad that we all are forced to accept the more blobs and „buy recent hardware“ policy from big techs. But perhaps this is inevitable reality.

thickfont, 2025-05-23:
Can you qualify what you mean by "forever"?

Is there reason to believe future software mitigations are impossible?

_I understand currently only CPU microcode updates remediate the vulnerability and all EOL CPUs aren't getting microcode updates._

I searched far and wide for any indications of software mitigations being off the table, but didn't find anything. Am I missing something? Are software mitigations CPU independent? If they are CPU dependent like microcode updates, then I can see a world where only >8th Gen CPUs get software mitigation solutions and can understand the pessimism. Some insight is appreciate on this.

From the Researchers' blog post:
https://comsec.ethz.ch/research/microarch/branch-privilege-injection/
> We have also evaluated several potential alternative mitigation strategies in software with overheads between 1.6% (Coffee Lake Refresh) and 8.3% (Rocket lake). Please refer to our [paper](https://comsec.ethz.ch/wp-content/files/bprc_sec25.pdf) for more details.

I've gotten the sense that a large part of the QubesOS/Heads community has given up on old hardware. Am I crazy to think it's too early to call it so decisively?

notgivenby, 2025-05-23:
@thickfont Perhaps it is not that bad as it looks…and you indeed may be correct. I think now there is no PoC on Qubes OS. Did you read the paper in more detail? I just did. They used KVM and run everything on host system kernel in hypervisor attack. So the situation in Qubes is different as from the Marek answer in matrix forum in Qubes channel dom0 is privileged VM. So perhaps, disclaimer i am not an expert in the field…it is still okay to use older hardware. In the paper, also a software mitigation was described: retpoline. But in should be proved, until then then we need to assume the worse….

tlaurion, 2025-05-23:
Updated threat modeling page to point to relevant Intel pages at https://github.com/linuxboot/heads-wiki/commit/68301d7c9e61c07e6f29918c168c0522b889a8ab

tlaurion, 2025-05-23:
> Updated threat modeling page to point to relevant Intel pages at [linuxboot/heads-wiki@68301d7](https://github.com/linuxboot/heads-wiki/commit/68301d7c9e61c07e6f29918c168c0522b889a8ab)
> 
> Still unsure on [#1976](https://github.com/linuxboot/heads/pull/1976) (renaming boards to LIKELY_INSECURE is way too long board names, to be honest Threat model page should be enough with warning in boards configs affected?)

I reverted radical changes (board rename) under #1976 : for now, vulnerable boards have big fat warning added under board configs.

@JonathonHall-Purism @marmarek further advice?

tlaurion, 2025-05-23:
> [@thickfont](https://github.com/thickfont) Perhaps it is not that bad as it looks…and you indeed may be correct. I think now there is no PoC on Qubes OS. Did you read the paper in more detail? I just did. They used KVM and run everything on host system kernel in hypervisor attack. So the situation in Qubes is different as from the Marek answer in matrix forum in Qubes channel „ dom0 is privileged VM.“ So perhaps, disclaimer i am not an expert in the field…it is still okay to use older hardware. In the paper, also a software mitigation was described: retpoline it was not helpfull against all types of attacks. But all the things should be proved, until then then we need to assume the worse….

[qubes-public matrix discussion input](https://matrix.to/#/!WtRrlYUTHOQjqGcSnn:invisiblethingslab.com/$fLRMUtLAEc0WuInaLVUbnLZ4VwEfzsZhNZlt3MMCJFc?via=invisiblethingslab.com&via=matrix.org&via=nitro.chat) by @marmarek

>> I am reaching you regarding the QSB-107. I read the QSB-107, the paper from ETH (https://comsec.ethz.ch/wp-content/files/bprc_sec25.pdf) and looked at their github repo. The guest to supervisor attack was done on KVM. Does the concept work on qubes/xen? Did somebody really check the PoC? To enable the PoC an update of the GRUB config is nessesary ´´´update the GRUB_CMDLINE_LINUX_DEFAULT by adding "nosmap nosmep clearcpuid=295,308 mitigations=off"´´´. I believe that must be done from dom0 first correct? Since, I lack the expertise the confirmation of the attack working on xen, taking in considiration the severity of the impact, could really help numbers of community members still using older hardware.
> their PoC would require very substantial changes to work on Xen, as it currently a) uses KVM API for creating a VM for test, and b) assumes you run it on the host system (the kernel which hosts hypervisor), while in Xen, you don't have that possibility at all - dom0 is just another VM (privileged one, but still a VM)

On which I have not much to add.

> 
> Edit: we perhaps just need to test the vulns first on our hardware with ubuntu installed. [@tlaurion](https://github.com/tlaurion) did you have a chance to test your x230? I will try to test my t430. [@thickfont](https://github.com/thickfont) can you check t480s? I now do not have any spare ssd to install ubuntu on my t480. Not sure wether live ubuntu will work…

My practical, QA perspective answer:
- We could test, yes. But my take is that qsb-107 makes me personally not want to recommend the x230 for qubesos users looking for a plug-and-pray experience anymore (where users are expected to run tor browser in a dvm alongside their vault's keepassxc open database and their LUKS container opened, decrypted... With untrusted qubes running overnight and trying to exfiltrate data at around 5.6kib/s for available, yet not reproduced, rate. I know its "use it at your own risk" but having sold hardware, having done paidd support for QubesOS users.... We have to be honest here. Most want a plug and pray experience. Ivy bridge x230: 3rd gen Intel CPU) up to 8th gen intel CPU will not receive any microcode updates anymore (gen 8th last microcode update to be received forever was 31 december 2024: this is t480 btw), so No I do not feel confortable recommending those machines for a plug-and-pray experience (hear me out:I am always against plug-and-pray experience, contributed to a lot of articles having been written over the year, but people don't read. One such good article is https://www.anarsec.guide/posts/qubes/ 

I am not anymore comfortable recommending hardware that is not tested anymore in any serious article (x230). 
I was felling uneasy at https://forum.qubes-os.org/t/short-list-of-laptops-desktops-that-work-well-with-qubes-os/5197/326, now i'm not confortable. Will retpoline realignment fix the day? Maybe.

But yet again, hear me out.
- using x230 for whisleblowing, to boot Tails on a single purpose workflow, where no states are to be kept on the machine: yes. Still yes. Lesser blobs level, people using Tails typically more conscious of states, or scared of them; they do not wish a plug-and-pray and will use the bare minimal needed in their work session and poweroff when they are done.
- using x230 for qubes workflow: means a lot of states to be protected. People using Heads (other side of tails) is typically to safeguard their qubesos workflows. So here again, if goal is to leave laptop powered on for days with unsafe workflows in paralell of doing private computing: this is not recommendable anymore. 

---

My theoritical background based answer:
The problem here @notgivenby with retpoline is that at the time it was introduced, there was traction for vulnerable systems that would/could not receive microcode updates already. Embargoed work happened under linux kernel in December 1997 up to public vuln disclosure in Jan 2018 for Spectre/Meltdown. To be noted that Spectre V2 was never fully patched by microcode updates for Ivy bridge and Ivy bridge was vastly used back in that time: those speculative vulnerabilities being a first in computing history with such massive impact and creating such havoc. That was 2018, retpoline since then went dynamic if no microcode available, and all was somewhat still ok with conditions to trigger other important vulns not all being theoretically met to recommend Ivy bridge to be retired; but this is somewhat different now. We are now in 2025: retpoline is used only for now really old hardware; there is recommen [... cut ...] ntains blobs, Risc-V still not fast enough and not completely open docs/IPs, ARM IP became a little more interesting somewhat...

TLDR: since 2018: the world accepted that CPU have programmed obsolescence as well as for their phone, cars, and anything having electonics, software and firmware (blobs), even more importantly for confidential workflows; microcode updates are offered for a platform around 6 years after manufacturing date and this is the world we live in now.

Sorry for the long post.

JonathonHall-Purism, 2025-05-26:
> QubesOS posted [QSB-107](https://github.com/QubesOS/qubes-secpack/blob/master/QSBs/qsb-107-2025.txt)
> 
>     * The description clearly states that CPU gen up to 7th gen were tested affected with all mitigations applied (xen + CPU microcode updates)

@tlaurion Am I missing this?  I see in the linked document:

>In particular, COMSEC observed
>that BPRC affects CPUs as far back as 7th Generation Intel Core. [4]

They tested _back to_ 7th for BPRC.  Intel confirmed BPRC for 8th and onward (and IIRC they received microcode updates).

This leaves _only_ 7th gen unaddressed by microcode.  And as you said retpoline is used for "really old hardware".  I believe 7th is in that group.

tlaurion, 2025-05-29:
Maybe `EOL` prepending all EOL boards not receiving microcode updates?
Point is that mostly all boards under Heads would, with exception of newer boards from Librems and Novacustom

PR #1976 (2025-05-30): Intel 8th gen and older ESU ended boards configs: add big fat warning, document BOARDS_AND_TESTERS.md
Proposition to fix #1975 

EDIT:
- not all 8th gen CPUs are the same. [qubes-public matrix channel post](https://matrix.to/#/!WtRrlYUTHOQjqGcSnn:invisiblethingslab.com/$BYrC39tttLRlZU7wzbeGT_cJZzsVVybRv2F0cCR_QFI?via=invisiblethingslab.com&via=matrix.org&via=nitro.chat):

> 
> - **SOME** Intel 8th Gen CPU are supposed to have received microcode updates and ARE NOT considered vulnerable to qsb-107 AFAIK:
>     - t480/t480s  is 8th gen Kaby Lake Refresh : Intel Core i7-8650U (ESU ended 12/31/202 [... cut]
closed 2025-05-30 by tlaurion