linuxboot/heads #1979: web.archive.org down/having issues. What to use when wb archive fails?!

tlaurion, 2025-06-06:
Which we rely on to download sinit blob directly at https://github.com/linuxboot/heads/blob/9fabe7f26cacea874fac6502ddded564ee2f8af5/blobs/xx30/optiplex_7010_9010.sh#L37  

since Intel removed sandy/ivy bridge sinit blob to fix https://github.com/linuxboot/heads/pull/1916  (3mdeb having same issue at https://github.com/linuxboot/heads/pull/1916)

```
--2025-06-06 15:07:39--  http://web.archive.org/web/20230712081031/https://cdrdv2.intel.com/v1/dl/getContent/630744
Resolving web.archive.org (web.archive.org)... 207.241.237.3
Connecting to web.archive.org (web.archive.org)|207.241.237.3|:80... connected.
HTTP request sent, awaiting response... 500 INTERNAL SERVER ERROR
2025-06-06 15:08:51 ERROR 500: INTERNAL SERVER ERROR.
```

```
--2025-06-06 14:53:28--  http://web.archive.org/web/20230712081031/https://cdrdv2.intel.com/v1/dl/getContent/630744
Resolving web.archive.org (web.archive.org)... 207.241.237.3
Connecting to web.archive.org (web.archive.org)|207.241.237.3|:80... connected.
HTTP request sent, awaiting response... 503 Service Unavailable
2025-06-06 14:53:33 ERROR 503: Service Unavailable.

```

mblanqui, 2025-06-19:
I suppose **someone who wasn't involved in Heads** could upload them to the IPFS, then Heads' build files can link to them via an IPFS gateway (perhaps with fallback options if one goes down?). An additional benefit would be that each chunk of the file distributed across IPFS is checked via a cryptographic hash, essentially making the blobs tamper proof.

tlaurion, 2025-06-20:
> I suppose **someone who wasn't involved in Heads** could back them up to the IPFS, then Heads' build files can link to them via an IPFS gateway (perhaps with fallback options if one goes down?). An additional benefit would be that each chunk of the file distributed across IPFS is checked via a cryptographic hash, essentially making the blobs tamper proof.

I guess that could work; while problem with this is the non-redistributive nature of blobs, here from Intel, on non-versionned archives, and the reliance we cannot have on web.archive.org forever, nor on IPFS "gateway" (so web url) that can one day become unavailable. Here we are lucky that someone took a snapshot of the tarball url, which contained the same blob (proven by hash) even if in an earlier version. This issue was created when archive.org couldn't provide that archive; where now Intel doesn't provide the blob itself anymore; having decided to recreate that archive without sandy/ivy bridge sINIT blob.... 

Again, Heads deals with this for most build related tarballs/zip, thanks to coreboot having mirror of buildsystem related tarballs, and purism hosting a mirror of everything heads tarballs related.... But when comes to blobs related archives, as of now we rely on Internet Archive, which proved to possibly fail as well in the future. And we cannot of course ask Purism to host a blob the is not redistributor  

TLDR:
- if archive.org goes down, or are attacked (again) and for some reason, the sinit archive cannot be downloaded: Heads commit won't succeed CI build for boards depending on it (and won't have a green check under github for that commit ever
- if for whatever reason, someone tries to build older commit https://github.com/linuxboot/heads/commit/a084cc7b4a147484429a070cb101441ebd411fc4 (which preceded using archive.org at commit https://github.com/linuxboot/heads/commit/a084cc7b4a147484429a070cb101441ebd411fc4 : it will fail forever in the future. 
  - This is a real limitation of "reproducible builds" in the sense that one won't be able to reproduce, ever, https://github.com/linuxboot/heads/commit/a084cc7b4a147484429a070cb101441ebd411fc4 because of that (the final rom will have different hash because fixing https://github.com/linuxboot/heads/commit/a084cc7b4a147484429a070cb101441ebd411fc4 will create another commit and therefore rom won't be reproducible).
- Same issue would happen with IPFS gateway in the future, when that gateway will fall because XYZ.

---

TLDR of TLDR: 
- @mblanqui but yes, I guess the ideal would be to have multiple sources for each blobs in a next commit, hoping that at least one of the archived archive will stay accessible forever
- This issue was opened as a consequence of a hiccup of archive.org; its fixed now.... until next time.